cedricp/ddt4all #6: Changes not applied

valenciakva, 2017-09-11:
Thanks in advance for your help.  I am using a ODBLink SX connected via USB to a Windows 10 laptop.  I installed the program, started the program, selected the USB device (Last one at the right), COM Port and the "Connected" option.  After I click in connect no error message is displayed but when I try to make a change by enabling Expert Mode and clicking on "Write" nothing is written.  I am attaching a screenshot. As you can see, it says connected in the bottom left but the ELM is not giving a response back.
The car is a Koleos 2016 with a RLINK 1 system.

To make these changes, I created a cable to be connected between the ODB port in the car and the ODBLinkSX:

ODBLinkSX -Car ODB connector
6-13
14-12
4-4
5-5
16-16
Note: 4 and 5 are joined on ODBLinkSX side

I have tested with the car switch in ON and the motor running with the same results.  I haven't find a way of reseting the RLink but I don't think that's the issue.

Thanks for your help.
[image]

gordon6, 2017-09-11:
Hello, Did you tried without the additional connector..?  Renault use C-CAN and M-CAN both.
Normal ECU used to C-CAN(without additional connector).  Multimedia devices used to M-CAN(with connector as you described).

valenciakva, 2017-09-11:
Thanks, yes, I connected without the custom cable, results were the same.  Considering this is a change to the Navigation system called RLink, I think I need to use the connection.

- Is my wiring OK?  I am not sure if 4 and 5 should be joined or even wired
- I am selecting "Connected mode" when the program starts, is this correct?
- I select COM5 (I confirmed this is the port for the USB SX) and then the icon at the right:

[image]

- After I click on "Connected Mode" I see the following screens:
[image]

As you can see, the option to write is in "yellow", not in red, is this an indication I am not connected?
[image]

- Do I need to change something in the COM port configuration? I tried to connect using the OBDWiz software that came with the dongle without success.

Thanks again for your help.

enandrei, 2017-09-12:
Color changes once you start modifying the options.

I am trying the same thing from ubuntu, no luck.

Setup : - Chinese ELM clone running 1.5 (tested with ELM identifier working with commands up to 1.4) (also other changes on normal can bus are possible without errors)
           - Pins remapped as described earlier in the topic
           - Running on a different car (2014 Megane 3)

I am getting some errors though :

ECU uses SDS 10C0
Bad ELM response : WRONG RESPONSE : Unknown
Bad ELM response : WRONG RESPONSE : Unknown
Sending request :StartAfterSalesSession
Switching to session mode 10 C0
Sending request :DataWrite.ECU
Sending ELM request:2E 6C 1C 38
Bad ELM response : WRONG RESPONSE : Unknown
ELM response : WRONG RESPONSE : Unknown
Sending request :StartDiagnosticSession
Switching to session mode 10 81
Bad ELM response : WRONG RESPONSE : Unknown
Bad ELM response : WRONG RESPONSE : Unknown

At a previous attempt i was getting "Wrong response CFC0" . Had 2 more tries and did not show up. Maybe we need not run flow control on multimedia can ?

There is a radio device on the same multimedia can, i was able to write to that one.
And also , all these devices are properly detected when i scan.

valenciakva, 2017-09-12:
Thanks.  I am receiving exactly the same reply (below) and please also notice that my buttons are not changing from yellow to red when I make changes.

Sending request :DataWrite.ECU
Sending ELM request:2E 6C 1C 38
ELM response : WRONG RESPONSE : Unknown
Sending request :StartDiagnosticSession
Switching to session mode 10 81

Any other ideas?  I saw in a thread in a Renault forum (here:https://goo.gl/TiFh6R) that this change was successful using your software and I am using the same cable described there.

Thanks.

cedricp, 2017-09-13:
Are you up to date ? I fixed a similar bug some times ago

enandrei, 2017-09-13:
Using latest master , also tries version from the 25th of august ( date used on the forum where it was working) , same issue , different message .

cedricp, 2017-09-13:
Ok, try to increase CAN timeout value to 1000 ms (bottom of the window), I had similar issue, and was solved this way

enandrei, 2017-09-13:
tried it, no luck.

here is a log https://pastebin.com/NvbjubgA
i get the error when loading the screen as well (radio ecu is fine, no errors)

also, after second scan errors are looping until i select an ecu

script output is this : 
Using custom DDT database
Initilizing ELM with speed 115200...
Trying to open port/dev/rfcomm0 @ 115200
Scanning ECUs...
1366 loaded ECUs in database.
Skipping CAN extended address (not supported yet)  2D
Warning, address 400 is not mapped
Skipping CAN extended address (not supported yet)  DA
Skipping CAN extended address (not supported yet)  D2
Skipping CAN extended address (not supported yet)  41
Skipping CAN extended address (not supported yet)  60
Skipping CAN extended address (not supported yet)  2D
Warning, address 400 is not mapped
Skipping CAN extended address (not supported yet)  DA
Skipping CAN extended address (not supported yet)  D2
Skipping CAN extended address (not supported yet)  41
Skipping CAN extended address (not supported yet)  60

during a previous session i got this as well : 
Scanning ECUs...
1366 loaded ECUs in database.
Traceback (most recent call last):
  File "/home/ion/02_ddt/ddt4all-master/parameters.py", line 79, in tester_send
    self.sendElm(self.tester_presend_command, True)
  File "/home/ion/02_ddt/ddt4all-master/parameters.py", line 691, in sendElm
    elm_response = options.elm.request(command, cache=False)
  File "/home/ion/02_ddt/ddt4all-master/elm.py", line 663, in request
    self.vf.write(tmstr + ";" + dnat[self.currentaddress] + ";" + req + ";" + rsp + "\n")
KeyError: '7e0'
Skipping CAN extended address (not supported yet)  2D
Warning, address 400 is not mapped

thank you for looking at this :)

cedricp, 2017-09-13:
Would be better to send me the log file in 'log' directory (elm_ddt)

valenciakva, 2017-09-13:
I am adding my logs in case that helps.  I am having similar issues.

[logs.zip](https://github.com/cedricp/ddt4all/files/1299532/logs.zip)

This is a extract of what seems to show the problem: 

############################################################
#connect to: BICA2A3 08 V12 Addr:59
############################################################
>[20:34:23.218]AT SH 714
<[0.016]AT SH 714
OK
>
>[20:34:23.234]AT CRA 734
<[0.016]AT CRA 734
OK
>
>[20:34:23.250]AT FC SH 714
<[0.016]AT FC SH 714
OK
>
>[20:34:23.266]AT FC SD 30 00 00
<[0.016]AT FC SD 30 00 00
OK
>
>[20:34:23.282]AT FC SM 1
<[0.016]AT FC SM 1
OK
>
>[20:34:23.298]AT SP 6
<[0.016]AT SP 6
OK
>
>[20:34:23.314]0210C0
<[0.208]0210C0
NO DATA

cedricp, 2017-09-13:
Try with last version (updated now), I'm quite sure the problem is CAN timeout, a user reported me it worked.

enandrei, 2017-09-13:
just updated , same thing (saw the 400 timeout, increased to 1000)

here is the log : https://pastebin.com/5VasDEbC

cedricp, 2017-09-13:
Are you sure you are using the right XML file, the ECU responds "7F 02 11" to request "10 C0". That means "11: service not supported"

cedricp, 2017-09-13:
Did you try to scan the CAN bus to autodetect ?

enandrei, 2017-09-13:
Yes , previously scanned , and that was the right XML

see https://pastebin.com/NvbjubgA -> Identified ECU : MFD_v4.3_20160317T110748.xml DIAGVERSION [20] SUPPLIER [329] SOFT [2D2D] VERSION [10CB]

cedricp, 2017-09-13:
Are you able to communicate with other ECUs ?

enandrei, 2017-09-13:
Yes, on same Multimedia can i found 3 devices : 

Identified ECU : Radio_2012_V2.4_20130604T134716.xml DIAGVERSION [116] SUPPLIER [BT2] SOFT [0300] VERSION [0110]
Identified ECU : BIC_A2_A3_v22_20170602T161817.xml DIAGVERSION [20] SUPPLIER [DX2] SOFT [0620] VERSION [0000]
Identified ECU : MFD_v4.3_20160317T110748.xml DIAGVERSION [20] SUPPLIER [329] SOFT [2D2D] VERSION [10CB]

The MFD and BIC error when i load screens, but the Radio one works fine with the exact same setup.

Also connected to the normal can bus and did not see errors (without the pinout adaptor ).

cedricp, 2017-09-13:
Would be interesting to lower speed communication...
Can you try to open XML file and replace 
`<CAN BaudRate="10400" Multipoint="1">`
with
`<CAN BaudRate="250000" Multipoint="1">`

yoh-there, 2017-09-13:
AFAIK the MM bus on Renault is "always" running on 250Kbps indeed. Not sure if this is a helpful comment ;-). It is on ZOE. Mainbus on 500kbps.

cedricp, 2017-09-13:
If the MM bus is always 250Kbps, trying to change the parameter worth it. Because actually DDT4all runs at 500Kbps.
.

valenciakva, 2017-09-13:
I wanted to report back my latest results.  I thinks this thread now contains two different reports that might be linked. One from "enandrei" and mine that started the thread.  For mine, please consider that I am testing with a Windows 10 machine in a Koleos 2016 with RLink 1.

I successfully changed a parameter in CAN 2 and activated a feature called "MirrorLink".  What I did is to scan the ECUs and the software found one described as **[Navigation] MFD v4.3**. In that one I found the parameter for MirrorLink and changed it.  A Reset was required and for that I used the first option in 11-Reset.  Affter that, a new option is available in "@Services" option in the RLink 1 screen.

[image]

The problem is that I wanted to activate a feature called **"Android Auto"**.  If manually select the ECU from the top list, it can be found in **[Navigation] MFD v4.6** (Notice that the otehr is 4.3).  Followed a similar procedure than before but I receive an error message when trying to activate "Android Auto":

**WRONG RESPONSE CFC0 Unknown**

[image]

So, not very encouraging.  I want to believe there is a way of enabling the **"Android Auto"** feature in this car and RLink 1 system but I am not an expert on CAN and ECUs.

Is it possible to enable this feature included in **[Navigation] MFD v4.6** ?

Thanks.

enandrei, 2017-09-13:
Hi, 

Our problems seems to be diverging. 

I am trying to do the same thing , on a megane 3 2014 with Rlink 1.  But i can not activate the MirroLink option in MFD 4.3 .

I just tried changing the baud rate to 250000 in the xml , with the same problem. 

Log is here : https://pastebin.com/3ryHNYbM

BaumInventions, 2017-09-13:
To activate all features you have to update RLink to the newest Firmware with the SD Card an your PC.
After the System is Up to date you can actiate all features of "MFD 4.6".

I used MFD 4.6 on a 2017 Captur and a 2014 Kangoo to enable Android Auto.
On 2017 Captur it worked straight from the dealer (because it alreads has the newest Firmware).
On 2014 kangoo i had to Update the Firmware to enable features like MirrorLink.

valenciakva, 2017-09-13:
My problem is solved and I was able to activate Android Auto and MirrorLink in the Koleos 2016.  The suggestion from BaumInventions was the final solution. I updated the SD Card with the latest Renault available update, repeated the process and I was able to activate the options in [Navigation] MFD v4.6. A new reset was required.

Thanks to everyone who took the time to reply with suggestions to this thread.  I will keep the issue open until enandrei solves the issue reported.

enandrei, 2017-09-14:
After seeing valenciakva's success , i upgraded the Rlink version on my ecu as well. 

Now it doesn't recognize it either as MFD 4.3 or 4.6 , but after the upgrade i was able to communicate with the ECU properly , using the 4.6 xml files. 

I was also able to write the settings required to the ECU , and no longer received any errors ( using exactly the same setup ).

The conclusion here is that the problem was not in the ddt4all code, but rather a mismatch in xml / ecu firmware. And probably i would have had the same problem using an official cable with ddt2000.

Thank's everyone for your input !

closed 2017-09-14 by valenciakva

srayala, 2017-09-27:
Before all, thank you for your assistance. I am using ODBLink SX connected via USB to a Macbook Pro running Windows 10 via Parallels Desktop. After reading this issue, I tried to enable Android Auto, but with no success. I also read about update R-Link, which is supposed to be done, since there are no new updates both the R-Link Store site and the R-Link toolbox. Here is the screenshot:
[image]
I am not sure if I need to do an ECU reset, since it shows a legend that says "Force configuration to be taken". 
[image]
Attached next is the elm_ddt log.
[elm_ddt.zip](https://github.com/cedricp/ddt4all/files/1338742/elm_ddt.zip)

Again, thanks for your help and my apologies for reopen this thread.
Best regards.

BaumInventions, 2017-09-27:
First: You have to use a Adapter to read and write the Multimedia CAN instead of the vehicle can. when its working properly all ERROR messages are gone.
Second: After changing options you have to reset the display with a " RESET HARD"

srayala, 2017-09-27:
Many thanks Bauminventions. When you say an adapter, is it something like this?
[image]
I am guessing that I need to connect this adapter to the OBDLink and then to the vehicle can, am I correct?

valenciakva, 2017-09-27:
Yes, what BaumInventions suggest is to use a custom cable between the OBDII connector in your car and the ODBLink SX with the following wiring:

ODBLinkSX -Car ODB connector
6-13
14-12
4-4
5-5
16-16
Note: 4 and 5 are joined on ODBLinkSX side

BaumInventions, 2017-09-27:
The "adapter" is a very simple thing. its a cable you have to vonnect between OBD port and OBDlink. it just connects pins 4 and 16 (vehicle can) to 12 and 13 (multimedia can). You can build it yourself form a obd extender cable, a OBD Y-cable or just plugs and cables from the scrapyard... No rocket sciens involved ;)

srayala, 2017-09-27:
I see what you mean. Unfortunately, I do not have any extender cable, nor spare plugs and cables lol. So, I think the best option is to buy the adapter I posted above.

Many thanks both valenciakva and Bauminventions.

BaumInventions, 2017-09-27:
The thing from the picture above looks not like a adapter wich is fitting...
Your adapter/coverter should have a male OBD plug on one side and a Female OBD plug on the other side...

srayala, 2017-09-27:
You are absolutely right! I will call the store where I bought the OBDLink and see if they can help me on this.

BaumInventions, 2017-09-27:
I could custom build a cable for you... But i think the shipping cost around the world is some kind of high...

valenciakva, 2017-09-27:
If you are brave, you could do something like this.
[image]

srayala, 2017-09-27:
You bet! I am in Mexico. I just called the guys who sold me the dongle and told me they have indeed multimedia adapters. Lets hope it will do the job and if not, maybe they can build the cable.

Thanks a lot!

srayala, 2017-09-27:
I think I am not valenciakva! lol

valenciakva, 2017-09-27:
Fuerza México! Saludos desde Colombia

srayala, 2017-09-27:
¡Gracias hermano!

jperezruiz, 2018-01-30:
@valenciakva, I have similar error, the ELM response is null and I have the latest version for R-Link 11.342
Did you make any additional change to the ddt4 config?

valenciakva, 2018-01-30:
No, my issue was fixed by updating the SD Card.  I suggest to make sure you have good communication with the car by testing without the custom cable first.  
I created a guide in Spanish: rlink.valenciasanchez.com  and here is one more in English: https://www.gps-rlink.com

jperezruiz, 2018-02-02:
@valenciakva thankyou, I'm following your guide :).
With the usb elm the "Elm benchmark" fails, I tried to update the drivers but it didn't work, so I bought a bluetooth one. 
I opened a different issue #32 
Thank you again

groteskzp, 2018-05-12:
Hi. After I changed in MFD 4.6 rs monitor slot 1 and slot 2 parameters from other car, and changed in bic scripts from 6 to 14. I can't connect to MFD and BIC. Anybody have experience with it?

shrlnm, 2018-07-10:
@groteskzp Hello, could you please share the dump from working D70F3628? It may help to many people who got the same trouble.

alexpalcau, 2018-07-20:
Can anyone explain how to activate de RS Monitor to r-link 1? i found „rs display configuration” but i cannot figure where can I find the parameters...

mishu28nmv, 2019-03-08:
> Hi. After I changed in MFD 4.6 rs monitor slot 1 and slot 2 parameters from other car, and changed in bic scripts from 6 to 14. I can't connect to MFD and BIC. Anybody have experience with it?
> Upd: I raplace BIC from other car and MFD work. I think mcu data damaged. I will now try read from working D70F3628
> Upd2: dumps exchanged and all work

I want to activate the RS Display on a Laguna without changing the BIC. Will this work?

m0xz0r, 2019-05-13:
RS Monitor cannot be enabled by just filling the 2 slots via ddt4all. A slight modification of the contents of an 8K EEPROM inside the R-link displayunit by means of an EEPROM programmer is required. I have successfully performed the hack on my 2017 Clio 120 TCe. However, even though all functions and screens can now be invoked, some of the gauges and measurements are only partially working. My guess is that the BIC is not performing all the necessary translations between the car's main CANbus and the multimedia CANbus. Any help or info on this matter is highly appreciated!

alexpalcau, 2019-05-13:
I managed to activate the rs monitor via eeprom. I attach a link with the result: https://youtu.be/OkrNv4tAD40
I am also interested in the above matter but I think it is a problem with the sensors communication.

m0xz0r, 2019-05-13:
My RS Monitor behaves exactly like yours.  Let's hope that somebody can shed a light on how to enable the inactive gauges and missing measurements. I am convinced that all the necessary data is present on the main CANbus but the BIC is not relaying them to R-link.

luissilva1971, 2020-10-30:
Hi, everyone.

I am trying to solve an issue... change the orginal radio of my megane 3 grandtourer with tomtom but no bluetooth (beleive R1_08 radio) to an original radio from renault but with bluetooth function (R2_08) i bought from a parts store of a scrap yard. The problem is that the radio came from a dismantled car with standard display, not tomtom display. Could you help me in identifying the correct localion of XML /ECU where i have to change radio config? And also the correct xml / ECU from tomtom navigation panel (no r-link, simply tomtom (non live version) where i can configure the type of radio (it shoulb be R1_08, and i have to change to R2_08 with bluetooth, i think). Thank you in advance for all help you can give.