cedricp/ddt4all #7: Font alignment??

hoonythink, 2017-09-12:
hi
first, i thanks for you.
and I have a request from you.
ddt4all does not support font alignment.
So the menu is messy.
I want to sort the font right.
Can font alignment be supported?

- ddt4all screen
[image]

- RenoLink screen
[image]

and... I wish I had the ability to select xml files.

thanks.

cedricp, 2017-09-13:
Which XML file is it ?

hoonythink, 2017-09-13:
thanks for your answer.

i tested "RadNav_v2.3_20170726T141106.XML"

===================================
피큐브(주)

조정훈

H.P : 010 - 3153 - 2012
E-MAIL : hoonythink83@gmail.com
===================================

hoonythink, 2017-09-13:
attaching XML file.

it is some edited for see menu text.

===================================
피큐브(주)

조정훈

H.P : 010 - 3153 - 2012
E-MAIL : hoonythink83@gmail.com
===================================

cedricp, 2017-09-13:
Fixed

gordon6, 2017-09-13:
Thanx, Hoony and Cedric...
It goes enhanced..!

closed 2017-09-13 by cedricp

hoonythink, 2017-09-13:
thanks ^^

===================================
피큐브(주)

조정훈

H.P : 010 - 3153 - 2012
E-MAIL : hoonythink83@gmail.com
===================================