cedricp/ddt4all #8: Testing the program and not work

tameratf, 2017-09-14:
I try use the master version and it is not work , i see error when I refresh one shoot and the Changes not applied .
And I try another version  the program freeze when I select the ECU .
I need make some changes on my car
I use windows 10 .

cedricp, 2017-09-14:
Please be serious, and give us information about your configuration : 
* OBD Cable type
* Vehicle your are trying to connect
* Log files
If you see ERROR in the parameters, you are probably facing a bad (cheap) ELM dongle, try an ELM benchmark and report

tameratf, 2017-09-14:
>  ELM 327 usb v1.5
> Renault duster 2015
>  log files 
[logs.zip](https://github.com/cedricp/ddt4all/files/1304595/logs.zip)

cedricp, 2017-09-15:
Try with latest commit

hoonythink, 2017-09-15:
thanks

===================================
피큐브(주)

조정훈

H.P : 010 - 3153 - 2012
E-MAIL : hoonythink83@gmail.com
===================================

hoonythink, 2017-09-15:
the same.
it still not working.

have a nice weekend.

closed 2017-09-16 by cedricp

tameratf, 2017-09-16:
The last version of master not read elm 327 usb .
i use Windows 10 .

commit 347a2c3f (2017-09-16): Fix #8 and #9 init_can line accidentally removed