cedricp/ddt4all #10: Can't read elm 327 usb

tameratf, 2017-09-17:
The last version of master can't read elm 327 usb .
i use Windows 10 .

cedricp, 2017-09-17:
Provide log file for analyse.Tested successfully on my car yesterday
Any error message in terminal ?

tameratf, 2017-09-17:
[image]

cedricp, 2017-09-17:
No port found, have you tried with another application ?
Try to disable antivirus too.

tameratf, 2017-09-17:
i tried with another version of master and the port found .
[image]

cedricp, 2017-09-18:
My bad, I merged a part of the winXP branch into master, reverted. Can you check ?

tameratf, 2017-09-18:
I tried the last version and it is work .

cedricp, 2017-09-18:
Great, so closing

closed 2017-09-18 by cedricp