cedricp/ddt4all #11: ELM 327 cable can't load data...

hoonythink, 2017-09-26:
last DDT4ALL version is perfect at OBDLINK SX cable.

but ELM327 cable have one problem.

"USA_Bosch_Gen5_V7.6_20150423T151842.xml " can't read all data.
 
check please.

thanks always.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

cedricp, 2017-09-26:
Have you tried to increase CAN timeout value to 1000 ms ? (in the status bar)
If it doesn't work, I'll need logs

hoonythink, 2017-09-26:
it works.
thanks. ^^

closed 2017-09-26 by hoonythink

cedricp, 2017-09-26:
Great :+1: