cedricp/ddt4all #12: WRONG RESPONSE MULTILINE CFC0 and TIMEOUT

mvmlt, 2017-10-01:
Thanks in advance for your help. I am using a ODBLink SX connected via USB to a Windows 10 laptop. My car is a MEGANE III 2014. ECus Database is the last (08/2017). Chinese ELM clone running 1.5 (tested with ELM identifier working with commands up to 1.4). Then pins are remapped (6-13/14-12).  CAN timeout value is set to 1000 ms. I have the last DDT4all version.

When I start the ECU and select the navigation screen, I have errors messages on the screen like that 
<hour>;747;3E01;WRONG RESPONSE MULTILINE CFC0 
<hour>;747;226C13;WRONG RESPONSE MULTILINE CFC0 
<hour>;747;226C0F;WRONG RESPONSE MULTILINE CFC0 

Attached file, elm.ddt.
Thank in advance.

mvmlt, 2017-10-01:
[elm_ddt.zip](https://github.com/cedricp/ddt4all/files/1347060/elm_ddt.zip)

cedricp, 2017-10-01:
No communication, check your ELM or connection.

Turbox35, 2017-10-02:
Hello,
Exactly the same problem. How do you solve it? (tested 4 ELM)

mvmlt, 2017-10-02:
Merci Cedric.
Strange ...

I check the cable and it is ok (ELM <=> CAR ...4-4; 5-5; 16-16; 6-13; 14-12) 

I test ELM with EasyODBII (without adapter cable) and it is ok
[image]

Je suis sec! No Idea

cedricp, 2017-10-03:
Does it work on the vehicle bus ? Can you connect to the engine control module ?
Have you read the issue #6 ?

Turbox35, 2017-10-03:
I succeed (meg 3 rs 2014 rlink1):
I had yellow button and "no data" in red"

elm compatible: 95/95
timeout 1000
canbus (bluetooth) 500k
If it can help. Good luck

mvmlt, 2017-10-05:
Sorry Cedric, ut I do not understand your question !!
TurboX ...What is your elm ?
my elm compatible score : 76/95 ... Is it the reason of the problem ?

cedricp, 2017-10-05:
As tu essayé de te connecter à l'ECU moteur sur le bus CAN standard ? Il faut aussi vérifier que l'elm ait bien le circuit intégré PIC original, sinon ça marche pas.

Turbox35, 2017-10-06:
the working elm was a Chinese Bluetooth. I've tried with 3 other elm (1 had 75/96) but none was working. 
With the working elm,  the scan of ecu found 2 other ecu that means the connection was OK and after I manually go in nav ecu.

mvmlt, 2017-10-07:
My elm do not have the original PIC. I will buy another.
Merci pour vos conseils Cedric et turboX.

mvmlt, 2017-10-17:
Hello,
Bien vu Cedric  - Merci ! 
Le problème venait bien de l'ELM (version USB acheté sur Ebay)
Après remplacement (version BT sur Aliexp), tout est ok!

[image]

closed 2017-10-17 by mvmlt