cedricp/ddt4all #14: Connect success but unable to read and write

ronnietam, 2017-10-09:
Try to enable CarPlay support 
I followed the instruction:
Update to latest version
[image]

USB ELM327 with PIN 14 to 12 , PIN 6 to 13 
[image]
[image]
[image]
Plug and run ddt4all, select the correct COM3 and select Connected Mode
[image]
Switched to after sales mode, select the three new options and set to "present" , click Write
[image]

The log shows

Sending request :StartAfterSalesSession
Switching to session mode 10 C0
Sending request :DataWrite.ECU
Blocked ELM request :2E 6C 1C F8
ELM response : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sending request :StartDiagnosticSession
Switching to session mode 10 81

But after reset (i only can reset by "Set to factory default setting") i can't see the new Phone Connect Icon in Rlink

Is the above log show a successful or fail update? BTW I click Read All in ddt4all but it only show no data. Does anyone here can help? Thank you

cedricp, 2017-10-09:
**Blocked ELM request** :2E 6C 1C F8
It seems that you are not in expert mode, so the request is for your security blocked...
There are other closed issues about this problem, please read them.

cedricp, 2017-10-09:
You may need to increase CAN timeout also

ronnietam, 2017-10-10:
Thank you your suggestion. I had screened the closed issues and tried increase the CAN timeout to 1000. However, it still not work
The message "Blocked ELM request" still there and I'm sure i have turn on the Expert mode with "After Sales" option. 

[image]

I'm using China ELM327 however, I only can connect with speed 38400, all other speed will show "No ELM is available" or "Window Error(5, 'Access is denied.')

I noticed that you had released a per-released version v1.1, i will have a try if no other way.
I also suspect the Antivirus program block ELM request, it is possible?

cedricp, 2017-10-10:
Do not attempt a write operation while there is "NO DATA" in the read parameters, it is normal to use 38400 bauds to connect ELM. Check if there's a PIC18F25K80 chip in the ELM.
Try to benchmark your ELM too.
And no, antivirus cannot block ELM request

ronnietam, 2017-10-10:
here is the Benchmark 
v1.0 ATZ [OK] v1.0 ATE1 [OK] v1.0 ATL0 [OK] v1.0 ATI [OK] v1.0 AT@1 [OK] v1.0 ATAL [OK] v1.0 ATBD [OK] v1.0 ATBI [OK] v1.0 ATCAF0 [OK] v1.0 ATCFC1 [OK] v1.0 ATCP 01 [OK] v1.0 ATCS [OK] v1.0 ATCV 1250 [OK] v1.0 ATD [OK] v1.0 ATDP [OK] v1.0 ATDPN [OK] v1.0 ATH0 [OK] v1.0 ATI [OK] v1.0 ATIB 10 [OK] v1.0 ATIB 96 [OK] v1.0 ATL1 [OK] v1.0 ATM0 [OK] v1.0 ATMA [OK/TIMEOUT] v1.0 ATMR 01 [OK/TIMEOUT] v1.0 ATMT 01 [OK/TIMEOUT] v1.0 ATNL [OK/TIMEOUT] v1.0 ATPC [OK] v1.0 ATR1 [OK] v1.0 ATRV [OK] v1.0 ATSH 001122 [OK] v1.0 ATSH 012 [OK] v1.0 ATSP A6 [OK] v1.0 ATSP 6 [OK] v1.0 ATCM 123 [OK] v1.0 ATCM 12345678 [OK] v1.0 ATCF 123 [OK] v1.0 ATCF 12345678 [OK] v1.0 ATST FF [OK] v1.0 ATSW 96 [OK] v1.0 ATSW 34 [OK] v1.0 ATTP A6 [OK] v1.0 ATTP 6 [OK] v1.0 ATWM 817AF13E [OK] v1.0 ATWS [OK] v1.1 ATFC SD 300000 [OK] v1.1 ATFC SH 012 [OK] v1.1 ATFC SH 00112233 [OK] v1.1 ATFC SM 1 [OK] v1.1 ATPP FF OFF [OK] v1.1 ATPP FF ON [OK] v1.1 ATPPS [OK] v1.2 ATAR [OK] v1.2 ATAT1 [OK] v1.2 ATDM1 [OK/TIMEOUT] v1.2 ATIFR H [OK/TIMEOUT] v1.2 ATIFR0 [OK] v1.2 ATIIA 01 [OK] v1.2 ATKW0 [OK] v1.2 ATMP 0123 [OK/TIMEOUT] v1.2 ATMP 0123 4 [OK/TIMEOUT] v1.2 ATMP 012345 [OK/TIMEOUT] v1.2 ATMP 012345 6 [OK/TIMEOUT] v1.2 ATSR 01 [OK] v1.3 ATCRA 012 [OK] v1.3 ATCRA 01234567 [OK] v1.3 ATD0 [OK] v1.3 ATFE [OK] v1.3 ATJE [OK] v1.3 ATJS [OK] v1.3 ATKW [OK] v1.3 ATRA 01 [OK] v1.3 ATS1 [OK] v1.3 ATSP 00 [OK] v1.3 ATV0 [OK] v1.4 ATCEA [OK] v1.4 ATCEA 01 [OK] v1.4 ATCV 0000 [OK] v1.4 ATIB 48 [OK] v1.4 ATIGN [OK] v1.4 ATPB 01 23 [OK] v1.4 ATRD [OK] v1.4 ATSD 01 [OK] v1.4 ATSI [OK] v1.4 ATFI [OK] v1.4 ATSS [OK] v1.4 ATTA 12 [OK] v1.4 ATCSM1 [FAIL] v1.4 ATJHF1 [FAIL] v1.4 ATJTM1 [FAIL] v1.4 ATJTM5 [FAIL] v1.4b ATCRA [FAIL] v2.0 ATAMC [FAIL] v2.0 ATAMT 20 [FAIL] v2.1 ATCTM1 [FAIL] v2.1 ATCTM5 [FAIL] Result: 86 succeeded from 95 ELM Max version:v1.4  

Is it good enough for the ECU write ? Thank you.

cedricp, 2017-10-10:
It should be enough. But it doesn't mean your ELM has no problem.
Try to connect to other ECU (on the vehicle CAN) to check if it's working correctly.

cedricp, 2017-10-10:
Can you provide log files for analyse ?

ronnietam, 2017-10-10:
Here is the log file. Sorry the data may be mess due to i try write multiple times. Thank you for your help

[elm_ddt.zip](https://github.com/cedricp/ddt4all/files/1370992/elm_ddt.zip)
[ecu_ddt.zip](https://github.com/cedricp/ddt4all/files/1370993/ecu_ddt.zip)

cedricp, 2017-10-10:
You have a lot of CAN ERROR reported by ELM meaning (ELM doc) :  
**CAN ERROR**
The CAN system had difficulty initializing, sending,
or receiving. Often this is simply from not being
connected to a CAN system when you attempt to send
a message, but it may be because you have set the
system to an incorrect protocol, **or to a baud rate** that
does not match the actual data rate. As with BUS
ERRORs, **the CAN ERROR might also be the result of
a wiring problem**, so if this is the first time using your
ELM327 circuit, **review all of your CAN interface
circuitry before proceeding**.

ronnietam, 2017-10-12:
Would like to appeicate your help. I can now connect the car and able to update the values. I just had wrong wiring before. Thank you and please feel free to close this ticket.

cedricp, 2017-10-12:
:+1: 
Great, so, closing ticket

closed 2017-10-12 by cedricp