cedricp/ddt4all #16: Connect to ELM327 wifi but freezes when scanning (CAN or CAN new)

LemonLyman91, 2017-10-17:
Im using Windows 10 and when trying to scan either with CAN or CAN new system the program freezes (not responding). What might be happening? I tried to use the ELM327 with a mobile app and it works.

Log:

############################################################
#connect to: SCAN Addr:1E
############################################################
>[13:40:12.424]AT SH 748
<[0.0]AT SH 748
OK

>
>[13:40:12.534]AT CRA 768
<[0.0]AT CRA 768
OK

>
>[13:40:12.643]AT FC SH 748
<[0.0]AT FC SH 748
?

>
>[13:40:12.768]AT FC SD 30 00 00
<[0.0]AT FC SD 30 00 00
OK

>
>[13:40:14.258]AT FC SM 1
<[0.0]AT FC SM 1
?

>
>[13:40:14.371]AT SP 6
<[0.0]AT SP 6
OK

>
>[13:40:14.481]AT ST FA
<[0.0]AT ST FA
OK

>
>[13:40:14.592]0210C0
>[14:08:07.404]ATZ
<[0.0]ATZ

ELM327 v1.5

>
############################################################
#[10/17/17 14:08:17.570] Init CAN
############################################################
>[14:08:17.570]AT WS
<[0.0]AT WS

ELM327 v1.5

>
>[14:08:17.899]AT E1
<[0.0]AT E1
OK

>
>[14:08:18.008]AT S0
<[0.0]AT S0
OK

>
>[14:08:18.117]AT H0
<[0.0]AT H0
OK

>
>[14:08:18.227]AT L0
<[0.0]AT L0
OK

>
>[14:08:18.336]AT AL
<[0.0]AT AL
OK

>
>[14:08:18.445]AT CAF0
<[0.0]AT CAF0
OK

>
############################################################
#[10/17/17 14:08:18.555] Init CAN
############################################################
>[14:08:18.555]AT WS
<[0.0]AT WS

ELM327 v1.5

>
>[14:08:18.664]AT E1
<[0.0]AT E1
OK

>
>[14:08:18.774]AT S0
<[0.0]AT S0
OK

>
>[14:08:18.883]AT H0
<[0.0]AT H0
OK

>
>[14:08:19.008]AT L0
<[0.0]AT L0
OK

>
>[14:08:19.102]AT AL
<[0.0]AT AL
OK

>
>[14:08:19.211]AT CAF0
<[0.0]AT CAF0
OK

>
############################################################
#connect to: SCAN Addr:1E
############################################################
>[14:08:19.321]AT SH 748
<[0.0]AT SH 748
OK

>
>[14:08:19.446]AT CRA 768
<[0.0]AT CRA 768
OK

>
>[14:08:19.555]AT FC SH 748
<[0.0]AT FC SH 748
?

>
>[14:08:19.665]AT FC SD 30 00 00
<[0.0]AT FC SD 30 00 00
OK

>
>[14:08:19.774]AT FC SM 1
<[0.0]AT FC SM 1
?

>
>[14:08:19.883]AT SP 6
<[0.0]AT SP 6
OK

>
>[14:08:19.993]0210C0
>[14:08:52.431]ATZ
<[0.0]ATZ

ELM327 v1.5

>
############################################################
#[10/17/17 14:09:20.576] Init CAN
############################################################
>[14:09:20.576]AT WS
<[0.0]AT WS

ELM327 v1.5

>
>[14:09:20.686]AT E1
<[0.0]AT E1
OK

>
>[14:09:20.795]AT S0
<[0.0]AT S0
OK

>
>[14:09:20.904]AT H0
<[0.0]AT H0
OK

>
>[14:09:21.014]AT L0
<[0.0]AT L0
OK

>
>[14:09:21.123]AT AL
<[0.0]AT AL
OK

>
>[14:09:21.233]AT CAF0
<[0.0]AT CAF0
OK

>
############################################################
#connect to: SCAN Addr:1E
############################################################
>[14:09:21.342]AT SH 748
<[0.0]AT SH 748
OK

>
>[14:09:21.451]AT CRA 768
<[0.0]AT CRA 768
OK

>
>[14:09:21.561]AT FC SH 748
<[0.0]AT FC SH 748
?

>
>[14:09:21.670]AT FC SD 30 00 00
<[0.0]AT FC SD 30 00 00
OK

>
>[14:09:21.779]AT FC SM 1
<[0.016]AT FC SM 1
?

>
>[14:09:21.904]AT SP 6
<[0.0]AT SP 6
OK

>
>[14:09:22.014]021003

cedricp, 2017-10-17:
I guess you're using a bluetooth adapter, right ?
There are no reason for the application to hang. Your adapter seems to not support some important commands : 
**[14:09:21.779]AT FC SM 1
<[0.016]AT FC SM 1
?**

**[14:09:21.561]AT FC SH 748
<[0.0]AT FC SH 748
?**

Try to benchmark it at first (if it's detected as a core arm one, you can trash it)

LemonLyman91, 2017-10-17:
Its a wifi adapter. This one: https://www.amazon.es/gp/product/B01HYX7HHE/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

Which adapter will work ok? Thanks!

cedricp, 2017-10-17:
It's a WiFi one, Ok, I solved the issue #15 recently, have you downloaded the latest version of the source code ?

cedricp, 2017-10-17:
About ELM327, you need the real one with PIC18D25K80 inside, else it won't work.
The benchmark test is on the first screen, the button on the right. Please do it before all.

mvmlt, 2017-10-17:
Hello, see issue #12. 
I use an USB ELM which is the same type and design of your wifi ELM. It won't work correctly. 
After using another ELM with PIC18xxx all is ok.

LemonLyman91, 2017-10-17:
What about a konnwei KW902 Mini Bluetooth? Should that work ok? I cant find a real ELM327 in Spain :S

cedricp, 2017-10-17:
I have a KW902 and it works very well.

cedricp, 2017-10-20:
I'd need your log files to analyse it. Are you connecting on the M-CAN BUS ?

LemonLyman91, 2017-10-20:
Thanks very much. There is no need anymore. I was able to change the parameters by locating the ECU on the database (the data was read and I could write in after sales mode). Thank you! You can close the ticket.

cedricp, 2017-10-20:
Awesome ! Is it possible to get your log files, though ? I'd like to understand why only 2 ECUs were found on your vehicle. Thanks

closed 2017-10-20 by cedricp

LemonLyman91, 2017-10-20:
Sure! Here you go: https://www.dropbox.com/s/pp4cljj44blx8vf/logs-Megane-IV-ELM327%201.4.rar?dl=0