cedricp/ddt4all #17: Where to put the ecus?

octav90st, 2017-11-11:
Please tell me exactly where to put the ecus. I have installed the program and everytime I copy them it seend that the software starts with 0 loaded ecus.

gordon6, 2017-11-12:
Dear,
You need to put at C:\ECUs or C:\Program files (x86)\DDT4All\ECUs.
Then the problem should prove...

octav90st, 2017-11-12:
thanks!

C:\Program files (x86)\DDT4All\ECUs is the right path and now it works to select zip database and in the final it says: permission denied ecu,zip

annoying thing. still can't use this program...

cedricp, 2017-11-12:
zipping database is not mandatory, you can work without this step. I'll try to fix the permission issue, though.

cedricp, 2017-11-28:
I added a file dialog to ask where to put the final archive, that should be fine now

closed 2017-12-21 by cedricp