cedricp/ddt4all #18: A-IVI parameters

Forscenic, 2017-11-19:
Hi Cedric.
First of all I'm thankful for your great job and I'm glad if you (or someone else) will help me on this problem I've found.
Then, I'm currently working on my Scenic IV to check if the digital tuner was already built on the multimedia device (R-Link2) and, of course, activate it to turn on the DAB parameter (named DR = Digital Radio on the Renault operating manual).
I use an ELM 327 V.1.5 with PIC18F25K80 wifi connected on Win10 and the switch to set Low/High speed bus on it. Everything seems works ok (I can read/write the RadioNav ECU on 500K and the MIU interface set on 250K) but, when I try to read the A-IVI ecu (doesn't matter if on a Aug. DB or an Apr. DB and I tried both speeds), the response is still NO DATA (blank value in the log window). I see that the 8 pins plugged on the car's OBD port are 4,5,6,12,13,14,15,16 and they are quite different than my ELM (2,4,5,6,10,14,15,16 and the switch between 6/13 and 14/12).
However, I see that DAB parameters are already checked in RadioNav ecu but, when I try to enable it, there's no success at all.
Do you have any solution ?
Thank you in advice.

BaumInventions, 2017-11-19:
From my experience with smaller platforms (like Clio iv/ Captur / Trafic) there is no DAB Tuner installed when you dont order it with the car. The whole Hardware Tuner is missing inside the Amplifire module, and the antenna with all cables is missing.

On Clio iv woth R-Link 1 and Evolution you have to replace the Amplifire Module and fit a DAB antenna to make it work...

You can tell from the Antenna Plugs on your ECU if somethin is not connected.
Yellow and White are the plugs for normal Radio and diversity antenna. Blue is GPS and Brown is DAB.
If there is no Brown plug on your ECU it is not supported.

Forscenic, 2017-11-20:
You're right and I completely agree but, as you may know, the public manual of the device R-Link2 (http://gb.e-guide.renault.com/eng/R-LINK2/LISTENING-RADIO and it's available in multiple languages) explain all the functions of the device and doesn't refer at the model of the car you buy/order. That's the reason why, after looking inside these features, I saw many functions were not "enabled" (AndroidAuto and CarPlay are 2 examples) and, using this "miracle" called ddt4all, I was able to start a little reverse engineering on the ECUs and activate all without problems, even functions and params that are not phisically presents (like the CD Drive). That's the reason why I think the issue is maybe inside the xml and/or in the OBD pinout in the car. However, if you have a link to send where is a "disassemble procedure", I can try to investigate on the cables (and colors), even if I'm not so experience.

Forscenic, 2017-11-20:
Apologize for my wrong check on the issue

closed 2018-01-15 by cedricp

Banana009, 2018-02-05:
I have the same problem. I am not able to read/write A-IVI_v0.9. I would like to activate LKA (I have park assist) . I am not successful.

lucamagilla, 2018-03-21:
Same problem also I can not read A-IVI 09 anyone who has any idea? I have tried all the modifications on the OBD but nothing help