cedricp/ddt4all #19: Megane 4 Support

Giuse1977, 2017-11-28:
Hi cedricp,
will ddt4all work with Renault Megane 4 in near future?

regards

Giuseppe

cedricp, 2017-11-29:
It should already work. What is the problem ?

Giuse1977, 2017-11-29:
Some users report that is possibile to read but not to write on it. I still had not time to try by myself, anyway..

cedricp, 2017-11-29:
Well, it depends on the ECU you want to write on, I guess you're talking about MediaNav functions, right ?
Check older issues, there are a lot of information about it.

Giuse1977, 2017-11-29:
Actually the head unit is R-Link2. Owners of other Renault cars with same head unit reported success, but so far no one on Megane 4 did. I repeat, I had no time to try by myself so far, unluckily 

Inviato da iPhone

MeganeIVgrandcoupe, 2017-12-22:
It works on Megane IV!

cedricp, 2017-12-22:
Perfect, so I close the issue

closed 2017-12-22 by cedricp