cedricp/ddt4all #20: Support for KKL 409.1 CH340

Inf3rnus187, 2017-11-29:
Salut et juste avant ma demande good job c'est vraiment top ce que t'as fait ;)  
Peu après que tu l'ai sorti j'avais commencer à faire un installateur mais étant pris à coté j'avais pas eu le temps de m'y remettre et je voie que t'as bien avancer ! 

Mon Elm est en puce ch340 bien reconnu et mon cable kkl aussi je me demandais ou faire deux trois modif afin de pouvoir le choisir au début dans la sélection du port ? 

En espérant que tu passe par là merci à toi ;)

Inf3rnus187, 2017-11-29:
Ben ce que je capte pas c'est justement la puce de mon elm 1.5a est la même....

Après oui c'est sur que  il est juste fait pour le protocole kwp2000 ce cable. 

Mais justement ma laguna 2 phase 2 de 2006 l'uch est en kwp et via ddt2000 et ce fameux cable ou même ddt4all et le cable ELM je n'arrive à rien dessus non plus, j'espérais que je pourrais me retourner avec ton soft est mon cable pour faire mes test. 

Juste au cas ou mon cable ELM est bien censer lire écrire sur le KWP2000 Fast init  ?

cedricp, 2017-11-29:
Oui la lecture/ecriture doit fonctionner, apres, faut verifier que les PINs de l'adaptateur vont au bon endroit... Pour ma part j'ai jamais pu tester, mais on a fait des essais avec quelqu'un qui m'a reporte que son dacia arrivait a lire/ecrire en KWP2000 FastInit.

closed 2017-12-21 by cedricp

justForHobby, 2021-11-23:
Hi, 
is there instructions in English for fix BUS INIT : BUS ERROR? 

I'm using Proscan automotive DDT4ALL adapter.
Issue is with Kangoo AC4 ph2-crash data reset