cedricp/ddt4all #21: DDT4ALL not working some ecu configuration

hoonythink, 2017-12-04:
hi, long time no see. 
i have Koleos II and use OBDLINK SX (USB) for ddt4all.
"ddt4all" works very well and satisfactory.

However, recently one problem has been found.

"ESC___1540_HFE_XZG_Version_2.3_20170620T142408.xml"

Click the "Write" button has no effect.
There is no error message.

[image]

but, it works in a program called "RenOLink".

and Some users ask me. 
When they click the "Write button", the setting value("Read" side value) changes.
However, the setting values are initialized when reconnected.
do you know about this?

I always thank you for your hard work.

cedricp, 2017-12-04:
It seems that the ELM doesn't wait enough to get the ECU response, try to increase **CAN Timeout** value to 600-800ms in the bottom of the screen and then send the request.

hoonythink, 2017-12-10:
sorry, i'm late.
I tried and succeeded as you told me. 
thanks.

and....
Could you answer me the second question?

cedricp, 2017-12-12:
I have no idea about the second point, sorry. Have you tried a soft reset ?

hoonythink, 2017-12-12:
my car and obdii cable is working well.
someone told me the problem. 
I thought the problem was due to Bluetooth OBDII model.
But people who use the same cable as me are also experiencing the same problem.

Thank you for your attention.

closed 2017-12-12 by hoonythink

Ylmz038, 2023-10-17:
Merhaba ddt4all ile ABS yada ESP ayarlarını bozdum galiba ABS Işığı aracımda sürekli yanıyor yardım edin lütfen 

Uploading VID_20231003_171717.mp4…

AndyIreland, 2023-10-17:
> Hello, I think I messed up the ABS or ESP settings with ddt4all. The ABS light is constantly on in my vehicle, please help.
> 
> Uploading VID_20231003_171717.mp4…

hi, either enable ABS/ESP back again or extinguish the light by going into your 'clustre' settings and choose 'no ABS' and/or ' no eps'