cedricp/ddt4all #22: Méthode gethexvalue de la classe ECU

FredLeudon, 2017-12-08:
Hello,

Je me pose une petite question concernant la méthode getHexValue

Dans le code, on sépare les traitements entre des réponses longues (big) et courtes (little)

Ors, si on prends la trame reçue en hexa, qu’on la convertit en chaîne représentant le binaire, avec les données firstbit, etc. il est plus facile de retrouver ses petits en extrayant la partie de cette chaîne binaire puis en la convertissant en hexadécimal.

En me relisant, je n’ai pas l’impression d’être très clair. 

On peut en discuter par mail is tu veux ;-)

cedricp, 2017-12-08:
En fait, la separation vient du fait que l'on traite l'info en **little** ou **big endian**, ce qui n'a pas de rapport avec la taille de la reponse mais plutot de l'ordre de stockage des infomations (on lit les bits depuis la droite ou depuis la gauche). J'espere que ca eclaircira ta comprehension.

FredLeudon, 2017-12-08:
Merci, je me coucherai moins bête ;-)

Frédéric RICHARD

closed 2017-12-21 by cedricp