cedricp/ddt4all #25: 9600 b/s  ?

fgs64, 2017-12-21:
Hi all,
first of all, bravo Cédric  ! 
One question :  I can't make my old usb elm327 v1.2 work (with W10), it works well with other software. Can you tell me where is the best place to add 9600 b/s speed please ? 
(Oui bon, j'ai aussi commandé la prise kivabien et la rallonge, c'est d'abord pour 'apprendre à lire' le python in the text.)
Thank's
Francis

cedricp, 2017-12-21:
Hi,
Thank you, you can try to modify these lines in **ddt4all.py** (around line 870) : 
```
class portChooser(gui.QDialog):
    def __init__(self):
        portSpeeds = [9600, 38400, 57600, 115200, 230400, 500000]
```
But I'm quite sure it won't work. The application needs some commands present in version 1.3 and 1.4.
Let me know if it works, though

fgs64, 2017-12-22:
Hé hé ! The oooold ELM327 made it, it coultd hear and speak sloooowly, and it didn't need to have new commands: Android Auto activated on RLink 1 on Laguna III 2015 :-)))

Closed, guy happy.

[image]

fgs64, 2017-12-22:
Thanks god Cédric !

cedricp, 2017-12-22:
Good news :)

closed 2017-12-22 by fgs64