cedricp/ddt4all #28: need help for Megane IV

igno77, 2018-01-02:
Hi Cedric,
first of all I want to say you thanks a lot for this fantastic software, it's very useful and complete! I use your software with a wifi obd with the PIC18F25K80 on my Megane IV (1.6 dci Bose) and works quite well.
I've seen that I've to reverse 2 pins in the way to better comunicate with the ecu, right? I've also to remove a resistor (120ohm) from the standard pcb? Can you explain me better when I've to switch from low/high signal and for what kind of ecus?
On the top of the software interface, when you click on the Einstein icon, there is a line where you can select:
-After sales default 10C0
-Start Diagnostic session default 1001
-Start diagnostic session reprog 1002
-start diagnostic session extended 1003
If I change from the 1001 to 1002 the display that is under modify (tachometer, rlink, etc) is switching off and I can't use the ddt4all anymore. After few seconds the device make a reset and wake up again. Can you explain me the meaning of these choose and when I've to use it?
Sorry for all these question but I love this software and I love modify my car!! Thanks again to you.

cedricp, 2018-01-02:
Hi,
Reversing the 2 pins and removing the 120 Ohms termination resistor is not normal, you shouldn't do that, but perhaps there's a design issue with your adapter. But if it's working this way...
You also should avoid to manually change diagnostic sessions (it's already done automatically by the software), it's just for experimentation. In you case, you ask your ECU to enter a **reprog** session, and it waits for incoming reprog data (hence the problem).

igno77, 2018-01-02:
Ok, now is clear....I will never touch it again!
About the 2 pins reversing, I've seen that if you want to comunicate with the Renault multimedia, you have to send the signal from pin 6 to 13 and 14 to 12, isn't it?? My obd2 interface it's a "common interface" with the right chip but I've a lot fo unreaded ecus... Could be the timeout to short?
There is also a way to check what's really acrtivated after the reset? If I click on turnkey off/on you will store the modification in the ddt4all software and when you open again you will see the modification enabled in anyway

cedricp, 2018-01-02:
Yes, on most Renault cars, the M-CAN (mutimedia) pins are 13 and 12. Others (ECM, Airbags, ESP, etc.) are on the normal bus.
If you get some unread data, you can try to increase timeout value in the bottom of the screen, it often helps. I don't have a recent car, so I can't test on my side, but you can record a video of your session and I'll try to help.

igno77, 2018-01-02:
Thanks a lot for your help, now I will buy an obd cable in the way to modify the pins (my obd interface is soldered directly to the pcb!) and I will try. This "famous" resistor is used for the comunication timeout and is placed in parallel from the pin 6 and 14 on the "virgin obd interface". If I will increase the timeout in the software (like 1000ms) it will be the same of the resistor removing? I will let you know 
Bye and thanks again

cedricp, 2018-01-02:
The resistor has nothing to do with timeout issue, it's just an impedance line adapter. My guess about that is that the resistor should not be there, because the bus already has termination resistors.

igno77, 2018-01-02:
Right, I will remove this resistor soon! When the cable for switching pins will arrive I will make a lot of tests and I'll let you know. Bye

igno77, 2018-01-04:
Just another question, when I see more ecu with the same name but different version (like Cluster_15-40 v3.30 and Cluster_15-40_v3.30.02) I've to use the last one? Or every cars have the proper ecus version?
When I try to scan the ecus with the button in the upper left side, the software hang on at 21% and it found only 5 ecus, it's normal?
Thanks

igno77, 2018-01-04:
Just another question, when I see more ecu with the same name but different version (like Cluster_15-40 v3.30 and Cluster_15-40_v3.30.02 or Radnav 2.2 and Radnav 2.4) I've to use the last one? Or every cars have the proper ecus version?
When I try to scan the ecus with the button in the upper left side, the software hang on at 21% and it found only 5 ecus, it's normal?
Thanks

cedricp, 2018-01-04:
You must (every time it's possible) use the detected (via scan) ECU file. About the hang, it's strange, do you have any error in the terminal view ?

cedricp, 2018-01-04:
Could you send me your log files in the **log** folder after a scan ?

igno77, 2018-01-04:
Can you send me an email where I can send you the log? I've tryed to use the "attach file" below but it doesn't work...
Here the python interface...
[img]https://ibb.co/foMCHG[/img]
If I click on elm benchmark nothing happen...

igno77, 2018-01-04:
Can you give me an email where I can send you the log file? I've tryed to use teh "attch the file" below but without success....
Here the Python interface:
[img] https://ibb.co/foMCHG [/img]
If I click on the elm benchmark nothing happen.... (obd2 wifi)

cedricp, 2018-01-05:
send it to cedricpaille@gmail.com

cedricp, 2018-01-05:
About the scan issue, could you try to modify the file **ecu.py** around line 1480 and replace : 

> soft_version = can_response.replace(' ', '')[6:38].decode('hex')

with : 
> soft_version = can_response.replace('FF', '').replace(' ', '')[6:38].decode('hex')

igno77, 2018-01-05:
I've sent my log by email and now I've downloaded your latest "master" update. I will overwrite all the files and today I will try on the car!

igno77, 2018-01-05:
Great!! Later I'll try and I'll let you know. Thanks a lot

closed 2018-01-15 by cedricp