cedricp/ddt4all #31: I connect with elm327 obd wifi but I can not reach the scan program interface

mrcfen, 2018-01-21:
hii cedricp
my car megane 4. elm327 obd2 I can not connect with wifi. program ddt4all with megane 4> Audio> RadNav v2.3 after selecting the program does not respond.

cedricp, 2018-01-22:
As I don't own a WiFi dongle, i'm currently not able to make test with it... I'll try to borrow one to make tests ASAP.

mrcfen, 2018-01-22:
thanks cedric ... i am waiting for your results

cedricp, 2018-01-30:
I tested a WiFi dongle this morning, but it's a bad Chinese clone, and indeed, there are a lot of bad behaviors. I think the program is not responsible. You better get a good USB dongle for that operation.
According to issue #15, wifi connection should work.

KarelSvo, 2018-04-02:
Problem solved?

Did you connect with Media Can or Vehicle Can?
The CLIP database says RadNav speaks on Pin 7 + 14 Vehicle Can.

Furtif, 2023-06-02:
for help or question uses https://github.com/cedricp/ddt4all/discussions please .

closed 2023-06-02 by Furtif