cedricp/ddt4all #32: Connection suddenly closed: Access denied

jperezruiz, 2018-02-02:
First of all, thanks for your hardwork.
I have a clio IV and I'm trying to enable Android Auto.
I'm using a bluethoot [ELM327](https://www.amazon.es/gp/product/B01BZPL70U/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) recommended by another user who executed the same process in another car.

I'm following the steps of the #6, the initial connection works but when I try to access to [Navigation] MDF 4.6 or 4.3 the ELM connection is lost with the error "5 access denied".
I'm also using the special wire:
ODBLinkSX -Car ODB connector
6-13
14-12
4-4
5-5
16-16

Some steps I tried:
- Enable/disable com ports in the device manager
- Disable windows defender
- Different bit rate
- 1000ms timeout
- Different ports (the elm provides com5 and com6)
- With and without the special wire 

The connection is lost when I click in the [Navigation] item, not before.

I also tried the "Test elm connection" from the first window, almost the half of the items shown "OK/Timeout".

Could you please bring me some light? :)
[Here the logs](https://drive.google.com/file/d/1adzJ6cyooImvYjuNtJkRm_1e73vpdw6B/view?usp=sharing), maybe could by helpful:
[image]

cedricp, 2018-02-06:
If you get "**Adapter with ARM incompatible**", you won't be able to connect any ECU of your car, I'm sorry to say that your adapter is a bloody Chinese clone.
It seems difficult to get a working adapter nowadays, unfortunately (I got 4 non working at home!).
If you really need a bluetooth one, try the KW902.

jperezruiz, 2018-02-06:
Thanks four your answer  @cedricp, I supposed that the problem was the connector so I bought a third one that shows in the description "original 25k80".
The connector arrived yesterday and I was able to connect and write the ecu, yai!
Thanks for your hard work :)
For anyone who may need it, the connector is Autool c1 V1.5 OBD. The price is around 13€

closed 2018-02-06 by jperezruiz

cedricp, 2018-02-06:
Thank you for the feedback :+1:

Sig2018, 2018-05-11:
Hello jperezruiz,

Is your RLINK 1 or RLINK 2?

I've found the program and some instructiosn to modify my navigation system but like others in this post I couldn't.

The program seems to work, conection with ELM327 is green but it can't comunnicate with ECU, After sending a command there is a blank field on ECU response. Scanning ECUS didn't find anythig.
I've tried with direct ODB2 and with modified cable to acces the Multimedia Can Bus but nothing.
I would appreciate those with Rlink 2 that were succesfully can help me with your experiences.
Thank you in advance.