cedricp/ddt4all #33: Bad ELM Response

radunre, 2018-02-06:
Hi there,

First of all, I would like to thank you so much for the idea and great work you have done for all Renault lovers.

I have started to figure out the ddt4all since 3 weeks and I have on my bench an Automatic Parking Brake unit from my Laguna 3/2010. I have connected it with an ELM Bluetooth 1.4 (as reported bu ddt4all) interface.
First issue is that ddt4all cannot find an ECU file for my unit. I have software/version D100/5A5A and database files have software/version B300 or B600/0000.
Tried and succeed to connect with the closest ECU file and it shows a lot of information.
Tried to do more, like actuating the motor and receive a strange reply, for ex. "Sending ELM request 30 18 00" followed by "Bad ELM response: WRONG RESPONSE : Unknown" 

What could be the reason for a "Bad ELM response" ?

cedricp, 2018-02-06:
I'd need your log files for analyses.

radunre, 2018-02-06:
Here are the logs.
[log files.zip](https://github.com/cedricp/ddt4all/files/1699277/log.files.zip)

cedricp, 2018-02-06:
Did you try to switch the diagnostic session to factory (then OK button) before ? If so, I found nothing that can help in the log files, just that the ECU refuses the sent frame...

radunre, 2018-02-06:
Yes, I previously switched to "Factory and after sales session" that has replied fine. The "Bad ELM response" came when I tried to manually control the unit or when I tried to send movements commands.

From the screen of "02-01 - APB process" I was able to reset, start repositioning etc. but from "01-05 Manual APB command" I never succed,always 

I thought it's about a too stupid ELM but with BCM I was able to send commands without issues.

radunre, 2018-02-06:
Sorry, button mistake on phone 😀