cedricp/ddt4all #34: Tutorial of how to modify OBD connector?

Seguro19, 2018-02-07:
Hello, i tried to connect with OBD WiFi and another via bluetooh and all ok, can connect to my car (Megane 3 RS) but i want to activate android auto and see Navigation and Audios ECUS and see another ecus but always when i click on scan appears the same 11 ECUS and not audio or navigation, i read in all sites that i have to modify my obd connector pins 12 and 14 but i dont know how or what i need to do that, can some home give some instructions of how to do it? (For noobs) haha.

Where can i find the cable and how to change the pins from my obd?

Thanks!!

cedricp, 2018-02-07:
https://github.com/cedricp/ddt4all/wiki#how-to-access-multimedia-can-bus

fparmentier, 2018-02-07:
Hi have modified a Ford OBD adapter with a PIC18F25K80 chip. These adapters have a toggle switch to switch between different CAN networks of Ford. I changed the connections from pin 3 to pin 13 and pin 11 to pin 12. This way you can switch between the multimedia CAN bus and normal CAN bus.
The only thing you will need is a soldering iron and a screwdriver.

https://www.aliexpress.com/item/100-PIC18F25K80-Chip-ELM327-V1-5-USB-Switch-ELM-327-For-Ford-HS-CAN-MS-CAN/32829576979.html

ohayyam, 2018-02-08:
Hello to all. 

Sorry for the delay but, between the work and the fever I have not had time to try to "explain" my developments. 
First I would like to warn you that, to complete the work, you need a little bit of familiarity with the use of soldering iron (preferably professional) and, in any case, as I said from the beginning, it's all at your responsibility. 
First step to where I started: make a PHOTO of the OBD socket of the car, as you can see (the pictures are all in the final link), has only 8 connected PIN. Here so I tried to find out what he was referring to the pin ( http://pinoutdiagram.com/renault-obd-ii-diagnostic-connector-pinout/) And I have tried to reproduce the same configuration interface ELM used for the various "procedures" well known. In this case, therefore, having to interact with the multimedia CAN bus, it is necessary to make a change of the cables which intercept the "high level" interface signal (6 and 14) and interposing a diverter that, using the same PIN, divert them toward those of "low level" (13:12) Renault (the diagram and photos of the links are always in the link below). Then, sore, it is necessary to remove the resistance (R04 from 120Ohm) which is located in the contact bridge (6 and 4) on the base of the ELM. Finally, once all closed, one may proceed by connecting the ELM to the machine, and then open the "mythical" DDT4ALL and enable (low level) in ' ECU A-IVI (In-Vehicle Infotainment Android) DAB parameter. Subsequently, returning to the ECU RadNav (high level), simply turn the DAB parameters that I had sent earlier to Loris. As always, at the end of it all it makes a nice reset the device and .... Bingo!

Read more at https://www.renaultclub.it/index.php?threads/radio-dab-su-scenic-iv-risolto.3172/page-5#06UxHvbSzy3GRtWh.99

he was mentioned to remove a resistance at forum. Any idea ?

Seguro19, 2018-02-08:
Ok great but where can u find the cables? Because i think there are data cable right? Where can i but It?

ohayyam, 2018-02-08:
https://www.drive2.ru/l/468057702971474426/    

But above link did not mention about removing resistance.

cedricp, 2018-02-08:
Indeed, a 120 Ohm resistor shouldn't be integrated, as the vehicle bus has already a termination on each ends.
Thank you for the feedback.

Seguro19, 2018-02-08:
Ok so... i recently get this 
[image]
[image]
Can i establish the cable betweet my obd elm to the new adapter using this on the obd elm:
[image]
And then connect it to my car?

ohayyam, 2018-02-08:
[image]

No need to second connection socket. just follow the link above for procedure.

Banana009, 2018-02-27:
I modified according to the instructions above, remove the resistance  of ELM adapters (USB, wifi adapters).  Adapters communicate with RadNav without problems  but not with A-VI v0.9.  
 am not able to to read/write to A-VI v0.9 .When I switched to LS CAN,  scaned it , DDT4ALL found only Central Panel 7/8.7 :-) Nothing else  :-1: 
Can someone help me?
[image]

ohayyam, 2018-02-28:
I am not sure but next step is try to find original ELM327 chip which is 21 USD. Chines clone can work limited. Just check obdlink sx ( Red one when we open the program it is proper for ddt4all. )

If it will not work means we have to wait for updated database...

Thanks & Regards,

[image]

lucamagilla, 2018-03-18:
On the Italian site only one seems to have succeeded

testeur70, 2019-03-14:
> Hi have modified a Ford OBD adapter with a PIC18F25K80 chip. These adapters have a toggle switch to switch between different CAN networks of Ford. I changed the connections from pin 3 to pin 13 and [...]

Hello Felix, I have the same elm and want to use it with ddt on my scenic. Can you confirm that you just have to switch  pin 3 to pin 13 and pin 11 to pin 12, to have a fully working ELM with high and low CAN ?
Thanks

fparmentier, 2019-03-14:
> Hello Felix, I have the same elm and want to use it with ddt on my scenic. Can you confirm that you just have to switch pin 3 to pin 13 and pin 11 to pin 12, to have a fully working ELM with high [...]

Hi,

Yes that's all. :)

mnigten, 2019-03-15:
I have this one from August last year and modified as discribed. It works great.

TimvdEijnden, 2019-08-31:
Yes this also worked great for me!

myshuker, 2021-01-27:
hey friend 
can I use this ?
https://www.aliexpress.com/item/4000955430101.html?spm=a2g0o.detail.0.0.6ae92d1cdsLP9L&gps-id=pcDetailCartBuyAlsoBuy&scm=1007.12908.197732.0&scm_id=1007.12908.197732.0&scm-url=1007.12908.197732.0&pvid=2560a7d1-bf0a-4eac-818e-ee46eb6946be&_t=gps-id:pcDetailCartBuyAlsoBuy,scm-url:1007.12908.197732.0,pvid:2560a7d1-bf0a-4eac-818e-ee46eb6946be,tpp_buckets:668%230%23131923%238_668%23888%233325%233_668%232846%238110%231995_668%232717%237561%23306__668%235154%2323664%23130

fparmentier, 2021-01-28:
Looks good to me :)

myshuker, 2021-01-28:
> Looks good to me :)
so it can be used for renault and dacia cars?

myshuker, 2021-01-28:
> Looks good to me :)

and witch chosse for renault cars is HS CAN or  MS CAN?

fparmentier, 2021-01-28:
> 
> 
> > Looks good to me :)
> > so it can be used for renault and dacia cars?

I think so, but only after you change the wires for the media can bus. But I assume you are aware of that ;)

myshuker, 2021-01-28:
> > > Looks good to me :)
> > > so it can be used for renault and dacia cars?
> 
> I think so, but only after you change the wires for the media can bus. But I assume you are aware of that ;)

thanks friend , but i assume they was already change the wires in this modul, because the y put a switch , is that correct?

fparmentier, 2021-01-28:
> thanks friend , but i assume they was already change the wires in this modul, because the y put a switch , is that correct?

You are right, but Ford uses another pin assignment for the media can bus.
Change the wires connected to the pins as follows:

Media CAN L - Ford pin 3 change to pin 13
Media CAN H - Ford pin 11 change to pin 12

myshuker, 2021-01-28:
ok ... BIG THANKS my friend

luissilva1971, 2021-01-28:
What about pic18F25K80 ... is it required or not for DDT4all and Renault/Dacia Cars? the one the link is showing does not have it... it apears!

fparmentier, 2021-01-28:
> 
> What about pic18F25K80 ... is it required or not for DDT4all and Renault/Dacia Cars? the one the link is showing does not have it... it apears! 

It says it does. But you can always ask the seller to be more sure.
[image]

myshuker, 2021-01-29:
[image]

myshuker, 2021-01-29:
[image]

VVitchfinder, 2021-07-13:
> You are right, but Ford uses another pin assignment for the media can bus.
> Change the wires connected to the pins as follows:
>
> Media CAN L - Ford pin 3 change to pin 13
> Media CAN H - Ford pin 11 [...]

Hey Felix, hoping you can help. I have a Ford elm 327. I know I need to change the pins. They seem to be solded straight to the pin. So don’t see any actual wires. 
[image]

how to I swap 3 to 13 and 11 to 12 like this? Do I disconnect the pins somehow, or can I solder a wire from 3 to 13 and 11 to 12. I’m a complete novice so any help would be really appreciated.

VVitchfinder, 2021-07-15:
I have an obdlink sx (red) do I have to mod this to get it to connect to my Renault? I have a new media nav with the system config message and I want to fix it in ddt4all. I was told it was just plug and play but it doesn’t seem to work.

AndyIreland, 2021-07-19:
https://andyrobus.wordpress.com/2021/07/18/wiring-up-an-obd2-splitter-cable-to-read-a-renault-r-link-radio/

charliehoward, 2021-09-17:
Can someone clarify, I connected the pins 6 to 13, and 14 to 12 in my adapter but it didn't work. What am I doing wrong?
[image]
[image]

charliehoward, 2021-09-18:
I'm not sure where I got those above pins from but I've removed them and followed the above using a Ford adapter -  pin 3 change to pin 13 and pin 11 change to pin 12. But it's still not working, when I plug it in I get loads of errors on the dash like braking system failure and ESC. Can someone help me out?

AndyIreland, 2021-09-18:
> I'm not sure where I got those above pins from but I've removed them and followed the above using a Ford adapter - pin 3 change to pin 13 and pin 11 change to pin 12. But it's still not working, [...]

When you do a 'find my ecu's' in the car using the DDT4ALL magnifying glass then sometimes the ABS light will light up and flash or flicker - and possibly a message about ESC as well and DDT4ALL is talking to the ECU's and connecting to them . 

What parameters are you trying to change on your car @charliehoward  ? - is it the rlink radio parameters or other ECU's? - if its the other ECU's of the car you do not have to make that adjustment of soldering the pins

charliehoward, 2021-09-19:
I've got the adapter working now, I'm trying to edit R-Link, I want to enable this when reversing for the rear sensors but not sure where to look.
[image]

myshuker, 2021-09-19:
whitch car do you have ?

charliehoward, 2021-09-19:
Megane 3 RS with R-Link retrofitted

myshuker, 2021-09-19:
> Megane 3 RS with R-Link retrofitted

could you please explaen or take picture how you connect the reare view camera in R-Link system and hoe you activate it on the system??? PLEASE

charliehoward, 2021-09-19:
I don't want the rear camera, just the reversing sensors like in the photo. I've been told it's possibly to enable I just don't know where to look

myshuker, 2021-09-19:
> I don't want the rear camera, just the reversing sensors like in the photo. I've been told it's possibly to enable I just don't know where to look

I know that my friend .
But I want to know how to connect and activate it because I have megan3 and i want install rear view camera.
please can you help me

charliehoward, 2021-09-19:
I haven't connected a rear camera so I can't help you sorry, if you do find out let me know

AndyIreland, 2021-09-20:
@charliehoward - so you have the ultrasonic sensors fitted in your rear bumper yes? - and you just need to enable them in rlink? is that what you mean?

charliehoward, 2021-09-20:
> 
> 
> @charliehoward - so you have the ultrasonic sensors fitted in your rear bumper yes? - and you just need to enable them in rlink? is that what you mean?

No, my car came with rear sensors from factory but I've retrofitted R-Link and want to enable the screen below with the sensors 
[image]

AndyIreland, 2021-09-20:
> @charliehoward - so you have the ultrasonic sensors fitted in your rear bumper yes? - and you just need to enable them in rlink? is that what you mean?
> 
> No, my car came with rear sensors from factory but I've retrofitted R-Link and want to enable the screen below with the sensors

OK - so , where did the beep beep warning come from these factory fitted rear bumper sensors ? - through the factory fitted radio/infotainment system or through a speaker in the boot or from dash where speedometer is?

charliehoward, 2021-09-20:
No idea, it was all there when it left the factory. All I want to know is how to enable that screen shown above
[image]

AndyIreland, 2021-09-20:
Right, OK lets see. Have you got this reverse  sensors showing up in your list of ECU's  (Valeo) because i Reckon the rlink just works with Valeo UPA sensors (I could be wrong but thats what I reckon) 
[image]

charliehoward, 2021-09-20:
https://user-images.githubusercontent.com/17574576/134045883-2d4767a7-1227-482f-a565-e3b31dc59581.mp4

This is what I had

AndyIreland, 2021-09-20:
I cannot see any UPA (universal parking assist) sensors at all in your list of found ECU's and to me I think you will need that ECU for the information you want showing up on the MFD there (I could be wrong though) - maybe someone else would know. - its not like it takes the info from the reversing camera to light up those segments it takes the data from the bumper UPA sensors

charliehoward, 2021-09-20:
Maybe for the rear sensors but I've been told it's in the R-Link part as it's just enabling a screen, I imagine it's in the same section as the rear camera, do you know where that is?

AndyIreland, 2021-09-20:
in that case then , maybe this is the settings you have to make to the MFD settings?

[image]

charliehoward, 2021-09-21:
I've managed to get this screen up for the rear camera
[image]
But can't find anywhere to make the sensors appear like this
[image]

charliehoward, 2021-10-09:
I haven't yet, I think you have to retrofit the newer parking module so I've made this adapter and will let you know if fitting the newer one works.
[image]

MellotronScratch, 2021-10-09:
You can also modify this VW cable and make a converter cable for the new type of pdc ECU. I've done it to make my sensors show up on the display.

https://a.aliexpress.com/_uju0dm

charliehoward, 2021-10-09:
Anything else you need to do other than just fitting that module?

AndyIreland, 2021-10-09:
> I haven't yet, I think you have to retrofit the newer parking module so I've made this adapter and will let you know if fitting the newer one works. [image]

You most probably know what you are doing but if its any help to you at all @charliehoward  - here is a wiring diagram for the Renault Zoe that uses that parking aid computer:

[image]

MellotronScratch, 2022-03-22:
Does anyone know if this VGate VLinker cable will work with dtt4all? Especially the automatic switching sound interesting.
This looks like a good alternative to an ELS27 adapter to me.

https://aliexpress.com/item/1005003116796694.html
[image]

MellotronScratch, 2022-03-26:
I decided to give it a try, the cable arrived a few days ago.
It seems ddt4all is able to read the main CAN bus (115200 baud or higher) with both ELM327 or ELS27 option selected. It is noticeably a lot faster than a normal ELM327 or OBDLink SX
I modified the connector to read pin 12 and 13 in stead of the Ford pins (3 and 11)  but unfortunately the Multimedia CAN bus is nog being read. I've attached some pictures of the PCB, it seems there are two CAN controllers (MCP2551) and two DPDT switches connected to a MIC3322. Unfortunately I cannot find any datasheet about it :(
Maybe somebody knows the code how the implement to read the CAN2 bus?
@cedricp 

[image]

AndyIreland, 2022-03-26:
@MellotronScratch  - is there a PIC18F25K80 chip in there , I cannot see it

MellotronScratch, 2022-03-26:
@AndyIreland No there isn't. The brain is a MIC3322.

Furtif, 2023-06-02:
if not still repply move question to https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif

vicentmartinezmartinez, 2024-08-08:
> [image]
> 
> * pin 6 en el lado ELM a pin 13 en el lado OBD
> * pin 14 en el lado ELM a pin 12 en el lado OBD

Thank you all, I manage to connect to my renault zoe to activate the rear camera.
this pins changes on the obd2 are necessary.

[image]

[4 more replies]