cedricp/ddt4all #35: Activation LKA Scenic IV

Banana009, 2018-02-08:
Did anyone activate LKA  (Lane-Keeping Assist ) or DW  (distance warning) or ACC ( Adaptive Cruise Control )?  I have Hands Free Parking so I have all the components . I am not able to activate LKA.  I set it in RadNav2.3 and then XFA_RACam_v3.7 - LKA.LKA_Configuration_Status. Function is active in Rlink2 menu, but does not work physically, side arrows are not green. Maybe it has to be activated  in A-IVI_v0.9 .?  A-IVI_v0.9 does not communicate with ECUs :-(

ohayyam, 2018-02-09:
You are at wrong way brother. You can try to activate "LDW | Lane Departure Warning System"  not LKA. My Talisman LDW working and i am trying to activate LKA which is different than LDW.  LDW just warning with sound and visual but LKA should keep car between lanes.

Banana009, 2018-02-09:
Why wrong way? New Scenic IV  has LDW and LKA. Talisman has only LDW.  
http://ie.e-guide.renault.com/eng/New-Scenic/LANE-KEEPING-ASSIST

ohayyam, 2018-02-09:
Talisman also have hands free parking and have all components and LKA can be activated thats why we are using DDT4All rigth ? I may miss undestand when u wrote side arrows not green means your LDW also not working rigth ? Thats why you are on wrong way. First u have to try to activate LDW then u can continue to try for LKA. I think if LDW not works LKA will not works...

Banana009, 2018-02-09:
Talisman has different components that do not allow LKA. Only new Scenic has LKA because it has a vibrating wheel. (haptic LKA) . 
I have active LDW but LKA is deactivated (the region rules) . See attach. the side arrows are still gray 
[image]

ohayyam, 2018-02-09:
I see.. Did you check this screen :)
[image]

Banana009, 2018-02-11:
Unfortunately it does not work. WRONG RESPONSE , see: 
[image]

cedricp, 2018-02-13:
Is your ECU auto detected with the scan button ? You probably are using the wrong file

Banana009, 2018-02-13:
This file is for Scenic IV, is not it true? Maybe LOW-HIGH can problem?

cedricp, 2018-02-13:
There are several files depending on the ECU software version and vehicles options. You should only use those that are detected by the scanner. For example, the injection unit is not the same if the engine is diesel or gasoline.

Banana009, 2018-02-13:
I will try it. Where áre updates of ECUs  .xml files ?

cedricp, 2018-02-13:
As indicated on the **readme** file, you must own that copyrighted database of xml files.

Banana009, 2018-02-13:
There is a problem that there are many files and nobody knows what they are doing.

cedricp, 2018-02-13:
Sorry, but this software is not an easy way to tune your vehicle, it's an engineer tool that needs some experience and study.

ynkpars63, 2018-02-19:
Hello 
there. my agent is using MeganE 4th program for a long time. In the program I could not find a section about the mirrors automatically falling back to the mirrors of the mirrors. I have a feature like this in talisman models and I am sure it is in ECU but I can not find it. I am very happy to help you with the subject.

Banana009, 2018-03-13:
The problem solved, . You only have to press the start button but not the pedals!! After that, I am able to write to CMF1_REPS.

closed 2018-03-13 by Banana009

tomka1118, 2018-05-10:
Please help me, I've made all the settings that you wrote here but doens't work for me.

Can I reset some of the devices over DDT4All to start working?

Banana009, 2018-05-11:
Every time you start the engine, you must turn the LKA manually from the R-link screen, otherwise it does not work :-(

tomka1118, 2018-05-11:
I'm new in this busienss.

Idon't have the selection possibility in the menu of LKA/LDW menu, only have the basic settings (Haptic level and strongness)

Could you write down a Quick guide?

Banana009, 2018-05-11:
It is very dangerous, if you do not have enough knowledge, it can damage the vehicle.
Are you able to read and write to ECUs?

tomka1118, 2018-05-11:
yes I can. 

I'm an IT guy with electrician PhD.

I bought a really good ELM 327 adapter witg original ELM chip.

So there can't be any problem.

tomka1118, 2018-05-12:
Have you tried to enable the R-sound option? it is working on Scenic 4 ?

tomka1118, 2018-05-12:
I made it! Don't know how but a made all the settings again and restarted the R-Link 2 and it worked.

I have the Screen Mirroring and the working LKA/LDW menu.

I also enabled all the functions thru driving. It works too.

I enabled the R-Sound but didn't appear the menu option. I have look after.

Banana009, 2018-05-14:
GOOD work, you have to reset R-link after avery change( long press start button for more then 10 sec) There are 3 types of reset, be carefull:
"Hard Reset" simulates a shutdown of the power supply.
"key off on Reset" simulates the drain and turn on the ignition with the key.
"Soft Reset" allows initialization of certain program units and their storage structures.
R-sound allows to add additional sound (Alpine), It works for me (Intense) , but I read that BOSE cars do not work. :-(

tomka1118, 2018-05-14:
Thanks,

I'm not in love with this new Bose, in my opinoin the sound in SCIII was better than actual.

This microphone volume level setting has more disadvantages and less advantage.

Wbr

Banana009, 2018-05-14:
I did not want BOSE because it has a large amp in the storage space :-(

tomka1118, 2018-05-14:
Why could they fit the complete amplifier under the passenger's seat in the SCIII? 

The actual has a "boom-box" sound and I don't like it. Nor my child in the second row...

payamt20, 2018-05-18:
Hello,
how I can active auto closure of sunroof and windows for talisman by DDT4all?
Please guide me. Thanks

tomka1118, 2018-05-18:
I don't know the Talisman. But.

First of all you have to have the initial package:
 - a good ELM (USB or Wifi),
 - this program (DDT4All) and the corresponding database (ecus.txt)

 When all you have, you should test your ELM compatibility, because a bad/fake ELM can destroy your cars ECU and all others...

If you don't know what do you want to change, first learn to use the DDT4all in the "Edit mode".

I found only one corresponding setup point, it is in the Audio/RadNav 2.3/Configuration/ECU Part II. Then name what I think it could be: HMI/Settings/ Auto lock up.

But it could be maybe the auto door closing?? Have to try.

In my Scenic IV (like the old one SCIII) if I push the closing button on the drivers door twice, all the windows come up.

Wbr,

SorakCz, 2018-05-24:
Hi tomka1118,
I am interesting about switch on LKA. I have GS IV Bose. Do you know if you lost warranty by this adjustment? Could you please share some video how it work?
Thank you

Banana009, 2018-05-24:
Also, I think the service center does not recognize change settings. I think ECU does not record changes. You can always change your settings backwards.
After each diagnosis by an authorized service center, the settings will be reset to factory settings (here in Czech Republic).
Unfortunately this LKA does not hold the car in the middle of the lane but only changes the steering when the vehicle detects a line. It looks like ping ping. Do you understand me? Other car makers can keep a vehicle in the middle of a lane (VW, Skoda), unfortunately Renault does not. LKA only helps in critical situations. What is uncomfortable that you must activate the LKA after each engine start.

payamt20, 2018-05-24:
Hello,
Where I can find Unlocking video playing while driving in Talisman.
Thanks

bostu, 2019-04-25:
Hello to all, 
not to open new post I would rather ask here.

do you think that LKA is possible in Espace V 2016... I also have Park assist and LDW by default.

When I look in catalogs for new talisman or espace there is no posibility to buy LKA in any version of the car.... it is only possibe as I see for 200€ at Scenic models. 

Did anyone try to activate on espace?

Thank you for the answers

Banana009, 2019-04-25:
Bad news for you, you never activate LKA on the Espace. You don't have RaCam, which calculates the trajectory.  Unfortunately, even on Koleus with RaCam, LKA cannot be activated because it has a different type of software.
Only the new Scenic IV can and the new Clio probably ...

bostu, 2019-04-25:
Thank you for the infolp Boštjan 
-------- Izvirno sporočilo --------Od: Banana009 <notifications@github.com> Datum: 25. 4. 19  08:57  (GMT+01:00) Za: cedricp/ddt4all <ddt4all@noreply.github.com> Kp: bostu <bostjan.stucin@gmail.com>, Comment <comment@noreply.github.com> Zadeva: Re: [cedricp/ddt4all] Activation LKA Scenic IV (#35) Bad news for you, you never activate LKA on the Espace. You don't have RaCam, which calculates the trajectory.  Unfortunately, even on Koleus with RaCam, LKA cannot be activated because it has a different type of software.
Only the new Scenic IV can and the new Clio probably ...

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/cedricp/ddt4all/issues/35#issuecomment-486543840",
"url": "https://github.com/cedricp/ddt4all/issues/35#issuecomment-486543840",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

bostu, 2019-05-17:
hello again

can I just ask you what kind of OBDII adapter do you use?

Thank you

ustunkaya07, 2020-06-13:
And what about LKA for megane 4? I have LDW, auto brake, Self parking etc. İs there any way to activate it?

scoobydoo511, 2020-06-14:
If you have all off those in your car, then you are probably able to activate LKA...

Dema323, 2020-06-15:
Did you find where you activate LKA for megane iv ?

ustunkaya07, 2020-06-15:
Unfortunately no, dude. Megane 4 and scenic 4 have different hardwares. To activate LKA, car Must have haptic steering wheel. Megane 4 has not :( the problem is depend on hardware, it seems

Dema323, 2020-06-15:
[image]
Just found this though

Banana009, 2020-06-15:
It is not possible to activare LKA on Megane. There is not Racam.

Dema323, 2020-06-15:
Well it can do ldw so don't see why lka wouldn't be possible.

ustunkaya07, 2020-06-15:
It does not make sense . There are radar, self parking, adaptive criuse control in Car. There are radar and all sensors but cannot activate LKA. There must be a way.

Banana009, 2020-06-16:
There is no chance to activate LKA on Megane because you do not have RaCam. The RaCam unit calculates the LKA. Buy a new car, Megane phase 2. :-)

Dema323, 2020-06-16:
And why is the racam not needed for ldw ?

micdesa, 2021-07-15:
Hi Guys, do you have any news about activating LKA on Megane IV? It's very strange that there are no wau to do it.
Thanks.

AndyIreland, 2021-07-19:
> Hi Guys, do you have any news about activating LKA on Megane IV? It's very strange that there are no wau to do it.
> Thanks.

what have you got a Ph2 Megane? there is a setting in navigation any help ? you most probably need extra hardware like camera in the windscreen if you havent already got it, and specially wired OBD2 cable or adaptor so you can read / write multimedia and OBD2 to not be secured by gateway 

[image]

micdesa, 2021-07-19:
Hi,
I have Megane Ph 1 Bose version...

AndyIreland, 2021-07-19:
what ECU does it find when you search with the magnifying glass in DDT4ALL? (you may need the specially wired OBD2 adaptor or cable to read the multimedia parameters through the CAN network)

AdameikoMaxim, 2022-01-08:
Sorry for asking again, but is it still no chance to activate for Megane 4 Ph 1? Even for Intens having all others assistants

AndyIreland, 2022-01-08:
> Sorry for asking again, but is it still no chance to activate for Megane 4 Ph 1? Even for Intens having all others assistants

So, i am predicting you have the easylink radio system and the OBD2 has got the security gateway block on it - if you have the camera for the LKA the setting is more than likely there in DDT4all but the security gateway has to be unlocked to make any changes

AndyIreland, 2022-01-08:
> > Sorry for asking again, but is it still no chance to activate for Megane 4 Ph 1? Even for Intens having all others assistants
> 
> So, i am predicting you have the easylink radio system and the OBD2 has got the security gateway block on it - if you have the camera for the LKA the setting is more than likely there in DDT4all but the security gateway has to be unlocked to make any changes - even then I am not sure after unlocking what is needed to change parameters on the easylink - on the rlink1 &2 you had to solder up some pins on the elm327 to read/write CAN-H  but I dont really know on the newer Renaults  with the security gateway if you still need to do that now or how you read/write to the easylink sorry

alquistus, 2022-05-13:
@Banana009 Hi, you wrote that it is impossible to activate LKA on Espace V ph1, because of HW. But is it possible to find RACam or steering wheel from Scenic for example and activate LKA?

gonzaloalbito, 2023-03-21:
I'm about to buy an Espace V from 2016. It lacks ACC and LDW. It has emergency brake assist and park assist so I'm not sure it has all needed sensors (cameras and proximity sensor) to enable LDW and ACC after buying it, or it will not be possible at all. Could anyone help me?