cedricp/ddt4all #36: about the main settings

ynkpars63, 2018-02-19:
Hello there. my agent is using MeganE 4th program for a long time. In the program I could not find a section about the mirrors automatically falling back to the mirrors of the mirrors. I have a feature like this in talisman models and I am sure it is in ECU but I can not find it. I am very happy to help you with the subject.

cedricp, 2018-02-25:
Not sure to understand your request, what are you trying to do ?

ynkpars63, 2018-02-26:
I want the mirrors to come down when I get back vitased.

tomidix, 2018-04-08:
I'm afraid that it won't be possible unless your agent has a driver's seat with electric controls and position memory enabled. I'm afraid that's a specific module which controls this. After you filter ECUs by Megane IV, you can take a look at ADP and LDCM groups of ECUs to try to enable memorization function. 
[image]
[image]

payamt20, 2018-05-18:
Hello,
I want active closing mirrors and sunroof  when locked doors by remote for Talisman.
Please guide me. Thanks

Furtif, 2023-06-02:
move you question to https://github.com/cedricp/ddt4all/discussions