cedricp/ddt4all #37: Mirrorlink on Talisman With RLink 2

sergegauthier, 2018-02-20:
Hi,
I am seaching some help and advices to use ddtall to implement mirrorlink on my Talisman car. 
My car was bought on 04-2016 and I would like to know if anyone has the experience to make this change. I have an ODB2 bluetooth jack that works. Thanks

cedricp, 2018-02-25:
Did you try to scan the vehicle bus ? If so, what are the detected ECUs ?

sergegauthier, 2018-02-25:
Hi, not yet. I'll do it soon and I tell you.

Thanks

OptionPlus, 2018-04-29:
hi i am doing the same thing 
i tried 2 different data base one of them detect radnav 2.3 and the other one detect radnav 2.1 both in red condition 
it can read data but cant write 
when i write elm respond 0 0 0

sergegauthier, 2018-04-29:
Hi OptionPlus, Could you tell me more about what you are doing ? Personnally, I finally managed to connect via a USB interface under linux. In the choice of vehicles, I can't find Talisman. Thank you for your advice.
cordially

OptionPlus, 2018-04-29:
Hi serge , finally i did my goal , do you whatsapp ?
i cant help you to reach your goal too
my number is +989121374004

Furtif, 2023-06-02:
if question reply ok if not ask as https://github.com/cedricp/ddt4all/discussions