cedricp/ddt4all #38: Timeout when accessing RadNav through ELM327 USB

MarbolanGos, 2018-02-21:
Hi,

I wanted to activate the mirroring tool (Waze in particular) in a Scenic IV where the option is not activated by default even if it is available...

My RLink2 version is 3.3.16.946
ddt4all version is the release from January 2018.

If I test my ELM USB device I have:
`v1.0 ATZ [OK] 
v1.0 ATE1 [OK] 
v1.0 ATL0 [OK] 
v1.0 ATI [OK] 
v1.0 AT@1 [OK] 
v1.0 ATAL [OK] 
v1.0 ATBD [OK] 
v1.0 ATBI [OK/TEMPS DEPASSE] 
v1.0 ATCAF0 [OK] 
v1.0 ATCFC1 [OK] 
v1.0 ATCP 01 [OK] 
v1.0 ATCS [OK] 
v1.0 ATCV 1250 [OK] 
v1.0 ATD [OK] 
v1.0 ATDP [OK] 
v1.0 ATDPN [OK] 
v1.0 ATH0 [OK] 
v1.0 ATI [OK] 
v1.0 ATIB 10 [OK] 
v1.0 ATIB 96 [OK] 
v1.0 ATL1 [OK] 
v1.0 ATM0 [OK] 
v1.0 ATMA [OK/TEMPS DEPASSE] 
v1.0 ATMR 01 [OK/TEMPS DEPASSE] 
v1.0 ATMT 01 [OK/TEMPS DEPASSE] 
v1.0 ATNL [OK/TEMPS DEPASSE] 
v1.0 ATPC [OK] 
v1.0 ATR1 [OK] 
v1.0 ATRV [OK] 
v1.0 ATSH 001122 [OK] 
v1.0 ATSH 012 [OK] 
v1.0 ATSP A6 [OK] 
v1.0 ATSP 6 [OK] 
v1.0 ATCM 123 [OK] 
v1.0 ATCM 12345678 [OK] 
v1.0 ATCF 123 [OK] 
v1.0 ATCF 12345678 [OK] 
v1.0 ATST FF [OK] 
v1.0 ATSW 96 [OK] 
v1.0 ATSW 34 [OK] 
v1.0 ATTP A6 [OK] 
v1.0 ATTP 6 [OK] 
v1.0 ATWM 817AF13E [OK] 
v1.0 ATWS [OK] 
v1.1 ATFC SD 300000 [OK] 
v1.1 ATFC SH 012 [ECHEC] 
v1.1 ATFC SH 00112233 [ECHEC] 
v1.1 ATFC SM 1 [ECHEC] 
v1.1 ATPP FF OFF [OK] 
v1.1 ATPP FF ON [OK] 
v1.1 ATPPS [OK/TEMPS DEPASSE] 
v1.2 ATAR [OK] 
v1.2 ATAT1 [OK] 
v1.2 ATDM1 [OK/TEMPS DEPASSE] 
v1.2 ATIFR H [OK] 
v1.2 ATIFR0 [OK] 
v1.2 ATIIA 01 [OK] 
v1.2 ATKW0 [OK] 
v1.2 ATMP 0123 [ECHEC] 
v1.2 ATMP 0123 4 [ECHEC] 
v1.2 ATMP 012345 [ECHEC] 
v1.2 ATMP 012345 6 [ECHEC] 
v1.2 ATSR 01 [OK] 
v1.3 ATCRA 012 [OK] 
v1.3 ATCRA 01234567 [ECHEC] 
v1.3 ATD0 [OK] 
v1.3 ATFE [OK] 
v1.3 ATJE [ECHEC] 
v1.3 ATJS [ECHEC] 
v1.3 ATKW [OK] 
v1.3 ATRA 01 [OK] 
v1.3 ATS1 [OK] 
v1.3 ATSP 00 [ECHEC] 
v1.3 ATV0 [OK] 
v1.4 ATCEA [OK] 
v1.4 ATCEA 01 [OK] 
v1.4 ATCV 0000 [OK] 
v1.4 ATIB 48 [OK] 
v1.4 ATIGN [OK] 
v1.4 ATPB 01 23 [OK] 
v1.4 ATRD [OK] 
v1.4 ATSD 01 [OK] 
v1.4 ATSI [OK/TEMPS DEPASSE] 
v1.4 ATFI [OK/TEMPS DEPASSE] 
v1.4 ATSS [OK] 
v1.4 ATTA 12 [OK] 
v1.4 ATCSM1 [OK/TEMPS DEPASSE] 
v1.4 ATJHF1 [ECHEC] 
v1.4 ATJTM1 [ECHEC] 
v1.4 ATJTM5 [ECHEC] 
v1.4b ATCRA [ECHEC] 
v2.0 ATAMC [ECHEC] 
v2.0 ATAMT 20 [ECHEC] 
v2.1 ATCTM1 [ECHEC] 
v2.1 ATCTM5 [ECHEC] 
Adaptateur ARM incompatible 
Résultat:76réussis sur95 
Version ELM maximum:v1.4`

When I try to get the data I have NODATA.  You can see the error on the screen : https://postimg.org/image/88anceq0z/

The logs shows in fact a timeout to access data : 
`>[07:32:47.074]ATZ
<[0.812]ATZ

ELM327 v1.5

>
############################################################
#[02/12/18 07:33:11.834] Init CAN
############################################################
>[07:33:11.834]AT WS
<[0.015]AT WS

ELM327 v1.5

>
>[07:33:11.849]AT E1
<[0.0]AT E1
OK

>
>[07:33:11.849]AT S0
<[0.0]AT S0
OK

>
>[07:33:11.849]AT H0
<[0.016]AT H0
OK

>
>[07:33:11.865]AT L0
<[0.0]AT L0
OK

>
>[07:33:11.865]AT AL
<[0.016]AT AL
OK

>
>[07:33:11.881]AT CAF0
<[0.0]AT CAF0
OK

>
############################################################
#connect to: RadNav v2.3 Addr:13
############################################################
>[07:33:11.881]AT SH 712
<[0.015]AT SH 712
OK

>
>[07:33:11.896]AT CRA 732
<[0.0]AT CRA 732
OK

>
>[07:33:11.896]AT FC SH 712
<[0.016]AT FC SH 712
?

>
>[07:33:11.912]AT FC SD 30 00 00
<[0.015]AT FC SD 30 00 00
OK

>
>[07:33:11.927]AT FC SM 1
<[0.0]AT FC SM 1
?

>
>[07:33:11.927]AT SP 6
<[0.032]AT SP 6
OK

>
>[07:33:11.959]021003
<[5.007]TEMPS DEPASSETIMEOUT
>[07:33:17.949]023E00
<[5.023]023E00
TEMPS DEPASSETIMEOUT
>[07:33:24.345]023E00
<[5.008]023E00
TEMPS DEPASSETIMEOUT
>[07:33:30.395]023E00
<[5.008]023E00
TEMPS DEPASSETIMEOUT
>[07:33:35.403]0210C0
<[5.007]0210C0
TEMPS DEPASSETIMEOUT
>[07:33:40.988]023E00
<[5.007]023E00
TEMPS DEPASSETIMEOUT
>[07:33:47.040]023E00
<[5.008]023E00
TEMPS DEPASSETIMEOUT
>[07:33:53.093]023E00
<[5.008]023E00
TEMPS DEPASSETIMEOUT
>[07:33:59.146]023E00
<[5.008]023E00
TEMPS DEPASSETIMEOUT
>[07:34:05.199]023E00
<[5.007]023E00
TEMPS DEPASSETIMEOUT
>[07:34:10.269]0210C0
<[5.007]0210C0
TEMPS DEPASSETIMEOUT
>[07:34:15.276]03222130
<[5.008]03222130
TEMPS DEPASSETIMEOUT
>[07:34:20.284]03222202
<[5.008]03222202
TEMPS DEPASSETIMEOUT
>[07:34:25.292]03222203
<[5.007]03222203
TEMPS DEPASSETIMEOUT
>[07:34:30.393]03222130
<[5.023]03222130
TEMPS DEPASSETIMEOUT
>[07:34:35.416]03222202
<[5.015]03222202
TEMPS DEPASSETIMEOUT
>[07:34:40.431]03222203
<[5.007]03222203
TEMPS DEPASSETIMEOUT
>[07:34:46.062]023E00
<[5.008]023E00
TEMPS DEPASSETIMEOUT
>[07:34:51.085]021003
<[5.008]021003
TEMPS DEPASSETIMEOUT
>[07:34:56.093]03222130
<[5.008]03222130
TEMPS DEPASSETIMEOUT
>[07:35:01.101]03222203
<[5.007]03222203
TEMPS DEPASSETIMEOUT
>[07:35:06.108]03222202
<[5.013]03222202
TEMPS DEPASSETIMEOUT
>[07:35:11.121]021001
<[5.007]021001
TEMPS DEPASSETIMEOUT
>[07:35:16.238]03222130
<[5.007]03222130
TEMPS DEPASSETIMEOUT
>[07:35:21.245]03222202
<[5.008]03222202
TEMPS DEPASSETIMEOUT
>[07:35:26.253]03222203
<[5.007]03222203
TEMPS DEPASSETIMEOUT
>[07:35:31.448]023E00
<[5.007]023E00
TEMPS DEPASSETIMEOUT
`
My ECU DB is DDT2000 DB 04.08.2017 

When connected the terminal output is: 
`Initialisation ELM a la vitesse 38400 
Tentative d'ouverture du portCOM3 @ 38400 
Balayage ECU 
1366 ECU chargées en base de donnée`

I tried on a Windows 7 and Windows 10 laptop because under Windows 10 the dirver of the USB ELM plug is obsoleted...

Is there another way to debug?

Regards,
MarbolanGos

PS: I can speak French if needed.

cedricp, 2018-02-21:
You can try to manually increase the **can timeout** value up to 1000ms (bottom of the screen), if it doesn't work, try  connecting another ECU (engine for example) in the car and check if you get something. If not, your ELM is probably faulty. Have you tried to scan the vehicle bus ?

MarbolanGos, 2018-02-22:
Thank you for your fast reply. I tried again to test with the timeout and it still failed.
As I tried with another application (ELM scanMaster delivered with the cable) and I was able to get live values then I thought the ELM plug was fine.
I am going to have someone who will lend me another plug to test.
Now, I am starting to believe the ELM is really faulty...

cedricp, 2018-02-22:
There are 2 buses on the vehicle (one for multimedia, the other for vehicle dynamics), you may have to swap pins on the ELM if the RadNav is on the M-CAN (see other issues)

MarbolanGos, 2018-02-22:
According to this (French post): it is not needed http://www.planeterenault.com/forum/r-link-waze-t49270-72.html#p1023824

Banana009, 2018-02-22:
Could you translate French post please? , I am not able to read even with M-CAN (Changed pins and disconnected resistor ) for example  A-IVI_v0.9. :-(

cedricp, 2018-02-22:
The post says that you don't need to connect to M-CAN (no connection modification, works as-is) for  V3.3.X

Banana009, 2018-02-22:
In RadNav is it possible of cource. 
I do not understand the communication, I did the modification OBD2 to  M-CAN according to the instructions but it does not work. I do not understand why? Does any function have a terminal resistance? I think not.

MarbolanGos, 2018-03-14:
Hello,

I have the plug and it works. This was due to my plug which was in fault. Thanks !

closed 2018-03-14 by MarbolanGos