cedricp/ddt4all #39: rlink2 problem

gvaldo, 2018-03-01:
parking sensor is not displayed on rlink2 kadjar

tomidix, 2018-04-02:
Try to restart the RLink2 system. While it's turned on, hold the On/Off button until the screen shuts off (long time, around 10-15s). Do this when vehicle is stationary as many of the functions will become unavailable, for example ambient lightning will stop working while reboot is in progress. If this doesn't help, then either the parking sensor option is turned off in vehicle settings or it is a hardware problem.

thomson56, 2019-06-12:
Hello
I have a problem how to get back to system 2.2.19.300 from 7.0.24.160, I tried probably all possibilities including Virgin mode niepey to no avail, the car does not see my account on My Renault. Please help.

AlMeu, 2021-04-01:
As I know there is no way to go back from v7 to v2, just bricking the unit.

Furtif, 2023-06-02:
if same issue move it to https://github.com/cedricp/ddt4all/discussions