cedricp/ddt4all #41: Activate wing mirror lights.

Potty06, 2018-03-10:
Hi. I hav a Megane IV 2016 I have keyless start and keyless entry. I want to activate something like external welcome function. I have lights under wing mirror. But they do nothing. I want make it works In light config in dd4all I did not found nothing like "Light wing mirror" Can you help me please?

gordon6, 2018-03-21:
Hi, Potty6...
Somebody called "Welcome Light", if you want to enable the Welcome Light function.
You can find HFM easily, then double click C1A_HFM_DDT_4_0 and you need select HFM Monitor, finally you can see HFM Monitor - Configuration.  That's it.
You can change "CUSTOMER_APPROACH_CF" value to TRUE.
May the Force be with U...

Potty06, 2018-03-21:
Thank you so much, friend. You are a hero!! I will try and I will tell you if this works. I have "one type of welcome light" DRL on, wing mirror oppening, rear led on when I came to the car. But no light under mirrors. Thank you so much I will try it. You are a hero

Potty06, 2018-03-21:
[image]

CUSTOMER_APPROACH_CF ok I found it
but what is WELCOME_GOODBYE_CF? (I am in EDITION MODE NOW not connected to the car, probabbly this is ON on my car, if is FALSE? I should put it TRUE ?

Potty06, 2018-03-22:
I tried to go in C1A_HFM_DDT_4_0 but I dont have that, I have C1A_HFM_DDT_3_6 (only this) and all colums are RED with NO DATA :( But in UCBIC/BFR > CMF1_BCM_DDT_11 > Monitor > Monitor - Configuration > CUSTOMER_APPROACH_CF it's setted on true.. and still no lights. How can I take C1A_HFM_DDT_4_0? it's a new data base?

tomidix, 2018-03-30:
I have them too. They actually work only at night. However there is a bug with them and sometimes they do not turn on even when the car is parked at 100% dark place. There are 2 events when they do turn on - approaching the car / turning the car off and opening the doors.

Regarding the version 4_0, this should be the new ecu version which I also haven't found.

I'd like to turn this lights on regardless the day/night.. Is this possible?

Furtif, 2023-06-02:
if not issue found ask in https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif