cedricp/ddt4all #42: Invisible commands in dd4all

Banana009, 2018-03-12:
I noticed that there are many commands in .xml files that are not visible on the Screen. For example this one: Is possible to read/write this parameter LDW_LKA.LDW_LKA_SelectionState_memorized ?
[image]

cedricp, 2018-03-13:
That should be possible, but you need to find the requests that read/write that data item before.

Banana009, 2018-03-13:
And how? I did not find this option in Screen menu.

Banana009, 2018-03-13:
Can I send/read a command directly from the bookmark Requests?

cedricp, 2018-03-13:
Unfortunately not, you must build a screen with parameters/buttons.

Banana009, 2018-03-13:
Would it do the next version of ddt4ll ? It would be very useful. :-)

Banana009, 2018-03-13:
How to add a write button?  According to your instructions, I have only a command, I can not read and write to the ECU.  See:  
[image]
Thanks for the help

Potty06, 2018-03-13:
Do you found somthing for puddle lights?(wing mirror lights)? I want to activate it

cedricp, 2018-03-14:
Go to the screen, enable "**UI EDIT**", right click on the screen -> "**add button**"
Go to button tab, select the request you want to send (_LDW_LKA.LDW_LKA_SelectionState_memorized_) and click "**ADD**"
Save your file, then reopen the app in connected mode, and you should be OK

cedricp, 2018-03-14:
You can use the plugin system to do it also, but it's a bit more complicated

Banana009, 2018-03-14:
I will try the first way.  :-)
By the way, why are not all commands in the Screen with buttons for read/write to the ECU?

cedricp, 2018-03-14:
Because the request must be tagged "manual" in the request spreadsheet (last column).

Banana009, 2018-03-14:
I can not open it in the Connected mode. Screens are empty. In the Edition mode I can save an ECU just  like .jcon .
There is only this option. I created testing ECU, see here: 
[image]

arifcalhan, 2019-03-05:
[image]
iyi günler ustatlar emniyet kemeri sekmesini nasıl aktif hale getirebilirim yardımcı olurmusunuz?

xormail, 2019-04-10:
Hi, I am trying to edit, but I just says:
To be able to edit your screen, first export it in JSON format

So I go File-Save current ECU-> give name, Save and no change. Also the Requests and Data tabs are "locked". Thanks

cedricp, 2019-04-10:
You need to convert your database in JSon format, go to File->Zip database. It's a long process...
Once done, you should find an ecu.zip file. Remove ecu directory (the one containing XML files) and you're done. You can now edit requests/data/screens.

arifcalhan, 2019-04-11:
yorumlarınız için teşekkür ederim fakat yapamadım json dosyasını oluşturup mail olarak paylamanız mümkün olabilir mi? cluster b95 L38 ph3 V192 fluence(x38) mail adresim ( arifcalhan@hotmail.com ) şimdiden çok teşekkür ederim iyi günler

Furtif, 2023-06-02:
seem a stuff but not a issue move question to https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif