cedricp/ddt4all #46: Help radio DAB Scenic 4

lucamagilla, 2018-03-21:
Hello to all I have a Scenic IV and I would like to activate via dd4all the dab radio but I have the problem that I can not read the screen A-IVI 09, I changed the obd socket by diverting the signal from the poles 6-14 to 13-12 but nothing I can not read. Does anyone have any ideas? Thanks

Banana009, 2018-03-23:
Unfortunately, no one has yet been able to activate DAB (read from A-IVI 09). Only one person at the Italian forum, which is strange.
I have tried 3 types of ELMs without success.
I would also like someone to solve this problem and have more success than I do.

lucamagilla, 2018-03-23:
Hi Banana, I'm Magilla from the Italian renault forum, I've also tried more obd jacks and at this point I have the doubt that it is a lie that of magellano of the italian forum

ohayyam, 2018-03-27:
Did you check your radio that have digitale antenna socket and digital antenna connected ?
[image]

gvaldo, 2018-03-27:
Hi ! I have a problem with parking sensors. they do not work, error message (help check parking). I can check with ddt4all where the error is.

Banana009, 2018-03-27:
I have tried 3 types of OBD adapters without success!! It's a lie. Do not waste money on other OBD2 adapters. It cost me around 100 EURO!!
Another problem is, as Ohayyam wrote, that we need another antenna for DAB. Is it not clear if an antenna is connected?

mishu28nmv, 2018-10-22:
I activated DABwith ddt4all. It showed on my Rlink1 display but there is no signal so... I need to buy an external antenna and a box from a site autodab. com and that will work only if you have the brown connector, wich Rlink1 doesn't.
picture: [https://imgur.com/a/LcP04XJ](url)

Rcarnero73, 2020-02-23:
> Did you check your radio that have digitale antenna socket and digital radio antenna connected ?
> [image]
> 
> As far as I understand, Renault only hides the ongoing features of the Renault engineers' development work, which are not fully functional, so the features that can be opened with ddt4all are hidden and not fully functional. I think that the features that do not work with DDT4all are caused by lack of hardware.

Do you have the complete pinout for Rlink 2?

ohayyam, 2020-02-23:
pinout diagrams depends your car and radio type. tell me your car and radio type ?

bokkoman, 2020-02-29:
> I activated DABwith ddt4all. It showed on my Rlink1 display but there is no signal so... I need to buy an external antenna and a box from a site autodab. com and that will work only if you have the brown connector, wich Rlink1 doesn't, at least on the back of the display.
> [Rlink1 display back connectors](https://imgur.com/a/LcP04XJ)

The DAB connector would be at the radio module, not behind the display unit.
Rlink has a display unit and a separate radio module/unit. In the Clio 4 it is behind the glovebox, which you can take it very easily if you look at the hinges (you can slide them to the side).

mishu28nmv, 2020-03-01:
> > I activated DABwith ddt4all. It showed on my Rlink1 display but there is no signal so... I need to buy an external antenna and a box from a site autodab. com and that will work only if you have the brown connector, wich Rlink1 doesn't, at least on the back of the display.
> > [Rlink1 display back connectors](https://imgur.com/a/LcP04XJ)
> 
> The DAB connector would be at the radio module, not behind the display unit.
> Rlink has a display unit and a separate radio module/unit. In the Clio 4 it is behind the glovebox, which you can take it very easily if you look at the hinges (you can slide them to the side).
> 
> Can you tell me how you have activated DAB with ddt4all?

[image]
With ddt4all go to  MFD vX.X then 1.Configuration then 12.ECU configuration then look for Digital Radio option.
I hope this help!

bokkoman, 2020-03-01:
> 
> [image]
> With ddt4all go to MFD vX.X then 1.Configuration then 12.ECU configuration then look for Digital Radio option.
> I hope this help!
Which database version do you have?
My mfd goes to version 4.6. You have a higher version.

mishu28nmv, 2020-03-01:
I have Laguna 3 with R-Link1 wich has/is MFD v4.6. It does not care what MFD you have. Just follow the steps.
As for a module and an antenna:
1. I don't know where is the radio module located in Laguna
2. I don't have an antenna (just an external gps antenna with Fakra connector😂)
3. I don't know what connectors has the module
4. I wish you luck 😉

bokkoman, 2020-03-01:
> I have Laguna 3 with R-Link1 wich has/is MFD v4.6. It does not care what MFD you have. Just follow the steps.
> As for a module and an antenna:
> 
> 1. I don't know where is the radio module located in Laguna
> 2. I don't have an antenna (just an external gps antenna with Fakra connectorjoy)
> 3. I don't know what connectors has the module
> 4. I wish you luck wink

Thanks haha. I have a Clio, i know where the module is located (behind glovebox). I was just wondering if replacing the module would make DAB work.

mishu28nmv, 2020-03-01:
I think we need to buy something like this:
https://www.niq.de/dab-autodab/dab-dab-modul-fuer-diverse-renault-modelle-mit-iso-stecker.html?language=en
and it's expensive at 230 € 😱
Or maybe something cheaper but I don't know yet what 🤣

bokkoman, 2020-03-01:
> I think we need to buy something like this:
> https://www.niq.de/dab-autodab/dab-dab-modul-fuer-diverse-renault-modelle-mit-iso-stecker.html?language=de
> and it's expensive at 230 € 
> Or maybe something cheaper but I don't know yet what

No. I'm looking at buying an official radio module for Rlink.
Rlink consists of 2 parts. The screen and a separate radio module.
My radio module only has am/fm. But there are newer Rlinks that came with DAB. Soooo what I'm looking at is new radio module with dab that came from a torn apart car (wrecked renault car).
I think that way i add an official antenna as well and then i have DAB radio.

mishu28nmv, 2020-03-01:
Ok. Good luck and have fun! 😁

myshuker, 2022-07-05:
Hey friends
anyone can help me where can I find 32 pin connector in Scenic iv  because I want to fit rear view camera ?
and if there are a video to how to remove radio unit ?

myshuker, 2022-07-05:
any one has 32 pin r-link 2 connector pins diagram ?

bokkoman, 2022-07-05:
> Hey friends anyone can help me where can I find 32 pin connector in Scenic iv because I want to fit rear view camera ? and if there are a video to how to remove radio unit ?

You need to find the radio unit.
The radio unit holds the rca plug for video. 
Then just activate it with ddt4all.

myshuker, 2022-07-05:
> 
thanks for replaying
but I want to know witch pin is for rear gear also please I have 12v camera,

bokkoman, 2022-07-05:
> > 
> 
> thanks for replaying but I want to know witch pin is for rear gear also please I have 12v camera,

You put the 12v wire to the reverse light. 
So mount camera, or red wire to reverse light 12v, black to ground. And the rca to the radio unit input.

myshuker, 2022-07-05:
I can just find 24 pin connector for Medianav on Clio4 on the internet 
but i cant fint 32 pin connector 
https://ibb.co/0q2DpD2

myshuker, 2022-07-05:
> You put the 12v wire to the reverse light.
in this case there are problem, because whe I relase the gear the screen be black for about 5 secound because there are not power on reverse light anymore to energize the camera

bokkoman, 2022-07-05:
> > You put the 12v wire to the reverse light.
> 
> in this case there are problem, because when I release the gear the screen be black for about 5 secound because there are not power on reverse light anymore to energize the camera

I have no clue beyond this.
Sounds like a trip to a (Renault) garage wouldn't hurt.

myshuker, 2022-07-05:
yeah,
But thanks for you friend 👍 
:)

closed 2023-06-02 by Furtif