cedricp/ddt4all #47: Renault Talisman config for ACC (adaptative cruise control)

games12345, 2018-04-01:
Hello,
I hope someone can help me. I have an Renault Talisman Initale Paris with normal Cruise Control/Limiter.
I want to change it to ACC (adaptative cruise control).
I already have fitted the switch for ACC/Limiter and the steering wheel.
Now I only need the config to activate the acc, I cant find it.
It is the 160 dci R9M 452 with DW6 dual clutch gearbox.
The car has everything inside but it dont have glas rooftop, the acc and the winter package (heated steering wheel and headlight washer).
I hope someone can help me to activate the acc for my car.
Thank you
Sry for my english im from Germany

tomidix, 2018-04-01:
Are you sure you have everything? Do you have sensor for ACC in the front logo and is the safe distance warning already available in your vehicle settings?

games12345, 2018-04-02:
Yes I have distance warning and speed limit sign identification. Also lane Assist (it beebs if i leave lane without flasher).

tomidix, 2018-04-02:
Ok, if you have Distance Warning (DW) working correctly then I believe you should be able to turn on ACC without a problem.
I have Lane Departure Warning (LDW) too in my Megane IV but it seems it uses different sensors than the DW as I haven't been able to turn the functionality on. After changing the DW setting to DW Present with Camera in ddt4all, I have successfully turned on the setting in the Rlink2. But after trying to turn it on in the RLink2 menu, it seems that the vehicle is aware (probably there is an exception in the code) that something is missing and it turns the setting to OFF by itself after 5-10 seconds. So I think it will be up to the sensor for DW if this will work on your vehicle.

1. You can find the management for your setting here: Tableu de bord -> Cluster_15-40_v3.30.02 -> Ecrans Configuration -> Conf Calculateurs -> UCEPresent_CF.Presence_ACC
[image]

2. While taking look at that, can you please look at: Tableu de bord -> Cluster_15-40_v3.30.02 -> Ecrans Configuration -> Conf Prestations VEH -> DW_CF? Is it set to the Radar or Camera option?
[image]

Good luck, and if you do enable the ACC successfully, be careful when using it. By default you have to be aware of the situation at all times as system may disengage itself and leave the reaction to the driver when it's not sure what to do or it has to brake more than 1/3 of the vehicle's braking capacity. It's by design.

Banana009, 2018-04-04:
Do you think it would be possible to activate ACC on Scenic IV? I have a radar because I have a DW. Unfortunately, I do not have a button on the steering wheel for ACC to set the distance of the vehicle in front.  I do not think it's necessary? 
I tried it in these units, but without success. Car says AEBS is not available:-(
[image]
[image]
[image]

tomidix, 2018-04-07:
Hi,
maybe you can. I think this is not the model dependent, but I think there still may be a hardware which has to be activated or is missing in your vehicle.. Regarding the distance button, I think it is not neccessary.
I would give a try with FCW setting in the second picture you've attached and see what it can enable..
[image]
It will invoke "Active braking" item in a driving assistance menu in the Rlink2 and maybe give you the possibility to activate the AEBS.. Be extra careful because you are actually messing with critical systems in a vehicle. Remember also that this is all at your responsibility.

games12345, 2018-04-09:
I have done everything you write. But everytime I start the car I get the error message Active braking off. I think there should be an other acc switch in other computers. I found one in the variable damper adjustment ecu.
[image]
Also I find in injection ecu EDC17C84_170B_780_7J0_830_8F0_A50_C00_V02_20170106T133042 this
[image]
But I dont find a way to write in injection ecu. I dont find the option I have marked.

Another question: Why I only have the marked buttons in data editor, not if I am connected to the car?
[image]

Hope you can help me.
Thank you

Banana009, 2018-04-09:
Try to press the start button but not the pedals!! After that you can read/write more ECUs

games12345, 2018-04-09:
I know how to activate ignition permanently.
I mean i dont find the Option in ddt4all

Banana009, 2018-04-09:
I think you have the opposite problem, you got a button on the steering wheel and you need to activate it, right? 
Did you change the steering wheel? What do you mean switch for ACC/Limiter?

games12345, 2018-04-09:
Yes I Need to activate button on the steering wheel.
I mean The Button under the Parking Break. I send Photos later

Banana009, 2018-04-09:
I know what do you mean, but this buton is the same tor ACC and normal Cruise Control. 
Do you have all cables for distance buton on the steering wheel? 
Do you have a picture of how the steering wheel looks with a button and without a button like?
It seems that activation of the ACC will be more complicated, because the same problem is with the distance button and no button.

games12345, 2018-04-09:
Sure i have all Cables. I replaced it because of heated steering wheel. I replaced the switches for wipers and lights/flasher too.

games12345, 2018-04-09:
Heating works very well

misncz, 2018-04-11:
@games12345 where did you get the steering wheel with distance button? I'm after the same setup as you are

games12345, 2018-04-11:
Hi i get it from a Renault dealer

misncz, 2018-04-11:
@games12345 I actually am about to achieve the same with the same car setup. Just wondered where to get the wheel with distance button. Actually wondering whether distance button is needed at all. I'll be looking to hear some good news from you :-)

misncz, 2018-04-11:
I tried to ask in our local Renault dealer. The technican was busy but I was told it should be possible because the car (in Initiale Paris) has everything required to run ACC. I go to visit them tomorrow again and try to ask him privately. He hopefully might say more during non-official discussion :-)

Banana009, 2018-04-11:
Maybe, ACC will need something else in the steering (brake)?? How would ACC be able to brake?

games12345, 2018-04-12:
Here is where i change the config to acc:
Frontal Radar
[image]

ESC
[image]

Radio (R-Link 2)
[image]

Cluster
[image]
[image]

Adaptive Damper
[image]

misncz, 2018-04-12:
Is it working for you then?

Banana009, 2018-04-12:
Did you RESET Cluster_15-40_v3.30.02 with button on the screen? That is important for this ECU. If it does not reset it, it does not work.

games12345, 2018-04-12:
No its not working. Yes i reset the cluster ecu but still not working

Banana009, 2018-04-12:
Did you write acc parameters to all ECUSs( without VADA)? Some ECUs allow write with the engine on, some off...

games12345, 2018-04-12:
Yes all acc parameters are write in all ecus with permanent ignition on.
Press start without press brake pedal to turn ignition on. Then put shifter on drive and activate parking brake. Again press and hold start button until the message hold start button

games12345, 2018-04-16:
Here some pictures of my steering wheel and the switch:
[image]

[image]

games12345, 2018-04-17:
I have no idea where to activate ACC with clip.

Banana009, 2018-05-11:
Here is tutorial, try it. https://www.youtube.com/watch?v=xmgBis5SFFo

Banana009, 2018-07-09:
Any impovement with ACC?

veroled, 2019-04-27:
Hi guys, did Anyone  solved the acc activation problem?

victor80gl, 2019-04-30:
Hi there !
 i am also busy . My talisman is without AEBS , but the ABS block have the same reference like the car with AEBS ( 476609440R ) so to activate is soft or calibration . I succeded to install onother soft from a car with AEBS04 but the configuration is not the same so i just configure automaticly . I search for more info . I have only 2 foult Active Brake desactivated and Function button RV:LV . I study also and the wiring diagram from Talisman(Without Radar ) and Espace(With radar and Acc) and i find only one difference ; to espace From activate button RV/LV it is one wiring more  to go in the UPC (+after contact ) pin 6 button to pin 3 green steker UPC ....

TonyChua28, 2019-05-07:
HI Banana009 will it be possible to share the tutorial on activating AEBS pls?

victor80gl, 2019-05-08:
Because i work in the Renault network i can use many software . Only with Clip . But i think the problem is not from the ABS software only from config. you can select with AEBS . The problem what i found it is in injection . Is the configuration from RV/LV .Configuration CCSL steering wheel push button . I read it to a car with ACC and is Whithout  and i read it to 10 cars without ACC and the configuration is With . So practically injection see another voltage how come from the button , because is one more button so more resistance ...
[image]

shrlnm, 2019-05-08:
Hi, is this voltage (4.6V) from wheel equipped with "Distance" button (5 buttons left module)?

shrlnm, 2019-05-08:
@victor80gl Could you please measure voltage while you pressing different buttons ( +, -, O, R and ACC)?

victor80gl, 2019-05-08:
The image is from a car with ACC . To the others car is also 4600mV . The voltage is the same . Computer work with 5V , But the difference is the output not the input . I think that with the change of the configuration is change of the level of the voltage .

shrlnm, 2019-05-08:
4600mV it is an output of resistive voltage divider inserted in wheel buttons module. We have already discussed it here https://github.com/cedricp/ddt4all/issues/126
This value is  available by LID 2039. I am using this parameter in my arduino projects to detect pushing of this buttons.  On wheel with 4 buttons the normal (unpressed) value is about 4000mV
'R' ~3V
'+' ~2V
'-' ~1V
'O' ~0V

It is interesting which values are on the 5-buttons module

victor80gl, 2019-05-09:
Here the value with ACC 
R 3.2V
O 0.19V
- 0.933V 
+ 2.5V 
D 1.7V (Distance button )

victor80gl, 2019-05-09:
Value from Clip
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Banana009, 2019-05-09:
The voltage value is different.  When no button is pressed  voltage is 4V, with ACC (distance button) is  4,6V. 
Is necessary to replace the steering wheel (or at least PCB). Then we can activate ACC :-)

victor80gl, 2019-05-09:
hi there

The value are from a car whit Acc from factury . so the valua are real . If
you measure the value without motor runing is 4 v ... but i deed with a
charger and maybe from there are a litle bit diference . The problem is
there in the config from injection .

shrlnm, 2019-05-09:
I suppose that the input voltage on the divider is after voltage regulator.  On my koleos it does not depend  on alternator output significantly. 
We need to define if ECM calibration different on cars with ACC and without. If calibrations ref numbers are the same then ECM detect  ACC presence  by measuring voltage output of these buttons.

victor80gl, 2019-05-10:
Yes is true , but is not congigured with Acc so for ECM is one error

veroled, 2019-05-17:
Hi guys, someone activated LKA and DW on Talisman?
On DDT4ALL, in FCAM there is only FrontCamera and nothing, why not compare XFA RACAM?

victor80gl, 2019-05-22:
guys i deed it ... everything is in the injection

victor80gl, 2019-05-22:
> @victor80gl <https://github.com/victor80gl> did you reflash injection ecu?

yes

victor80gl, 2019-05-22:
reprog ABS (for AEBS ) 
config dashboard with ACC
config radio with ACC 
reprog injection for the 5 button from steerwheel 
config Radar

Banana009, 2019-05-22:
What do you mean exactly?

TonyChua28, 2019-05-22:
Victor probably can you share the tutorial here so we can try on my scenic iv as well?

Banana009, 2019-05-23:
I have also SCENIC IV. I suppose it will be easier, we have RACAM and AEBS as a standard :-)

scoobydoo511, 2019-07-02:
Hi guys...
Im confused by reading the hole comments and i still dont know how to activate ACC on Renault Talisman or Scenic.
Actually, i dont know if there is any solution to acivate the ACC...
Any help would be appreciated...

victor80gl, 2019-07-02:
you must to reprog injection . There is the function from cruise control.

scoobydoo511, 2019-07-02:
Hi victor80gl
thx for your quick replay.
I understand everythinh you have write but not how to reprogram injection from steerwheel.
Cold you please explain to me how to do it please?
And, i have Scenic IV, i already have activated LKA and DW and everything is working well.
Thx for your help

victor80gl, 2019-07-02:
this is the big chalange ... i deed it with ddt but with the help from a
guy who werk there . Bicause in the last year the ECU must to be unlock to
cold be rited . Whit Clip is the same ... so i don t know how you cand
reprog the ECU . Really i don t have a ideea .

victor80gl, 2019-07-02:
sorry ... there to Renault ... The unlocking can be able whit a token

mar., 2 iul. 2019, 20:41 Victor Tanasescu <victor80gl@gmail.com> a scris:

scoobydoo511, 2019-07-02:
There is always a problem :))))
I will try to do something by my self.
Thx for your help victor80gl.

victor80gl, 2019-07-02:
succes ... I have the oportunity to take every soft from Renault ... the
question is how it is possible to reprog ???

scoobydoo511, 2019-07-02:
If you get the answer on how to reprogram, please call me ;))))))

victor80gl, 2019-07-02:
chip tuning ... :)
It is always a way . The guys from chip tunning they are able to rite the
program ...soo!!!

TonyChua28, 2019-07-02:
Hi Victor, will it be possible to take a screenshot of the page where the injection is to be reprogram. Much appreciated

scoobydoo511, 2019-07-05:
I agree with TonyChua28.
That could help us to solve the problem of injection ECU.
Thx :)

[272 more replies]

closed 2023-06-02 by Furtif