cedricp/ddt4all #48: error

adicretu97, 2018-04-01:
i use ddt4all with Vgate icar3 WIFI and i have this red error's, i open some ECU's but i can modify them or make something, all options is with red pls help me
[image]

KarelSvo, 2018-04-02:
RED in your photo because your exact ECU software version is not listed in the ECU file.

adicretu97, 2018-04-02:
Aha, thx
And what can i do? Looking for another database with more ecu's?

KarelSvo, 2018-04-02:
Your ECUs are yes in the database.
No one hundred percent match is no problem, just a beauty bug.

cedricp, 2018-04-02:
Yes, it looks like your version is very old, try updating it and retry. Some bugs have been fixed for KWP2000 protocol handling.
Red lines with **no relevant ecufile found**  mean that there's no XML file to support the found ECU.
Those with **no perfect match** mean that XML file is found but not with the exact soft version.

KarelSvo, 2018-04-02:
I would say that all relevant ECUs are available in the database.
Only the maintenance of the data status of the software versions is not particularly good.

I had to complete the software version of two ECUs of my vehicle, otherwise these will not be found during automatic scanning.

I would not overrate this data comparison.

cedricp, 2018-04-02:
Interested to get that patch, would you share it ?

KarelSvo, 2018-04-02:
I have only opened the ECU xml in the text editor and updated with my actual ECU data.

<AutoIdent DiagVersion=" " Supplier="  " Soft="    " Version="    "/>

adicretu97, 2018-04-02:
I will try tomorrow with kkl interface 409.1 with usb
Thx for answers

cedricp, 2018-04-02:
Not sure it'll work with _kkl interface 409.1_, it's not ELM327 compatible...
@KarelSvo : Ok, I understand

adicretu97, 2018-04-02:
I need elm with a specific chipset code no?

closed 2018-04-16 by cedricp