cedricp/ddt4all #52: Radio  Numérique "DR" - MEGANE 4

S4CABB, 2018-04-16:
Dans très peu de temps, la DAB (Radio Numérique Terrestre) arrive en France. 
Si vous n'avais pas l'option "DR" dans la sélection des bandes de la radio, 
c'est que l'option n'est pas validée dans les paramètres de l'ECU. 

Comme déjà montré plus haut, voici pour ma part le printscreen de l'option validée par l'ECU "RadNav 2.3". 

Cela fonctionne immédiatement... et la qualité du son est remarquable. 

Bon déblocage. 
[image]

lucamagilla, 2018-04-16:
J'ai un Italian Scenic 4 et pas assez pour activer l'option dans Radnav mais je devrais l'activer dans A-IVI 09 mais malheureusement je ne peux pas lire

S4CABB, 2018-04-17:
> Message du 16/04/18 23:37
> A : "cedricp/ddt4all"
> Copie à : "S4CABB" , "Author"

Bonjour Luca,
 
Moi aussi A-IVI 09 ne passe pas dans la MEGANE, 
c'est pour cela que j'ai utilisé RadNav 2.3
 
Je viens de vérifier pour le scénic IV, tu as la possibilité de le faire.
 
Ci-joint le printScreen...
N'oublies pas de passer en Mode "After Sales"
 
Salutations
 
Sylvain

lucamagilla, 2018-04-17:
Bonjour Sylvain A partir de radnav 2.3 ou 2.4 la dab est activée mais il n'y a pas d'affichage sur r-link de la bande DR
Ciao Luca

S4CABB, 2018-04-17:
Luca ,

As tu redémarré complètement le RLINK, (En appuyant plus de 3s sur le bouton) , j'ai déjà lu qu'il est nécessaire de le rebooter pour que le RLINK reprenne les options validées.
Ne pas avoir d'inquiètude, mais il faut presque 1mn pour que la tablette redémarre...

Cdt

lucamagilla, 2018-04-17:
Redémarré par le bouton et d'abord par la touche reset de Ddt4all

S4CABB, 2018-04-17:
Reset de DDT4ALL ? quel reset ?

lucamagilla, 2018-04-17:
Touchez sur le bouton de "Key on off reset " en haut à gauche

Furtif, 2023-06-02:
> Messieurs, Vous devez d'abord connaître toutes les caractéristiques de votre voiture en vérifiant le numéro de châssis de votre véhicule VIN sur infotech.renault.com. (site officiel de renault et vous pouvez acheter un accès encore cher ou vous pouvez demander des services officiels de renault, les jetons usb de cotech ont un accès gratuit.Une autre option pour en savoir plus sur votre voiture renault est http://carinfo.kiev.ua/ Si vous pouvez entrer votre numéro de vin, alors vous pouvez obtenir des informations très détaillées sur votre voiture renault.Je veux dire d'abord, nous devons apprendre des informations détaillées sur nos voitures alors nous devrions commencer à jouer avec ddt4all pour activer les fonctionnalités cachées ... Essayez de votre mieux :) Par exemple, comme je peux voir plus de 6 différents types de radio utilisés comme rad66, rad68, rad82 etc. dépend du modèle et / ou de la date de production de la voiture ... PS: Cedric vous pouvez poster cette info comme votre phrase à l'explication du logiciel ddt4all si vous aimez ..
> 
> Gentlemen, First need to know all the features of your car by checking your vehicle's chassis number VIN at infotech.renault.com. (official renault web site and you can buy acces even expensive. or you can request from renault official services, cotech's have usb token for free access. Another option to learn more about your renault car is http://carinfo.kiev.ua/cars/vin/renault/ . If you can enter your vin number then you can get very detailed information about your renault car.
> 
> I mean first we have to learn detailed info about our cars then we should start to play with ddt4all to activate hidden features... Try your best :)
> 
> For example as i can see more than 6 different types of radio used like rad66, rad68, rad82 etc depends model and/or production date of car...
> 
> PS: Cedric you can post this info as your sentence to explanation of ddt4all software if you like..

cedric is bussy but uses https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif