cedricp/ddt4all #53: Avaible options for a Clio 4 of 2015

nasskach, 2018-04-16:
Hi,

I'm wondering what are interesting features that could be enabled on my Clio 5 Initiale Paris 2015. I already enabled Android Auto.

Thanks !

blade909, 2018-04-16:
clio 5??sure you want say clio IV
tried econometer light in dashboard and hour
some pals with medianav evo tried rear sensors displayed on it

nasskach, 2018-04-16:
@blade909 Yes sorry, it is a clio 4 of course.

man0s, 2021-06-27:
@blade909 Did they managed to enable rear sensor display? If yes, do we have a tutorial/setting's name to enable it?

blade909, 2021-06-28:
only way to enable rear sensor display is in micom settings on medianav,
you have to enter micom manager and enable HMI button on second line on menu,
depends of medianav version of course

man0s, 2021-06-28:
Thanks for your response @blade909 👍

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/834

closed 2023-06-02 by Furtif