cedricp/ddt4all #55: bad elm response wrong response cfc0 unknown

mrhappychappy83, 2018-05-08:
Hi there, I too am trying to activate Android Auto in my Renault Megane III with R-Link 1.  I have followed all steps, made the modified cable, and my OBD scanner seems to be communicating with the ECU but not properly.  When I try to write the new Navigation settings, I get the error in DDT4All which reads: "bad elm response wrong response cfc0 unknown".

I have no idea what this means or what could be causing the problem. Can anyone help? Attached are the log files also. Appreciate your time in advance.

[elm_ddt_log.zip](https://github.com/cedricp/ddt4all/files/1983754/elm_ddt_log.zip)

mrhappychappy83, 2018-05-08:
BTW, my R-Link is updated already to the latest version

cedricp, 2018-05-08:
It looks like a bad ELM327 clone... Did you try to start a benchmark on the first window ? If it ends up with "**incompatible ARM core**", you'll need to get a _real_ ELM with PIC18F25K80 chip.

mrhappychappy83, 2018-05-08:
Thanks for the fast reply, Cedric!I ran the benchmark test blue and got 0 of 95 successful and the incompatible ARM core message :(
Just to confirm, when I run this test, is ok if the cable is not connected to the car? I only connected to the computer by USB.

cedricp, 2018-05-08:
Yes, you can do the test disconnected from the car. Your interface is definitely not the good one.

mrhappychappy83, 2018-05-08:
OK that's a shame but thank you for helping me work this out. I will buy one with the proper chip and try again.

fd007s, 2018-07-25:
I also have a clone chip " incompatible ARM core message 90/95" . I managed to resolve "bad elm response wrong response cfc0" by changing the driver.
Now i have a new problem: I have a red column with " No data" after i enter on dt4all- megane 3 -navigation-...etc. I can't flash anything.
I don't know what seems to be the problem: wrong chip, wrong wiring.
So i made a test: i connected the elm device to laptop but not to car.
i had the same result "No data"
Is there a way to check the cable or the connection between elm device and car?
Sorry for my bad english.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/834

closed 2023-06-02 by Furtif