cedricp/ddt4all #56: where find new ecu.zip file?

igno77, 2018-05-08:
Hi to all,
I use this fantastic software from a long time with my new Megane 4 Bose with a great success! Sometimes I've few problem in some section but nothing so bad. From 2 months I've also a new Captur Bose with R-Link evolution (android auto) and seems so similar to a Clio system. The problem is that I want to try to change the front flasher (in the Bose is a led bar but in the Inital Paris is a sequantial led bar) and the two headlight are the same with the same serial number! i think that I can do it BUT....the last ecu.zip file is from mid of 2017 so too "old" for this model and not all the calculator are visible or accesible!
Do you know where i can find a new ecu file list?? Thanks a lot

KarelSvo, 2018-05-26:
https://www.drive2.ru/b/498093336985338243/
with ecus until march 2018
DDThas a lead of about one year for series production. The automatic ECU identification by manufacturer and software version therefore does not work very well.
Better is, you know which ecus are built into your vehicle and connect manually to the latest ddt ecu version.

jirikeller, 2019-05-06:
Likely very similar qustion, I am trying to solve this error:

Found ECU [Power steering] (no relevant ECU file found) :DIAGVERSION [17] SUPPLIER [231] SOFT [0010] VERSION [0000]
Found ECU [EPS] (no relevant ECU file found) :DIAGVERSION [17] SUPPLIER [231] SOFT [0010] VERSION [0000]
Found ECU [DCM Renault] (no relevant ECU file found) :DIAGVERSION [0F] SUPPLIER [ITL] SOFT [47] VERSION [17]

Is there any fresher ecu.zip file ? Or can be the filenames (namely EPS one) somehow find out ?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/834

closed 2023-06-02 by Furtif