cedricp/ddt4all #57: General upgrade forum for Scenic IV

tomka1118, 2018-05-09:
Hi there,

I'm new here as an owner of a Grand Scenic IV since a month.

I've got my "original" ELM327, with the original ELM chip, i hope it will work fine. Today can I try it.

So, I think that it would be great, if we could make a thread to share all the available and tested hidden/deactivated functions for this type.

1. The phone mirroring function (I found the description somewhere else)

2. The LKA fucnction (aka Banana009)

3. I'm missing the Loudness setting in my Bose, don't know they are excluding each other?

But if this descriptions exist somewhere else please share with me.

Wbr,
Tom

tomka1118, 2018-05-10:
LKA doesn't work, all seems good, but nothing.
Phone mirroring also looks bad.

Somebody tried to write the HMI part of the ECU? There are many useful settings, like video in motion or Rsound. These are restricted for mer at the moment. Should I change the settings and write them back?

payamt20, 2018-05-28:
How I can active video in motion for Talisman?

tomka1118, 2018-05-28:
First of all you have to have the initial package:

   - a good ELM (USB or Wifi),
   - this program (DDT4All) and the corresponding database (ecus.txt)

When all you have, you should test your ELM compatibility, because a bad/fake ELM can destroy your cars ECU and all others...

If you don't know what do you want to change, first learn to use the DDT4all in the "Edit mode".

Are you ready with these steps?

closed 2023-06-02 by Furtif