cedricp/ddt4all #63: Windows Multilanguage installer

bovirus, 2018-05-14:
I posted a pull-up request to add to Windows installer multi.-language option.

I addedc all languages available as interface as language for the instalelr.

Please take care that also we need to add a custom message for the operation after installation about database.

Source: "DDT4ALL.BAT"; DestDir: "{app}"; AfterInstall: AfterMyProgInstall('**_Do not forget to install database to_** ', ExpandConstant('{app}'))

I will do it a in a next step.

I'm the official translator for Inno Setup.

The updated Italian language for Inno Setup is here

[InnoSetup-5.5.3-Language-Italian-2018.05.03-ANSI.zip](https://github.com/cedricp/ddt4all/files/2000274/InnoSetup-5.5.3-Language-Italian-2018.05.03-ANSI.zip)

Please unpack the .ISL file and overwrite the same file in Languages sub-folder of Inno Setup directory.
Thanks.

bovirus, 2018-06-12:
@cedricp 

Could you please rebuild Windows installer/portable version based on latest code fixes?
There are lot of changes vs. latest installer/portable version.

cedricp, 2018-06-12:
I'd like to be sure that all is working properly before releasing it, and I currently don't have a Windows install to do it.

bovirus, 2018-06-12:
Today to have for Windows a working DDT4ALL and updated version the user should have to do 

- Install installer version
- Download, unzip and overwrite the fiels in the installed folder.

I believe that you can build a beta version of teh installer with all changes to enable the user to test it in place and send you feedback about it.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/834

closed 2023-06-02 by Furtif