cedricp/ddt4all #76: someone had tryed ELS27 with Megane 4?

igno77, 2018-05-23:
Hi to all,
I've seen in last version of ddt4all the new icon of ELS27. After a quick reserch of information I know that this interface is better than ELM (faster, stable and use Mcan and Lcan at the same time) but....we need to modify some wiring for use it with a Megane 4? It seems this interface born for Ford and Mazda cars, not for Renault. On a russian website there is a modification of the els27 but no one have had success with the connection. Do you have more info? What kind of chip we must have insiede?
Thanks

KarelSvo, 2018-05-26:
You only must to modify the ELS27 if you want to use the media can bus. 
https://a.d-cd.net/40c326ds-960.jpg
Simply get yourself an experience with els27 from aliexpress from 13 euro.
Please your note, obdlink interface and els are preset to baudrate 115200. If firmware is not up-to-date then just update to scantool.net.
I must to use windows 10 for update  ELS.

pakoto, 2018-05-26:
Check photo. Description says 3-12 11-13: wrong, and photo shows 3-13, 12-14: wrong because you connect canbus and mm canbus into the car obd, dangerous!).

premultiply, 2018-05-26:
No. "Normal" CAN-Bus is on 6+14 (unmodified).
Only the second independant CAN-Interface needs to be briged to Renault MM-CAN on 12+13. Ford-Pins 3+11 does not have any connection on Renault vehicle OBD port so there is no need to remove or disconnect them on ELS27. Just wire these two connectors and you can use it on Ford and Renault vehicles.

pakoto, 2018-05-26:
Sorry, i mean that zmiterm bridge 6-13 and 12-14 and use out the car. See green red cables in the photo and compare position with generic OBD pinout photo.

ohayyam, 2018-05-26:
I tried many combinations but no success. Even i disconnect obd socket side and main board connections and made many more combinations according to ford and renault pin out diagrams but all fail. I never read succesfull story. does really anyone use els27 with any renault ? I already asked to zmiterm on drive2.ru no answer. I believe els27 may fake ??? I contact russian seller of original els27 he said 75 usd but he did not know about cable modification for renault. He said original els27 will work with renault. I am not sure and waiting to read success story with els27 and renault.

ohayyam, 2018-05-26:
I already mail latest ddt4all v.3.4.5 program with mar2018 database to cedric to check ;) download link on drive2.ru site by zmiterm. 

My els27 may damage during combination try :)) never connected or it was fake because i tried it without cable mod. It did not work for me.

premultiply, 2018-05-26:
http://ffclub.ru/topic/311647/

premultiply, 2018-06-12:
@cedricp This looks like the protocol description to change the CAN interface for MM-CAN access:
https://www.scantool.net/downloads/118/mx_protocol_commands.pdf
You have to select a "MS-CAN" preset first an then adjust the baud rate to 500000.

cedricp, 2018-06-12:
I saw that, but I don't have the hardware to test it yet.

premultiply, 2018-06-12:
I have one here. How can I assist you?

premultiply, 2018-06-12:
First try run (without connected to car X10 ZOE) with els27 branch:
```
Traceback (most recent call last):
  File ".\ddt4all.py", line 514, in changecanspeed
    self.paramview.setCanLine(item)
  File "C:\Program Files (x86)\ddt4all\parameters.py", line 534, in setCanLine
    self.initELM()
  File "C:\Program Files (x86)\ddt4all\parameters.py", line 537, in initELM
    connection_status = self.ecurequestsparser.connect_to_hardware(self.canline)
  File "C:\Program Files (x86)\ddt4all\ecu.py", line 1015, in connect_to_hardware
    options.elm.set_can_addr(short_addr, ecu_conf, canline)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1170, in set_can_addr
    self.lf.write('#' * 60 + "\n#connect to: " + ecu['ecuname'] + " Addr:" + addr + "\n" + '#' * 60
+ "\n")
TypeError: cannot concatenate 'str' and 'NoneType' objects
Scanning ECU FRRAD
Scanning ECU INV-ME_29b
Scanning ECU Aide au parking
Warning, address DB is not mapped
Warning, address DC is not mapped
Skipping CAN extended address (not supported yet)  DA
Scanning ECU WCGS
Scanning ECU FCAM
Scanning ECU BCB
Scanning ECU PEB
Scanning ECU CAN Adapter / BAC
Scanning ECU VSP
Scanning ECU AAU
Scanning ECU IDM
Scanning ECU UCH
Scanning ECU ABS-VDC - ABS-ESP
Scanning ECU Direction assiste
Scanning ECU HVAC
Scanning ECU LBC2
Scanning ECU HFM
Scanning ECU E-ACT-EBA_29b
Scanning ECU BCB-OBC_29b
Scanning ECU DCM Renault
Scanning ECU UBP
Scanning ECU UPC-EMM
Scanning ECU Frein de parking lectrique
Scanning ECU Airbag-SRS
Scanning ECU LBC (EV)
Scanning ECU (H) EVC
Scanning ECU PLC/PLGW
Skipping CAN extended address (not supported yet)  2D
Scanning ECU Audio
Scanning ECU MIU
Scanning ECU Navigation-UCC-ITM
Scanning ECU TDB
Skipping CAN extended address (not supported yet)  D2
```

cedricp, 2018-06-12:
This error is not related to **els27** work, which ECU are you trying to connect ?

cedricp, 2018-06-12:
Just commited a fix for this kind of problem.

premultiply, 2018-06-12:
Ok, did a test on my car (ZOE) now but switching to "MediumSpeed CAN" does not change anything.
All communcation stays on HS-CAN.

Switched to "MediumSpeed CAN" after startup, selected my vehicle from list and pressed scan.
All ECUs from HS-CAN where found but nothing from MM-CAN (Navigation/MFD, ...) was found.

```
Verwendung einer benutzerdefinierten Datenbank
libpng warning: iCCP: known incorrect sRGB profile
Starten des ELM mit einer Baudrate von 500000...
Versuche Anschluss zu oeffnenCOM17 @ 500000
Checking els
Test der Baudrate: 500000
Starten mit Baudrate: 500000
2933 ECUs wurden geladen.
Scanning ECU FRRAD
Scanning ECU INV-ME_29b
Scanning ECU Aide au parking
Warning, address DB is not mapped
Warning, address DC is not mapped
Skipping CAN extended address (not supported yet)  DA
Scanning ECU WCGS
Scanning ECU FCAM
Scanning ECU BCB
Scanning ECU PEB
Scanning ECU CAN Adapter / BAC
Scanning ECU VSP
Scanning ECU AAU
Scanning ECU IDM
Scanning ECU UCH
Scanning ECU ABS-VDC - ABS-ESP
Scanning ECU Direction assiste
Scanning ECU HVAC
Scanning ECU LBC2
Scanning ECU HFM
Scanning ECU E-ACT-EBA_29b
Scanning ECU BCB-OBC_29b
Scanning ECU DCM Renault
Scanning ECU UBP
Scanning ECU UPC-EMM
Scanning ECU Frein de parking lectrique
Scanning ECU Airbag-SRS
Scanning ECU LBC (EV)
Scanning ECU (H) EVC
Scanning ECU PLC/PLGW
Skipping CAN extended address (not supported yet)  2D
Scanning ECU Audio
Scanning ECU MIU
Scanning ECU Navigation-UCC-ITM
Scanning ECU TDB
Skipping CAN extended address (not supported yet)  D2
```

cedricp, 2018-06-12:
Did you rewire the MS-CAN lines on the OBD connector ? AFAIK, it's not connected to the good pins.

cedricp, 2018-06-12:
Just fixed an issue with MS-CAN scan, sorry, can you retry ?

premultiply, 2018-06-12:
Yes of course it's rewired.
Yep, I will try again in a few minutes.

premultiply, 2018-06-12:
Sorry, still only ECUs from HS-CAN.
And manual MFD selection still does not get any data.

premultiply, 2018-06-12:
There are some more command descriptions in https://www.scantool.net/scantool/downloads/234/stn1100-frpm-preliminary.pdf which are removed from later versions of the documentation.

cedricp, 2018-06-12:
That's weird, can you send me your _elm log_ over ?

cedricp, 2018-06-12:
You can also try to add a line at the end :
**elm.py 1192** 
```
        if canline == 0:
            if 'brp' in ecu.keys() and ecu['brp'] == "1":  # I suppose that brp=1 means 250kBps CAN
                self.cmd("AT SP 8")
            else:
                self.cmd("AT SP 6")
        else:
            self.cmd("STP 53")
            self.cmd("STPBR 250000") # this line
```

premultiply, 2018-06-12:
No problem. Sent you the log by mail.
But there is no STP command inside at all :-(

premultiply, 2018-06-12:
Mmmmh. I downloaded the changed files only. Maybe I've forgotten something...
I will try with full els27.zip again.

premultiply, 2018-06-12:
Dry-run els27.zip without car connected still does show any STP commands in log.

cedricp, 2018-06-13:
That is really strange... I checked the code several times...

cedricp, 2018-06-13:
Ok, just found and fixed the problem... You can retry if you have time

premultiply, 2018-06-13:
After several hours of testing I think all ST commands are sent out as raw data because they are filtered
https://github.com/cedricp/ddt4all/blob/7b28b65f613559ed1780ed2eb60af8d0cb7f244b/elm.py#L737 so the never reach the ELS27. This is why they never show up in any log...

cedricp, 2018-06-14:
Well spotted ! I fixed that issue

pakoto, 2018-06-18:
works multimedia canbus using bridges or still pending tests?

cedricp, 2018-06-18:
Still waiting for last commit testing ... (_els27_ branch)

premultiply, 2018-06-18:
Software side seems to be ok now.
But using the modified els27 on the car disturbed and crashed my radio ECU on MM-CAN. Had to pull the fuse to make it work again.

I am not able to verify that the modified hardware does what it should do.

pakoto, 2018-06-18:
ford switch canhigh 6-3 and canlow 14-11: 3 and 11 are mm canbus.
renault switch CH 6-13 and CL 14-12: 13 and 12 are mm canbus.

For conversion i think you need to connect mm CH 3-13 and mm CL 11-12. Other combination connect mm canhigh with mm canlow, or worse shortcircuit canbus with mm canbus.

IMHO

premultiply, 2018-06-18:
Yes it seems you are right. The polarity is wrong.
I have to change it on my ELS27.

pakoto, 2018-06-18:
Remember that obd pinout front view images from internet are mirrored with els27 front view.

I check with multimeter pins using this ELS27 front view 

\   .. 8  ..  7 .   6  .  5  .  4  .  **3** . 2 .  1  .  /
.  \ 16 15 **14 13** 12 .**11**. 10  .9 /

premultiply, 2018-06-18:
The pin numbers are always the same for rear or front view  ;-)

pakoto, 2018-06-18:
@premultiply i need your help with off-topic  if you are creator of ZOEdisplay:
We are trying to add DPF soot level into dacia-renault DCIs using arduino. ddt2000 database obtain it using:

Send 22242C
Receive 62242C0000
Start byte: 4
Bits: 16
Unit: gr
Value: (AX+B)/C
A=1
B=0
C=100

Do you know how to translate to arduino code using mcp2515 module? need to ask every x secs to ecu?

premultiply, 2018-06-18:
@pakoto I don't know how to poll data from ECUs. Never thought about that. All I need to do for ZOEdisplay is listen to the bus.

pakoto, 2018-06-18:
i don´t know how it works. When we connect to motor ecu using ddt4all, we can get too much sensor info than using stardard tools like android torque. We are trying to take some parameters and send to arduino lcd.

cedricp, 2018-06-18:
It's because each ECM (engine control module) CAN ISO_TP frames are differents. This is why you need an engineer tool like DDT4all to get them.
Select an ECU and look at _request_ and _data_ tabs in **edition** mode to know how frames are built.

cedricp, 2018-06-18:
I started a project for Arduino UNO with MCP2515 here : 
https://gitlab.com/DrQuark/x84_can
There are ISO_TP and CAN_RAW sources examples.

cedricp, 2018-06-18:
Send 22242C
Receive 62242C0000
Start byte: 4
Bits: 16
Unit: gr

as the wanted value starts at byte 4 (index 3) and we have 16 bytes (2 bytes) :

receiv_buffer = {0x62, 0x24, 0x2C, 00, 00};
short my_value = (receiv_buffer[3] << 8) | receiv_buffer[4]);

should be good

premultiply, 2018-06-18:
New modification 3->13 and 11->12. Only one wire needed ;-)
[image]
[image]
[image]

pakoto, 2018-06-18:
Yes. I have same mod. Works?

premultiply, 2018-06-18:
Yes, hardware does work now!
But the ddt4all part is not working correctly.

First problem was: MM-CAN on my ZOE car is 500 kbd not 250 kbd. I fixed it manually in code ("ST PBR 500000")
2nd unsolved problem: But even with that modification only parts of the ECU data is retrieved by ELS27 compared to an modified ELM327 on MM-CAN. But my ELS27 STN117 has FW 3.3.1 and I am not able to update it to lastest FW 4.2.4 (https://www.scantool.net/downloads/updates/stn1170/).

pakoto, 2018-06-18:
i can´t update my fw either. Aliexpress ELS27 clone i suppose.

I´ll test with my dacia dokker connecting to medianav and post results.

pakoto, 2018-06-19:
minor issue: elm benchmark don´t work with this branch.

pakoto, 2018-06-19:
Test elm fails
Connect works

remember, minor issue. els27 don´t need elm check because it works.

[image]
[image]

premultiply, 2018-06-19:
And by the way the serial port speed selection dropdown for ELS27 is useless. ddt4all does only auto detection. But the autodetection should run bitrates 2000000, 1000000, 500000, 115200, 38400.

The ELM test seems to try a wrong port etc. Same result for me.

cedricp, 2018-06-19:
Ok, I see

pakoto, 2018-06-19:
yes, port selection could get some improvements:
-bluetooth, usb, and obdlink selection could be one. Works selecting COMx and usb icon. Tested.
-option to remember expert mode and check recommendations. You could add a big scary message first time and a "don´t show again" checkbox.

cedricp, 2018-06-19:
```
yes, port selection could get some improvements:
-bluetooth, usb, and obdlink selection could be one. Works selecting COMx and usb icon. Tested.
```
Ok
```
-option to remember expert mode and check recommendations. You could add a big scary message first time and a "don´t show again" checkbox.
```
Not sure it's a good idea, I already had bad surprises when working on the first version when expert mode was not yet in place...

premultiply, 2018-06-19:
Here is some log extract from ELS27 session tests to MFD on ZOEs MM-CAN. You can see that some commands do not get an answer. Maybe the specific AT commands needs to be sent AFTER MS-CAN preset call?

```
>[19:19:51.504]ATZ
<[0.874]ATZ

ELM327 v1.3a

>
############################################################
#[06/18/18 19:20:24.632] Init CAN
############################################################
>[19:20:24.632]AT WS
<[0.031]AT WS

ELM327 v1.3a

>
>[19:20:24.663]AT E1
<[0.015]AT E1
OK

>
>[19:20:24.678]AT S0
<[0.016]AT S0
OK

>
>[19:20:24.694]AT H0
<[0.016]AT H0
OK

>
>[19:20:24.710]AT L0
<[0.015]AT L0
OK

>
>[19:20:24.725]AT AL
<[0.016]AT AL
OK

>
>[19:20:24.741]AT CAF0
<[0.015]AT CAF0
OK

>
############################################################
#connect to: MFD v4.9 Addr:58
############################################################
>[19:20:24.756]AT SH 747
<[0.016]AT SH 747
OK

>
>[19:20:24.772]AT CRA 767
<[0.016]AT CRA 767
OK

>
>[19:20:24.788]AT FC SH 747
<[0.015]AT FC SH 747
OK

>
>[19:20:24.803]AT FC SD 30 00 00
<[0.016]AT FC SD 30 00 00
OK

[... 597 lines cut ...]
<[0.047]023E01
017EFFFFFFFFFFFF

>
>[19:23:01.880]023E01
<[0.062]023E01
017EFFFFFFFFFFFF

>
>[19:23:03.393]023E01
<[0.047]023E01
017EFFFFFFFFFFFF

>
>[19:23:04.906]023E01
<[0.063]023E01
017EFFFFFFFFFFFF

>
>[19:23:06.420]023E01
<[0.062]023E01
017EFFFFFFFFFFFF

>
```

cedricp, 2018-06-19:
There is communication between the OBD tool and the ECU
- the start diagnostic session is accepted
- Tester present accepted too
- This frame too : 
\< 03-**226C15**
\> 04-**626C1506FFFFFF**

premultiply, 2018-06-19:
Yes it is but the result in screen is incomplete (no data).
If the same is run on modified ELM327 dongle everything is complete.

cedricp, 2018-06-19:
Have you tried to connect to another ECU on CAN line 2 ? Maybe a firmware issue on the ELS side

premultiply, 2018-06-19:
Yes, its the same (bad) result with other known ECUs like Radio.
Unfortunately, I am not able to update the firmware.

pakoto, 2018-06-26:
Now els27 branch is master?

premultiply, 2018-06-26:
Yes, looks like Cedric has merged them.

@cedricp I think https://github.com/cedricp/ddt4all/blob/e0cf3c4d63d23ac664196888205a84a6990795a1/elm.py#L692 also needs ST-Handling, isn't it?

pakoto, 2018-07-01:
Tested with my dacia Dokker and mod els27. Canbus ecus works ok. Medianav ECU dont detect using scan. 

We need yo know how forscan works.

pakoto, 2018-07-03:
Tested again with last version. Canbus 2 @500k works ok changing options into dacia medianav, but... when i select canbus 2 @500k and push scan ecus, no ecus found. After scan medianav shows NO DATA all options. Need to change canbus 2 500k to other option, change again to canbus 2 500k and push refresh.

[12 more replies]

closed 2023-06-02 by Furtif