cedricp/ddt4all #82: dddt4all.py - Localization issue

bovirus, 2018-06-27:
@cedricp 

Hi.

Creating the master language (**_ddt4all.pot_**) with **_pot_create.bat_** you can get an error

..**_\ddt4all.py:509_**: warning: 'msgid' format string with unnamed arguments cannot be properly localized:
                            The translator cannot reorder the arguments.
                            Please consider using a format string with named arguments,
                            and a mapping instead of a tuple for the arguments.

The problem is in **_ddt4all.py_** line **_509_**

                print _("Cannot load plugin %s, %s") % (plugin, traceback.format_exc())

Thanks.

bovirus, 2018-08-02:
@cedricp 

Is there any way to fix this problem on localization?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/discussions/833

closed 2023-06-02 by Furtif