cedricp/ddt4all #83: bad ELM respons

tomasvs, 2018-07-03:
After changing something in my EMM_lightning ecu of my Clio IV, I can't write anything in these parameters. I keep getting the notice: BAD SLM Response : WRONG RESPONSE : Unknown.

What should be the issue? by now my lights aren't working and I can't get them back to factory settings.

Please help.. 
[logs.zip](https://github.com/cedricp/ddt4all/files/2160259/logs.zip)

cedricp, 2018-07-03:
What is the command you send. I found some "wrong responses" but for the request "3B62" (TPMS WHEEL ID 2 write command)
Screenshot ?

cedricp, 2018-07-03:
Sorry I was on the BCM file, found on EMM one. Are you sure the file you're using is the good one ? Did the scan worked ?

cedricp, 2018-07-03:
I think you used a bad file for this ECU : 
Found ECU [UPC-EMM] (**not perfect match**) :EMM_EDISON_DDT2000_SW10_V2 DIAGVERSION [4] SUPPLIER [414] SOFT [C100] VERSION [**9500 instead 1008**]
You should not try to write with a file that doesn't match perfectly

tomasvs, 2018-07-03:
I think thats the issue. How can I fix this? Should the dealer update the ecu?

cedricp, 2018-07-03:
You need to either find the correct file to put the correct parameters or ask the dealer to put the parameters back.
You can try to get an ECU from a broken car too.

tomasvs, 2018-07-03:
Every dealer can do that?

cedricp, 2018-07-05:
Don't know...

cedricp, 2018-07-09:
I rebuilt your original frame from parameter reading. Open a screen with the file "EMM_EDISON_DDT2000_SW10_V2" and click on the icon "Ox".
-> set diagnostic session to 10 C0
-> paste this line to input : 
3B63015E00190AFC00C8000000000000000000000AFC0AFC0AFC07310A120AFC0AFC0AFC03E800000001000000030AFC0AFC0AFC0000000000002EE00AFC0AFC003C001E00C80A0BB8003200
then type "enter" to send the command. That should bring back your original parameters.

ClioIV, 2020-05-16:
Bonjour, avez-vous trouvé la solution ? J’ai le même problème que vous

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/discussions/833

closed 2023-06-02 by Furtif