cedricp/ddt4all #84: 0 ecus loaded  on the database

MLPMLPMLP, 2018-07-06:
Hello,
Reading the already discussed topics  I found that the "0 loaded ECUs on the database" in my case , is not solved by having XML files in the ECUs folder of DDT4all.  Archiving them did not help. I am using the DDT4all installer for XP version (Maybe it helps).
 I Used the ECU's folder  extracted  from DDT2000 DB 04.08.2017.7z from the link 

 https://www.google.fr/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjP1ID9r4rcAhWjPZoKHRZUDZwQFggnMAA&url=https%3A%2F%2Fcartechnology.co.uk%2Fshowthread.php%3Ftid%3D23842&usg=AOvVaw0tlEpElrBCXMRZLPBdVpOw.

Please help !

cedricp, 2018-07-06:
Hi, the XP version is a bit outdated now, but if you copy the whole _ecu_ directory in the program file location (somthing like _C:\Program file\ddt4all_) that should work. Check that the file _eculist.xml_ is also present inside _ecu_ folder.

MLPMLPMLP, 2018-07-06:
Hello
Thanks for fast response.
Indeed XP is quite obsolete (a workaround at me now...)
I tried to copy all XML files and also in archive mode, everywhere and it is not working .
What is interesting is that ECUlist.xml does not exist in that database...

Should this be a part of the "ECUs " folder from the archive called "DDT2000 DB.7Z"  ?
Thanks !

cedricp, 2018-07-06:
Ok, so it won't work without that file, try to get a good one (the file **must** be inthere)

MLPMLPMLP, 2018-07-09:
Update: it's working with an older database(2013) which contains the ECUlist.xml .
I need to find something not older than 3 years , and everything I found (2017) is missing the ECUlist.xml...

Anyhow, thanx a lot for guidance !

KarelSvo, 2018-07-22:
Current ECUs can also be found at our Russian colleagues :) 
https://www.drive2.ru/b/498093336985338243/

bovirus, 2018-07-22:
@KarelSvo 

Is there a way (emai?) where can I contact the author of the post to improve the two .exe package?

Are only in English (I added Italian/French/Russian language for the installer)
The -py files in ddtplugins sub-folder are not complete (only one inside
in the tools sub-folder are missing all.bat files to handle localization

KarelSvo, 2018-07-22:
> The -py files in ddtplugins sub-folder are not complete (only one inside
In the full version  all ddtplugins are available.
The dacia version does not need any renault plugins.

alireza6631, 2019-01-04:
hello
I downloaded and installed the last software.
but I need downloaded for the data base for the ddt4all
please help me for downloaded data base ecu.
thanks

KarelSvo, 2019-01-04:
> hello
> I downloaded and installed the last software.
> but I need downloaded for the data base for the ddt4all
> please help me for downloaded data base ecu.
> thanks

https://github.com/cedricp/ddt4all/issues/84#issuecomment-406865869

b3none, 2020-04-16:
> > hello
> > I downloaded and installed the last software.
> > but I need downloaded for the data base for the ddt4all
> > please help me for downloaded data base ecu.
> > thanks
> 
> [#84 (comment)](https://github.com/cedricp/ddt4all/issues/84#issuecomment-406865869)

Can you do a dump of that database and upload it somewhere?

Furtif, 2023-06-02:
if not data base ... is normal uses discutions https://github.com/cedricp/ddt4all/discussions/833

closed 2023-06-02 by Furtif