cedricp/ddt4all #87: Interprétation des fonctions

diasilatia, 2018-07-16:
bonsoir cedric et merci pour ton super travail que tu fais,
en utilisant le ddt4all, j'ai remarqué que c'etait difficile d'interpreter certaines fonctions ex : ocs_present / hfm_present... etc.
ma question est : est-ce qu'il n'y aurait pas quelque part une interprétation de ces fonctions genre : 
- Front_sbr_inhibition = configuration signale sonore oublie ceinture de sécurité.
- Econometer_on_off = activer/désactiver indicateur de style de conduite.
.... etc.
ce serait vraiment très utile d'avoir un document pareil sous la main et pourquoi pas, l’intégrer à ddt4all..

merci d'avance et bonne continuation...

cedricp, 2018-07-16:
En effet un lexique serait pas mal, on pourrait demarrer une page WIKI pour ça, après, il faut trouver leur differentes significations.

diasilatia, 2018-07-16:
bsr cedric,

effectivement, cela serait très intéressant pour la page wiki.
en attendant, je voudrais atteindre certaines options dans ma Clio 4 phase 2 (GT-Line) de 2018 (je pense que c'est la x98ph2, corrige moi si je me trompe), grâce à ton programme j'ai pu activer l'indicateur de passage de vitesses, l'econometer et l'heure + la température sur l'ordinateur de bord.

par contre, je voudrais activer d'autres fonctions mais je ne sais pas comment les atteindre, alors les options sont les suivantes :
1- monté de la vitre conducteur (la seule qui soit à impulsion) en verrouillant les portes avec la carte et/ou en s’éloignant du véhicule.
2- allumage des feux au déverrouillage des portes (par carte et/ou par appui sur le bouton de poignée).
3- Désactiver les leds du plafonnier quand ont éteint le moteur.
4- Extinction temporisée du moteur quand la carte n'est pas détectée dans l'habitacle (je pense que ça n'existe pas à l'origine sur ce modèle mais soyons fous, vu que le système détecte bien l'absence de la carte quand on s’éloigne avec du véhicule et affiche "carte non détectée", ce serait une bonne prévention contre le car jacking).

j'ai aussi une autre question : est ce qu'il y a une différence quand on scanne les ECUs moteur tournant, car quant on insert la carte le contact temporise et entre en veille?

merci d'avance cedric, je sais que j'en demande bcp mais j'imagine que cela servira même aux personnes désirant en connaitre plus sur la puissance de votre logiciel et leurs voitures.

Furtif, 2023-06-02:
not a issue please uses https://github.com/cedricp/ddt4all/discussions