cedricp/ddt4all #88: RADNAV ver2.3 Configuration Reset button problem

mansour-isa, 2018-07-22:
Hi,
Unfortunatly I clicked on Configuration Reset button(default value)  in DDT4ALL at Audio->RADNAV ver2.3  and now the screen is black and nothing to show,
how can I fix it ?, could anyone help me please?

shrlnm, 2018-07-25:
Unfortunately, you are not the first ane. I may try to help. Give me your email

mansour-isa, 2018-07-26:
Thank you Shrlnm,
my email address is victor_omidi@yahoo.com, but I tried myself by reading   Audio->RadNav v2.3->Configuration->Engineering Configuration parameters from other Talisman and setting up in my car then now it is working properly,
[image]
[image]

closed 2018-07-26 by mansour-isa