cedricp/ddt4all #89: Problem ddt4all-bad elm response Megane 4

simin87, 2018-07-22:
Hi Cedric,sorry for my English.I have an error in ddt4all:bad elm response-wrong response
I tried two different obd tools and two different ddt4all version(exe and master).
How can I do?
My car is a Renault Megane 4 with rlink2
I send you screen
[image]

shrlnm, 2018-07-25:
"Wrong response" with a  high probability means then you sent not supported command to your ECU or you are using bad ELM. The details you may look in elm* log file. Are you sure that you are using the appropriate xml?

simin87, 2018-07-25:
How can I find log files? I have tested elm with a program and it is compatible..what is xml appropriate?

shrlnm, 2018-07-25:
Usually the log is written to folder logs inside the ddt4all folder. There are should be two. Most interesting one have a name started with elm prefix

shrlnm, 2018-07-25:
6 from 8 fields a picture above have an ERROR values. Usually it means that you use wrong xml. Did you use scanning and ddt4all determined this xml?

KarelSvo, 2018-07-25:
In the photo a very old ddt4all version.

shrlnm, 2018-07-25:
then look into logs for details. Search the command you sent before WRONG RESPONSE but without spaces. For example from your picture we  need to find 2E209301

KarelSvo, 2018-07-26:
[image]
[image]
[image]

simin87, 2018-07-26:
The folder ddt4all can not find it .. I have to try the procedure again with the exe and then look for the log.
I scanned the ecu and found them all ... I think that elm works well.. 
Where is xml?

simin87, 2018-07-26:
Sending the wrong command?thank you

shrlnm, 2018-07-26:
As I can see, your car is equipped with a simple "FrontCamera". It could not do more powerful features like a LKA, ACC or AEBS witch may be realised by "RaCam". You may try to enable TSR if it is not enabled by default.

simin87, 2018-07-26:
Ok, how do I verify Tsr enable?

shrlnm, 2018-07-26:
it depends on a set of other ECUs (RadNav and TDB) 
Usually you need:
1) enable TSR in FrontCamera
2) enable OSP in RadNav by ddt or "Developer side"
3) reboot RadNav by long press on power button
in some cases you need pp4
4) reboot TDB by command 1103 (it may reset some saved parameters in TDB)

igno77, 2018-07-26:
Someone had tryed to activate distance warning (DW) on this simple camera?? In the ecu file is possible to set as activate, in the radnav is possible to set on camera detection instead of radar detection but....it seems that doesn't work!

simin87, 2018-07-26:
Is it right?It's in demo mode.

[image]
[image]
[image]

KarelSvo, 2018-07-26:
Your ddt4all version is still too old, can not distinguish phase 1 and 2.
C1A not yet in production. This is Megane 4 Phase 2.
Uninstall everything completely and use this exe https://www.drive2.ru/b/498093336985338243/

shrlnm, 2018-07-26:
[image]
[image]

KarelSvo, 2018-07-26:
But please do not use the camera xml from HFE Kadjar.
[image]
[image]

shrlnm, 2018-07-26:
Yes sure. The right way is to use scanning for determining right xml for your car

KarelSvo, 2018-07-26:
Scanning does not always help if the software references in ddt_xml are not up-to-date.
With a current ddt4all version and the right vehicle selection, you can filter out the relevant ecu very well.
Or the way via pyrene with clip database identifying the ecus.

simin87, 2018-07-26:
Hi Karel,in that page what should I activate?also in radnav 2.4 page?

simin87, 2018-07-27:
Nothing .. I tried and it's all active by ddt4all..Probably it can not work.no bad elm response error..
How to send log files?unknown exstension.

KarelSvo, 2018-07-28:
Maybe the log folder zip compress?
> How to send log files?unknown exstension.

shrlnm, 2018-07-28:
it is better to use something like google disk and share the link

simin87, 2018-07-28:
https://drive.google.com/open?id=1NhE-re5muM_lXhGIBje8gimguv3txiTq
https://drive.google.com/open?id=13z6hKnyuPWTKIbAtzxX6QAb4phxo3tIz
https://drive.google.com/open?id=1nED4a1OcdxlvMif5PB_9zB7E-wM4IxGB

shrlnm, 2018-07-28:
Log shows that all commands to RadNav and FrontCamera2014_ECU_V3.0_20150120T125936.xml were accepted and have positive response

But command to DAE "LKA_LDW_Activation" was not accepted and returned Negative Response

742;2E01BD03;7F2E12;

12 at the end means "SubFunction Not Supported"

simin87, 2018-08-14:
hi, possible solutions?is it a hardware problem?

alireza6631, 2018-12-30:
Hi

alireza6631, 2018-12-30:
Hello, I use the elm to encounter error

Furtif, 2023-06-02:
ok seen a problem but not at program here , ask as https://github.com/cedricp/ddt4all/discussions and rapport discution of

closed 2023-06-02 by Furtif