cedricp/ddt4all #90: How can enable to Talisman icon Traffic sign read function ?

karalahana, 2018-08-04:
Hello everyone;
I need to help. I have a talisman icon car and unfortunatelly Traffic sign read function ( on the way side has speed limits 90-70-50etc. )and driver tired detection function is deactive to my country. I would like to know how can i find that menu on ddt4all software. I have a rlink 2.2 version. Thank you for corporation.

karalahana, 2018-08-04:
UP

mansour-isa, 2018-08-08:
To read traffic sign your car should have front camera, are you sure your car equipted by front camera?
and for driver tired detection you can go through https://www.foroclub.es/megane4/viewtopic.php?f=446&t=20881
it is same as talisman

Furtif, 2023-06-02:
make question to https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif