cedricp/ddt4all #92: Possibly bricked R-Link

robnoo, 2018-08-10:
I wanted to unlock Android Auto on my Clio IV with R-Link. I first updated the R-Link software through the SD card to the newest version and checked version number with the various guides. Next I connected with DDT4ALL through a Konnwei KW902 and a modified cable. Initially had some problems with connecting and had an incorrect modified cable. At last I got a connection, but things didn't go as planned. Still had trouble to read values and an ECU scan only gave a yellow match with the navigation ECU. When trying to write values, the program started to hang and after a manual force close on the computer side, nothing had changed on the system. I believe it hang with a 10 81 code. Tried again with same results. I did not press any reset buttons inbetween.

Finally decided to write complete ECU, which also hang with the same code, but after a restart of the R-Link, don't remember if it was a hard reset from DDT4ALL or pressing the home button 5 times, it broke to a "Please add EOL settings" screen with a yellow triangle. After this, a bit scared, tried to reset the system from DDT4ALL, I think with Reset Configuration, the screen went completely black. At this point, I decided to go to the dealer. After 45 minutes, they managed to fix my last mistake and brought it back to the EOL settings screen. At this moment, they couldn't fix this error, due to central server problems in France. Brought the car back at a later moment, at the end of a day, and they still couldn't fix the EOL settings problem. The ECU doesn't respond at all and they told the R-Link is dead and needs replacement. Took my car unrepaired for now and decided to first try some more. Also, as it was at the end of day, I feel like they might not have tried everything, as they seemed to want to go home. 

When connecting to DDT4ALL at this moment on the navigation CAN, I can't find a single ECU. When scanning for ECUs, I get an (error) log as attached. When connecting to the normal CAN, I have no problems. Has anyone encountered this problem before or knows a solution to this?

At this moment I can think of three possible solutions:
- Rescue current system in some way
	- Reprogram without dealer -> is this even possible?
	- Try other dealer for second opinion
- Replace R-Link with second hand from broken/crashed car
	- Is this possible or does the R-Link needs to be linked to the car in some fashion?
- Replace R-Link by dealer

To be clear, this is my own fault as I made these mistakes:
- Continueing while response seemed weird -> also had no data as mentioned in: https://github.com/cedricp/ddt4all/issues/14#issuecomment-335381907
- Choosing write all, while ECU was not completely the same
- Wrong reset option -> https://www.speakev.com/threads/android-auto-update-done.97393/page-5#post-1903065

[image]
[image]

Logs:
[elm_ddt.txt](https://github.com/cedricp/ddt4all/files/2277527/elm_ddt.txt)
[ecu_ddt.txt](https://github.com/cedricp/ddt4all/files/2277528/ecu_ddt.txt)

shrlnm, 2018-08-15:
Have you solved the problem?
The correct xml for your car should be 
Radio_2012_V2.4_20130604T134716.xml
or
Radio_2012_V2.2***.xml from old databases

Check if the VIN is correct in your R-Link

robnoo, 2018-08-15:
Thanks for the response, unfortunately not yet.  Just connected the car with Radio_2012_V2.4_20130604T134716.xml.

All settings were empty, including the VIN. I added and wrote the VIN, but that did not solve the problem. Screen stays the same, also after reboot. I don't what the 4 charactars in the separate field should be. 
[image]
[image]
[image]

Would it be safe to let the ECU reset all to default settings with the ECU reset option under "11 ECU - reset"?

I would like to point that when searching for ECUs, it fails during the search for audio ECUs.

shrlnm, 2018-08-15:
CRC digits for your VIN is '341B' you may look them in responses from other ECUs

shrlnm, 2018-08-15:
[image]
[image]
[image]
[image]
[image]
If it doesn't help, then write me your email

shrlnm, 2018-08-15:
Also keep in mind that your vin is in log and on every picture you posted above

shrlnm, 2018-08-15:
I am pretty sure that EOL is a group of acoustic parameters defined by commands

2E2B2A
2E2B2B
2E2B2C

Check what values they have now

222B2A
222B2B
222B2C

cedricp, 2018-08-16:
I cannot help with this issue, but I fixed the ECU reset button problem in the master branch.

shrlnm, 2018-08-16:
The parameters looks good. They should be 

622B2A924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002924802000000200000001002FFFF0300FFFF00004594000000000000000000001A000000340000001A0000008A52C0FFE1AD1F0069000000D20000006900000011A5C0FF965C1F00D5D61F005552C0FFD5D61F008A52C0FFE1AD1F00B6A51F0093B4C0FFB6A51F0092B5C0FF6C4C1F0069000000D20000006900000011A5C0FF965C1F00FFFF1F0000000000000000000000000000000000FFFF7F00FFFF7F00FFFF1F00610BF6FFDB0E000088880800FF00000000FF00000000FF00000000FFFFFF00000000FFFFFF00000000FF00000000FF00000000FFFFFF00000000FFFFFFFF0000FFFF0000FFFF0000FFFF00008D2F00002122220003030000070600000303000085C2C1FF8C491E00C1211F007DBCC1FFC1211F0085C2C1FF8C491E00323304000000000000000000CECC04000000000000000000000000FF0000200000000000000000000000000000000000000020000000000000000000000000000000000000002000000000000000000000000000000000000000200000000000000000000000000000000000000020000000000000000000000000000000000001000000000000000000000000000000000000000000000000000000000000000000000000000000
622B2B00006B00001B0000770000470000580000017FFFFF19816C1211441211017FF7FF7FF7FF7FF7FF22616144DEC23E7E8BE8176304C6C6C9964E63AF01789030712C613D274FEB601E02403DC486CC784EE522616144DEC23E7E8BE8176304C6C6C9964E63AF01789030712C613D274FEB601E02403DC486CC784EE567E00C4FCC1943C7E4ED87CA004795C0959F7F5EA268700150FC032027FCF01099002133C045237FBF0167E00C4FCC1943C7E4ED87CA004795C0959F7F5EA268700150FC032027FCF01099002133C045237FBF012F13FB21C8092F13FB5B97F61F1C090004000000000000000000000000000004000000000000000000000000002F13FB21C8092F13FB5B97F61F1C0900040000000000000000000000000000040000000000000000000000000058E11F497DF658E11F02A7115A0CAB3310A130FEF83310A107F7623EFC2B0C40B60C40B600000026336900000058E11F497DF658E11F02A7115A0CAB3310A130FEF83310A107F7623EFC2B0C40B60C40B60000002633690000004FD302222A1A4FD302420747349C2F000400000C000000000ED1B00000004FD302222A1A4FD302420747349C2F000400000C000000000ED1B00000000004000000000000000000000000000004000000000000000000000000006A13F92BD80C6A13F94F67F3271C0C000400000000000000000000000000000400000000000000000000000000651392580E612D121D28019E6DDE4F3A030419BA373A03042DD62C1CCD142E90055D3C0A7977F4E9140A00D014C1B0AF7E4E730AD00F15BC1E79C7DEEE5092D256B8E6710AE16428F2016D0022DCC043CC7FBF0119D07C33CCF81C767FEB60F7DF0EBDECC053CA83890007D67B9C1E36D00F1B6FAA7FD1EDC020FC00107E000000FFE782E827CAC1B00D000FF2FBC000400000000000000000000000000000400000000000000000000000000000400000000000000000000000000000400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
622B2C00001400001400002E00002E0000010000017FFFFF19819816C16C1011017FF7FF7FF7FF7FF7FF22616144DEC23E7E8BE8176304C6C6C9964E63AF01789030712C613D274FEB601E02403DC486CC784EE522616144DEC23E7E8BE8176304C6C6C9964E63AF01789030712C613D274FEB601E02403DC486CC784EE567E00C4FCC1943C7E4ED86E20075C5C0F7717EFE6716D0022DCC044357FBF01099002133C045237FBF0167E00C4FCC1943C7E4ED86E20075C5C0F7717EFE6716D0022DCC044357FBF01099002133C045237FBF012F13FB21C8092F13FB5B97F61F1C096403E605181A0004007AE7E51C0C196403E655A81C0004002A67E31C0C192F13FB21C8092F13FB5B97F61F1C096403E64BB81B0004003447E41C0C196403E636881F0004004977E01C0C1958E11F497DF658E11F02A7115A0CAB3310A130FEF83310A107F7623EFC2B0C40B60C40B600000026336900000058E11F497DF658E11F02A7115A0CAB3310A130FEF83310A107F7623EFC2B0C40B60C40B60000002633690000004FD302222A1A4FD302420747349C2F000400000C000000000ED1B00000004FD302222A1A4FD302420747349C2F000400000C000000000ED1B00000006403E605181A0004007AE7E51C0C196403E64BB81B0004003447E41C0C196A13F92BD80C6A13F94F67F3271C0C6403E655A81C0004002A67E31C0C196403E636881F0004004977E01C0C19651392580E612D121D28019E6DDE4F3A030419BA373A03042DD62C1CCD145F30063E6C0D1847F2E7306D00C0DCC187E57E6E81728010652C212AF7DBECC092D256B8E6710AE16428F2016D0022DCC043CC7FBF0119D07C33CCF81C767FEB60F7DF0EBDECC053CA83890007D67B9C1E36D00F1B6FAA7FD1EDC020FC00107E000000FFE782DFE7CAC1B00D000FF2FBC000400000000000000000000000000000400000000000000000000000000000400000000000000000000000000000400000000000000000000000000019819819819819819800000000000000000000000000000000000000000000000000000000000000000000

robnoo, 2018-08-16:
@cedricp Thanks for the commit, I can now hard reset correctly.

@shrlnm My values are indeed the same. I feel like the radio ECU is complete at this moment. I originally started with the MFD 4.6 ECU. Do you also have the values for that ECU? My mail adress is r3e3r3 [a] gmail.com btw. (will remove this when not needed anymore). Or do you know which other ECUs exist on the M-CAN on the Clio IV Phase 1?

Gg261993, 2018-08-21:
Hello, I have a megane 3 and actually I have the same problem. Did you fixe it ? Or not yet. Sorry for my English.

mnigten, 2018-10-01:
Hi there, did you solve the issue already? Seems that i have the same issue after a reset.

mnigten, 2018-10-06:
I tried to find some information but there are not a lot of people with the same issue as I can find. When i push 5 times the home button I get a logo of TomTom so my assumption is that the bootloader is working but stops then and popups the triangle you mentioned and the EOL error.  Is there a possibility that the R-link application kernel is gone?

robnoo, 2018-10-09:
Could well be, in my case it's the same. My guess would be that by writing some non-existing settings, actually some other settings have been overwritten(possibly from another ECU if they are physically on the same storage). 

If you connect to ddt4all and execute a screen test(changing the screen to black or white etc.), does your system respond? Mine doesn't respond to that either. 

shrlnm sent me some Russian links a while ago, one of which include an image for sd cards with a tool on it in /system/bin/eoltool. I don't know how to run or use it though, he thought about adb. This might be difficult though if a system is not booting. 

http://www.club-renault.ru/forum/topic/62166-navigator-r-link/?do=findComment&comment=2187516 
Options include:
- 1 : Clean EOL configuration (UNCONFIGURED)
- 2 : Configure the device (CONFIGURED)
- 3 : Export/Import EOL settings to/from a file
- 4 : Show Virgin Bits
- 5 : Export Virgin Bits to a file
- ! : Main menu
- ? : Reprint menu
- q : Quit tool

closed 2019-01-18 by robnoo

mnigten, 2019-01-27:
Hi There,

Because issue is closed. Did you find a solution already? i could not find the tool mentioned above. mine is still not working.
Even my BIC was freezed so complete CANHS was dead but found a solution to reset the BIC and get it booted again to the screen with the error message.

ciapata, 2019-01-27:
How did you reset BIC? I have one that seems dead, no response at all. I have configured R-Link with Chinese Clip.

mnigten, 2019-01-27:
The problem is that most of the times a wrong script is loaded. For mine Clio the script needs to be on 14. So the bic is sleeping and not awaking by the can. The way to solve this is to awake it. Then all Canhs ecus will be found. Just connect pin 2 (empty in connector) on the bic with 12v. I used pin 3. That’s 12v. Leave the connection until you reporammed the script to the correct version.

ciapata, 2019-01-27:
Thanks, I'll try it and see what happens. To answer the question about adding R-Link EOL settings, at least partially, it is possible with Chinese Clip. It's broken now, so I can't connect anny sniffer and record what messages are sent, but there's no dealer help neccessary. If I remember correctly, you must have Dialogys and Clip installed. There is no "Add EOL" button in Clip software per se, but there is a link directly in the diagnostic/repair manual, you have to click to run EOL procedure.

robnoo, 2019-01-29:
In my case, I wasn't able to repair the system, neither was the dealer. In my case, I think it's possible that something was broken before, as I didn't get a good connection to the system before making changes. 

I closed the issue, as it was a problem with a system, not with the software in this repo.

nazmi02551, 2019-07-01:
Hello there.  I'm having the same problem.  Did you solve the problem?

pietroferri, 2019-12-18:
Some problem, Clio 4 pre-restyling R-Link - 2016

nazmi02551, 2019-12-18:
> Same problem, Clio 4 pre-restyling R-Link - 2016...

my problem is solved. renault technical service, said it was a problem that has not encountered before. They reinstalled the navigation device with can clip. there was something to be considered here. After the operation with can clip, the connection cable needs to be disconnected to continue the process.

pietroferri, 2019-12-18:
Thank you nazmi02551, tomorrow I will go to Renault service and I'll let you know.

seb0031, 2020-07-05:
Bonjour, 

je suis dans la même galère, avec le message d'erreur "eol settings". pouvez-vous m'aider ? j'ai une zoe de 2013. j'ai voulu mettre a jour android auto avec ddt4all. depuis je peu plus rien faire. help me help me

cordialement

seb0031, 2020-07-05:
> Merci, je vais l'essayer et voir ce qui se passe. Pour répondre à la question sur l'ajout de paramètres EOL R-Link, au moins partiellement, c'est possible avec Chinese Clip. Il est cassé maintenant, donc je ne peux pas connecter anny sniffer et enregistrer les messages qui sont envoyés, mais il n'y a pas d'aide nécessaire du revendeur. Si je me souviens bien, vous devez avoir Dialogys et Clip installés. Il n'y a pas de bouton "Ajouter EOL" dans le logiciel Clip en soi, mais il y a un lien directement dans le manuel de diagnostic / réparation, vous devez cliquer pour exécuter la procédure EOL.
Bonjour, 

je suis dans la même galère, avec le message d'erreur "eol settings". pouvez-vous m'aider ? j'ai une zoe de 2013. j'ai voulu mettre a jour android auto avec ddt4all. depuis je peu plus rien faire. help me help me

cordialement

seb0031, 2020-07-05:
> Merci, je vais l'essayer et voir ce qui se passe. Pour répondre à la question sur l'ajout de paramètres EOL R-Link, au moins partiellement, c'est possible avec Chinese Clip. Il est cassé maintenant, [...]

Bonjour, 

je suis dans la même galère, avec le message d'erreur "eol settings". pouvez-vous m'aider ? j'ai une zoe de 2013. j'ai voulu mettre a jour android auto avec ddt4all. depuis je peu plus rien faire. help me help me

cela a t'il marché en changent le "script" ?

cordialement

seb0031

gaouar, 2020-08-07:
bonjour , 
je suis sur le même; configuration EOL
vous aviez trouvez la solution ?

luissilva1971, 2020-11-07:
Hi guys. I was reading this issue and it is dome kind related to o r thing i am desperate to do but still did not do it. I have a megane 3 sw 2010 1.5 dci. Car had standatd rad01 with tomtom carminat (no bluetooth). I want to replace radio to a rad02_08 with bt function so i bought from junkyard a 281159389r radio that came from a car without tomtom. From what i have read from this site and another forumd if is possible to achieve that changing the rad02 parameters indicating that car has nav1g1 navigation and changing in my nav system the rad01 to rad02. Still the rad02 has the vin from the other car. Maybe not needed but can the vin of rad02 radio be changed to the vin in my car. For that it will be needed the vin.crc? How can it be calculated with ddt4all? Thanks in advance for all help . Regards

seb0031, 2020-11-07:
Concessionnaire Renault

mnigten, 2020-11-07:
You can read the vin from other module. The new module should be visible . With the calculator in de programm at the top calculate the crc and program it to the new module. 
But maybe there are other settings needed. I never tried this.

mnigten, 2020-11-07:
Do not forget this is the Can HS

Franel44, 2021-01-13:
Bonjour,

Je suis dans la même situation, ddt4all pour Android auto et après redémarrage, triangle jaune.

J'ai essayé de réécrire les ecu dans navigation, enlevé le fusible... Rien n'y fait 🥴.

Avez-vous une solution ?

ripper1979, 2022-03-15:
Pour info, j'ai trouvé un solution à ce problème. Il suffit de réinjecter une configuration.

Par contre , l'option d injecter une config sous ddt4all je ne la trouve pas. Je le fais avec ddt2000

garzazum, 2022-06-10:
> Pour info, j'ai trouvé un solution à ce problème. Il suffit de réinjecter une configuration.
> 
> Par contre , l'option d injecter une config sous ddt4all je ne la trouve pas. Je le fais avec ddt2000

Hi @ripper1979 , could you please advice what version of ddt2000 you used (with some standard ELM?) , and what exact configuration you pushed, something to MFD or rather to radio/audio ?

ripper1979, 2022-06-13:
> Hi @ripper1979 , could you please advice what version of ddt2000 you used (with some standard ELM?) , and what exact configuration you pushed, something to MFD or rather to radio/audio ?

Hi, to use DDT2000 instead of DDT4all, you may need anothers vci (derelek usbdiag or the new bosch).

And it's the mfd configuration of m car  that i used to reconfigure another mfd which have lost it own

garzazum, 2022-06-14:
Merci @ripper1979  pour votre réponse, j'avais peur de cela, cordialement

maglez81, 2024-09-06:
Hi. I'm experiencing the same issue. have you found a solution?

ripper1979, 2024-09-06:
> Hi. I'm experiencing the same issue. have you found a solution?

you have the "eol fault"? if yes, you can use pyren and the pyren bot to make an autosetup