cedricp/ddt4all #95: Mod to use with small screens: netbooks, windows tablets...

pakoto, 2018-08-26:
[image]

I made some modifications to code. I´m not a programmer, only cut copy paste code. I´m trying to maximize main screen space. File menubar removed. "FAVORITE" button loads previously saved eculist named "default.ecu". When push arrows icon, menu bar expand down to show SAVE ECU LIST. I don´t know how to add load xml, load ecu lists, plugins, etc.

I have more ideas, but i dont know how to do it: Convert ecu name treedock to combobox and add to upper menubar, to save space. Also i´d like to hide or reduce log window.

pakoto, 2019-04-21:
I have no idea about python and qt, and i only get a workaround using copy paste and google search. 

Added height limit to log and ecu. Should be better using combobox or something similar.

Added SAVE ECU LIST and FAVORITES buttons that save ecu list to newcar.ecu and read newcar.ecu when push it

invert zoom buttons

remove expert mode button and enable it by default.

[image]

        self.treedock_logs = widgets.QDockWidget(self)
        self.logview = widgets.QTextEdit()
        self.logview.setReadOnly(True)
        **self.logview.setMaximumHeight(100)**
        self.treedock_logs.setWidget(self.logview)
        self.treedock_ecu = widgets.QDockWidget(self)
        self.treeview_ecu = widgets.QListWidget(self.treedock_ecu)
        self.treedock_ecu.setWidget(self.treeview_ecu)
        **self.treedock_ecu.setMaximumHeight(70)**
        self.treeview_ecu.clicked.connect(self.changeECU)

pakoto, 2019-04-21:
Remove expert mode

        self.toolbar.addSeparator()
        self.toolbar.addAction(self.log)
#        self.toolbar.addAction(self.expert)    <-----------
        self.toolbar.addSeparator()
        self.toolbar.addAction(self.autorefresh)
        self.toolbar.addAction(self.refresh)
        self.toolbar.addSeparator()
        self.toolbar.addAction(self.diagaction)

Al promode false -> true
    def changeScreen(self, index):
        item = self.treeview_params.model().itemData(index)
        if qt5:
            screen = item[0]
        else:
            screen = utf8(item[0].toString())
        self.paramview.pagename = screen
        inited = self.paramview.init(screen, self.screenlogfile)
        self.diagaction.setEnabled(inited)
        self.hexinput.setEnabled(inited)
        self.cominput.setEnabled(inited)
        self.expert.setChecked(False)
        options.promode = True   <-------- False to True
        self.autorefresh.setChecked(False)
        options.auto_refresh = False
        self.refresh.setEnabled(True)

        if pc.mode == 0 or pcres == widgets.QDialog.Rejected:
            exit(0)
        if pc.mode == 1:
            options.promode = True
            options.simulation_mode = False
        if pc.mode == 2:
            options.promode = True
            options.simulation_mode = True
            break

pakoto, 2019-04-21:
I´m trying to convert saved ecu list to combobox, but i don´t know why.

This is an "easy" example of working combobox

        self.canlinecombo = widgets.QComboBox()
        self.canlinecombo.setFixedWidth(150)
        self.canlinecombo.currentIndexChanged.connect(self.changecanspeed)
        self.canlinecombo.addItem("CAN Line 1")
        self.canlinecombo.addItem("CAN Line 2@500K")
        self.canlinecombo.addItem("CAN Line 2@250K")
        self.canlinecombo.addItem("CAN Line 2@125K")

I think that is possible to read each item and sending to combobox and put into top menu. 

        vehicle_file = "vehicles/mycar.ecu"
        jsonfile = open(vehicle_file, "r")
        eculist = json.loads(jsonfile.read())
        jsonfile.close()

        self.treeview_ecu.clear()
        self.treeview_params.clear()
        if self.paramview:
            self.paramview.init(None)

        for ecu in eculist:
            item = widgets.QListWidgetItem(ecu[0])
            self.ecunamemap[ecu[0]] = ecu[1]
            self.treeview_ecu.addItem(item)

P.D.
2 zoom buttons for app (creative commons i think)
https://www.onlinewebfonts.com/icon/429149
https://www.onlinewebfonts.com/icon/429148

Furtif, 2023-06-02:
it seems to be complicated but it really has nothing to do with a problem with the code given for free, please review this if it is not resolved at https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif