cedricp/ddt4all #96: Renault Kadjar Bricked Main Screen

semih8989, 2018-09-05:
Hi all,

I have a problem with my screen , I wrongly pressed a button in system menu and now it is fully dark :) Is it possible to turn it back with ddt4all ?

shrlnm, 2018-09-05:
Yes. You need to define right screen parameters. They depends on size of your screen

semih8989, 2018-09-05:
I have 2015 Kadjar with standart screen , and how can find that screen parameters , and from which section of ddt4all can I change the values ?

shrlnm, 2018-09-05:
"Configuration/Egineering configuration"
You may copy these from other similar car or give me you email and I try to find these parameters for you

semih8989, 2018-09-11:
When I try to search the ECUs on my car , it is freezing. you may see the log below. What is the problem you consider ? I tried to load the original parameters for screen, but it is still not working

Scanning address ECU Airbag
Scanning address ECU BCM/UCH
Scanning address ECU Tachometer
Scanning address ECU ABS/ESC
Scanning address ECU ACC
Scanning address ECU Power steering
Scanning address ECU AMP
Scanning address ECU RADAR
Scanning address ECU LDCM
Scanning address ECU Video
Scanning address ECU HFM
Scanning address ECU BVA
Scanning address ECU SOW Left
Scanning address ECU MC HEV FSCM
Scanning address ECU Audio
Traceback (most recent call last):
  File ".\ddt4all.py", line 92, in scanselvehicle
    self.parent().parent().scan_project(project)
  File ".\ddt4all.py", line 561, in scan_project
    self.scan()
  File ".\ddt4all.py", line 617, in scan
    self.ecu_scan.scan(self.progressstatus, self.infostatus)
  File "C:\Program Files (x86)\ddt4all\ecu.py", line 1536, in scan
    if not self.identify_new(addr, label):
  File "C:\Program Files (x86)\ddt4all\ecu.py", line 1483, in identify_new
    self.check_ecu2(diagversion, supplier, soft, soft_version, label, addr, "CAN")
  File "C:\Program Files (x86)\ddt4all\ecu.py", line 1651, in check_ecu2
    delta = abs(int('0x' + tgt.version, 16) - int('0x' + version, 16))
ValueError: invalid literal for int() with base 16: '0x19300\xff\xff\xff\xff\xff'

shrlnm, 2018-09-11:
Hi

>> What is the problem you consider ?

Send this error log to Cedric. It is an error of code and only he may fix it

I suggest you to try pyren. It contains mod_ddt program which is analog of ddt4all, but it may get the dump of ECU configuration. The dump may be compared with an etalon configuration for your VIN.  
You may get the code, databases and instructions there

https://www.drive2.ru/l/9214434/  

>>I tried to load the original parameters for screen, but it is still not working

Did you do the reload of r-link after configuration changing? You need to do it by long press on power button or by disconnecting the battery

pcboys, 2018-09-24:
touch screen default parameters for kadjar

1	DebugData0	0
2	DebugData1	0
3	Touch.NumberOfRows	24
4	Touch.NumberOfColumns	32
5	Touch.TouchResolutionX	1024
6	Touch.TouchResolutionY	768
7	Touch.ScreeSizeX	17764
9	Touch.ScreenSizeY	13248
1	CentroidThreshold	28
11	NoiseThreshold	8
12	NegativeNoiseThreshold	1500
13	FingerThreshold	5
14	FingerVelocityX	2048
15	FingerVelocityY	2048
16	FingerDebounce	2
17	NumberOfTXClocks	11
18	Touch.TXDividerValue00	125
19	Touch.TXDividerValue01	87
20	Touch.TXDividerValue02	88
21	LargeObjectThreshold	25
22	FatFingerThreshold	10
23	LowPowerInterval	100
24	RefreshInterval	10
25	ActiveModeTimeout	2000
26	WakeUpSource	2
27	COMM_INTPulseWidth	100
28	XYFilterMoveThold	10
29	XYFilterFastThold	8
30	XYFilterSlowCoeff	16
31	XYFilterFastCoeff	16
32	FatFingerDeflectionLimit	10
33	IDACHoldOffValue	2
34	FreqHoppingOperationalMode	0
35	FreqHoppingNoiseThreshold	230
36	FreqHoppingRefreshCounterThreshold	50
37	BaselineUpdateRate	0
38	AdvancedFingerSeparation	1
39	Touch.InnerEdgeGain	16
40	Touch.OuterEdgeGain	3

1	Display.VerticalPeriod	525 line
2	Display.VerticalSyncLPulseWidth	2 line
3	Display.VerticalBackPorch	8 line
4	Display.VerticalFrontPorch	35 line
5	Display.VerticalActiveArea	480 line
6	Display.FrameRate	60 Hz
7	Display.PhaseDifference	0 PCLK
8	Display.HorizontalPeriod	1056 PCLK
9	Display.HorizontalSyncLPulseWidth	128 PCLK
10	Display.HorizontalBackPorch	10 PCLK
11	Display.HorizontalFrontPorch	118 PCLK
12	Display.HorizontalActiveArea	800 PCLK
13	Display.PixelClockFrequency	3333MHz
14	Display.TouchscreenReset	Present
15	Display.PixelClockEdgeSelect	Present
16	Display.I2CPassThroughAll	Not Present
17	Display.DeserializerI2C	2c
18	Display.SCGControl	00
19	Display.TouchscreenControllerI2CBoot	69
20	Display.TouchscreenControllerI2COperating	67

semih8989, 2018-09-25:
Hello all,

I have replaced my wifi elm327 dongle with bluetooth version. I still have a problem with ddt4all search action. When I select the car type(HFE KADJAR) , it starts to search the ecus and stops at 25% , finds only 2 ecus. And If I try to search without selecting the car tupe , it founds about 11-12 ecus but doesnt shown under the action bar. What is the problem you consider? Now my screen doesnt work but, radio and bluetooth function is active.

cedricp, 2018-09-25:
First, try to update from sources
Second, send a screenshot of the terminal window, there should be an error reported there.

cedricp, 2018-09-25:
This has normally been corrected in newer versions, get the code and replace it in the application directory. The problem is the \xff characters and it they should be removed in last version

cedricp, 2018-09-26:
From what I see, you have to use "RadNav v2.4" file to perfect match the ECU soft ID.
The "write touch" button will write all parameters at once, for information. Did you try to write values given  by @pcboys ? Don't forget to hard reset the ECU to validate changes.
If it still doesn't work, send me your log files over, so I try to recover original values.

semih8989, 2018-10-02:
[image]

My defect is still going on. ı couldnt find the true ecu data for radnav. I attached an example of developer menu that I firstly depressed the display mode button.

shrlnm, 2018-10-03:
[image]

killuadk, 2018-11-04:
Hello i have the same problem gyus is there a solution white ddt4all !!

kadjarmio, 2018-11-14:
Stesso problema,  schermo nero , passato in officina , dato che sono fuori garanzia mi hanno fatto pagare 40€,  e mi ritrovo il display in orizzontale, aiuto per favore, grazie

semih8989, 2018-11-15:
hi all again , I solved the bricked screen on kadjar. You have to install the software in the shop. if the screen orientation stays as horizontal, you need to select the display mode in mid via developer menu. that is all :) Now it is working and I can watch videos while driving .thx for all your help.

closed 2018-11-15 by semih8989

mitakbas, 2019-03-06:
I've entered the developer mode but accidentally I've touched the first top right option related to 'Screen display', I think it's switched to 'high'...
Now the screen is off (blank), the functions are working but the screen is blank.
I tried more than once to switch off the engine, restart the display unit, and even i tried to pull out the fuse, but nothing worked...
Please help me
renault kadjar

mitakbas, 2019-03-24:
I solved the problem, I entered the following values in the central panel-display parameters 1 Display.VerticalPeriod 525 line
2 Display.VerticalSyncLPulseWidth 2 line
3 Display.VerticalBackPorch 8 line
4 Display.VerticalFrontPorch 35 line
5 Display.VerticalActiveArea 480 line
6 Display.FrameRate 60 Hz
7 Display.PhaseDifference 0 PCLK
8 Display.HorizontalPeriod 1056 PCLK
9 Display.HorizontalSyncLPulseWidth 128 PCLK
10 Display.HorizontalBackPorch 10 PCLK
11 Display.HorizontalFrontPorch 118 PCLK
12 Display.HorizontalActiveArea 800 PCLK
13 display.pixelclockfrequency 3333mhz
14 display.touchscreenreset Present
15 display.pixelclockedgeselect Present
16 Display.I2CPassThroughAll Not Present
17 display.deserializerı2c 2c
18 display.scgcontrol 00
19 display.touchscreencontrollerı2cboot 69
20 display.touchscreencontrollerı2coperating 67
After resetting the image came side by side, then the developer entered the mode and set the image to mid adjust everything

hero92500, 2019-04-02:
hello i talisman intial displaymode click on mid and since then my screen is black is someone for help please thank you

Bozobobo, 2019-06-17:
> I solved the problem, I entered the following values in the central panel-display parameters 1 Display.VerticalPeriod 525 line
> 2 Display.VerticalSyncLPulseWidth 2 line
> 3 Display.VerticalBackPorch 8 [...]

Hello, great job
I had same problem.
Can you describe step by step how you did it? (ddt4all / developper mode, RadNav 1.5/2.x/3.x, ...)
Thanks

Younes481, 2020-02-14:
I clicked on the mid button in developer mode and since my screen is flipped in fact I have a new kadjar life version rlink2 9.0.33 with 7 inch screen without navigation. Any help. Younes481@gmail.com

foggiamanuel, 2021-07-27:
Salve, anche io ho lo stesso problema. Come posso fare? Ieri sono stato alla Renault e mi hanno arrivato il video ma lo schermo era ruotato. Come posso fare? Vi prego. Aiuto. foggiamanuel@gmail.com

foggiamanuel, 2021-07-27:
La modalità di visualizzazione a metà come si fa a farlo? Grazie

Younes481, 2021-07-27:
Bonjour,
Tu peux utiliser ddt4all pour rendre l"écran vertical ça existe sur youtube
Cordialement

foggiamanuel, 2021-07-27:
Ho cercato qualcosa del genere ma non ho trovato nulla in merito. Lei sa
come si può fare?

foggiamanuel, 2021-07-27:
Con ddt4all non riesco a farlo. Dove si può andare per apportare la
modifica su ddt4all?

foggiamanuel, 2021-07-27:
Poi un’altra cosa adesso ho lo schermo nero di nuovo. Come posso attivarlo?
Grazie infinite se mi può dare una mano.

AlMeu, 2024-02-15:
@moshetal402 leave an email and perhaps I can help you.

moshetal402, 2024-02-15:
> @moshetal402השאר אימייל ואולי אוכל לעזור לך.

mo3107539@gmail.com 
Thank you

Fre3DriiC89, 2024-06-30:
Hi, I have a Espace 2016, like all others i accidentally clicked on the wrong button... But my screen turned sideways and was splitted in the middle. Works fine but my wife is mad at me 🤣

fredrichultmark89@gmail.com if you can help me.