cedricp/ddt4all #97: Application freezes on test mode

felipetimoteo, 2018-09-07:
Hi, thank you for your hard work. I'm trying to unlock the android auto feature with a modified cable on my renault fluence dynamique plus. I'm using a wifi obd 2 scanner. The matter is when I try to launch the MFD v.4.6  the application freezes suddenly. I have unstalled the software and installed it again but it continuing freezing. I have tried the test button (without the modified extension cable) to see if i could get some error but it freezes during the test too. I'm using win 10 64 bits. I opened the log file, and here is the result.  could you help me ? 

>[00:30:02.234]ATZ
<[0.866]ATZ
ü

ELM327 v1.5

>
>[00:30:03.100] ATZ                   
<[0.87] ATZ                   
ü

ELM327 v1.5

>
>[00:30:04.011] ATE1                  
<[0.067] ATE1                  
OK

>
>[00:30:04.082] ATL0                  
<[0.068] ATL0                  
OK

>
>[00:30:04.154] ATI                   
<[0.069] ATI                   
ELM327 v1.5

>
>[00:30:04.228] AT@1                  
<[0.075] AT@1                  
OBDII to RS232 Interpreter

>
>[00:30:04.309] ATAL                  
<[0.068] ATAL                  
OK

>
>[00:30:04.385] ATBD                  
<[0.075] ATBD                  
00 00 00 00 00 00 00 00 00 00 00 00 00 

>
>[00:30:04.470] ATSP4                 
<[0.081] ATSP4                 
OK

>
>[00:30:04.550] ATBI                  
<[0.068] ATBI                  
OK

>
>[00:30:04.625] ATCAF0                
<[0.066] ATCAF0                
OK

>
>[00:30:04.700] ATCFC1                
<[0.068] ATCFC1                
OK

>
>[00:30:04.784] ATCP 01               
<[0.067] ATCP 01               
OK

>
>[00:30:04.859] ATCS                  
<[0.067] ATCS                  
T:00 R:00 

>
>[00:30:04.933] ATCV 1250             
<[0.079] ATCV 1250             
OK

>
>[00:30:05.018] ATD                   
<[0.068] ATD                   
üOK

>
>[00:30:05.092] ATDP                  
<[0.076] ATDP                  
ISO 14230-4 (KWP 5BAUD)

>
>[00:30:05.174] ATDPN                 
<[0.067] ATDPN                 
4

>
>[00:30:05.250] ATH0                  
<[0.067] ATH0                  
OK

>
>[00:30:05.326] ATI                   
<[0.069] ATI                   
ELM327 v1.5

>
>[00:30:05.402] ATIB 10               
<[0.074] ATIB 10               
KBusBaud=10400 OK

>
>[00:30:05.483] ATIB 96               
<[0.076] ATIB 96               
KBusBaud=9600 OK

>
>[00:30:05.565] ATL1                  
<[0.068] ATL1                  
OK

>
>[00:30:05.641] ATM0                  
<[0.068] ATM0                  

OK

>
>[00:30:05.715] ATMA                  
>[00:32:40.869]ATZ
<[00:32:40.934]STOPPED

><shifted>ATZ
<[00:32:40.934]TZ

?

><shifted>ATZ
>[00:33:50.716]ATZ
<[0.866]ATZ

ü

ELM327 v1.5

>
>[00:33:51.581] ATZ                   
<[0.869] ATZ                   
ü

ELM327 v1.5

>
>[00:33:52.488] ATE1                  
<[0.068] ATE1                  
OK

>
>[00:33:52.560] ATL0                  
<[0.067] ATL0                  
OK

>
>[00:33:52.631] ATI                   
<[0.069] ATI                   
ELM327 v1.5

>
>[00:33:52.706] AT@1                  
<[0.075] AT@1                  
OBDII to RS232 Interpreter

>
>[00:33:52.784] ATAL                  
<[0.066] ATAL                  
OK

>
>[00:33:52.853] ATBD                  
<[0.075] ATBD                  
00 00 00 00 00 00 00 00 00 00 00 00 00 

>
>[00:33:52.940] ATSP4                 
<[0.078] ATSP4                 
OK

>
>[00:33:53.018] ATBI                  
<[0.068] ATBI                  
OK

>
>[00:33:53.093] ATCAF0                
<[0.067] ATCAF0                
OK

>
>[00:33:53.164] ATCFC1                
<[0.067] ATCFC1                
OK

>
>[00:33:53.238] ATCP 01               
<[0.068] ATCP 01               
OK

>
>[00:33:53.311] ATCS                  
<[0.068] ATCS                  
T:00 R:00 

>
>[00:33:53.396] ATCV 1250             
<[0.079] ATCV 1250             
OK

>
>[00:33:53.482] ATD                   
<[0.067] ATD                   
üOK

>
>[00:33:53.554] ATDP                  
<[0.074] ATDP                  
ISO 14230-4 (KWP 5BAUD)

>
>[00:33:53.633] ATDPN                 
<[0.066] ATDPN                 
4

>
>[00:33:53.708] ATH0                  
<[0.068] ATH0                  
OK

>
>[00:33:53.787] ATI                   
<[0.069] ATI                   
ELM327 v1.5

>
>[00:33:53.864] ATIB 10               
<[0.074] ATIB 10               
KBusBaud=10400 OK

>
>[00:33:53.941] ATIB 96               
<[0.076] ATIB 96               
KBusBaud=9600 OK

>
>[00:33:54.019] ATL1                  
<[0.077] ATL1                  
OK

>
>[00:33:54.104] ATM0                  
<[0.067] ATM0                  

OK

>
>[00:33:54.181] ATMA

felipetimoteo, 2018-09-08:
Can I fix this issue or I have to buy another ELM ?

KarelSvo, 2018-09-08:
Either search until a Chinese elm clone found, which supports all necessary AT commands.

Or buy an interface with original elm firmware.
obdlink stn1110, stn1130
els27 stn1170, stn 2120
Polski ELM327 v2 stn1110
Polski ELM327 v3 stn1170
Elmino ELM 327 2.2 SM

KarelSvo, 2018-09-08:
version 1.5 is chinese clone
chance 50/50 works or not
https://lista.mercadolivre.com.br/obdlink#D[A:obdlink]

Furtif, 2023-06-02:
bretter speech in https://github.com/cedricp/ddt4all/discussions ....

closed 2023-06-02 by Furtif