cedricp/ddt4all #99: save default ECU settings megane 3

aliloup85, 2018-09-11:
Hi all,
i'm New here, same as many of ddt4all application, i've manipulated the ECU and provoked errors on my car. ie: my mirrors don't close anymore, there is no display of the external temperature ... i ask anyone Who wants to give help to upload the save of default settings present on megane 3 1.9 dci, year 2009. 

br,

aliloup85, 2018-09-12:
thanks for reply,  i want to know if a configuration of another Renault Mégane 3 1.9 dci could make my car function normally same as it was before.

cedricp, 2018-09-13:
Send me the log files, maybe there are some information about old config in there.
Tell me which ECU you modified too.

aliloup85, 2018-09-13:
cedric, merci bcp pour votre réponse, cependant je ne me souviens plus de quel calculateur il s'agit. dans ce cas je vais devoir me tourner vers un spécialiste pour qu'il corrige ce que j'ai gâché. merci encore une fois pour le grand travail que vous accomplissez.  bon courage à vous.

cedricp, 2018-09-13:
Avec les logs, je peux essayer de voir quel sont les parametres changes. Il faut effectivement etre assez prudent et faire des copies d'ecran des parametres avant ecriture. Il est aussi important de scanner le vehicule pour etre sur d'utiliser le bon fichier correspondant a l'ECU. Je pense qu'il doit s'agir de l'unite de controle habitacle qui est deparametre (UCH)

aliloup85, 2018-09-13:
j'aimerai bien que vous me disiez où se trouvent les logs M.Cedric?

cedricp, 2018-09-13:
Dans le dossier de l'application il y a un dossier "log"

aliloup85, 2018-09-13:
En effet, j'ai trouvé deux fichiers logs: ecu _ddt et elm_ddt. Je vous les transmettrai pas mail si possible.

cedricp, 2018-09-13:
C'est bien ca..

KarelSvo, 2018-09-14:
only one entry found with 2E

#connect to: CLUSTER_B95_L38_ph2_V1.2 Addr:51
>[21:13:50.931]042E012210
<[0.075]042E012210
037F2E12FFFFFFFF

shrlnm, 2018-09-14:
Please be careful. There are logs from at least two different cars.

In logs most changes were made in UCH  (CANid = 745) . It is STD-A so the write command starts with 3B (not with 2E)

cedricp, 2018-09-14:
Right,
There is **T4_VS_BCM_DDT2000_SW16_5** and **BCM95_SWB10_VB10_4**

shrlnm, 2018-09-14:
Too many changes (probably because second car is in the log)

I do not know which one is related to mirror closing

###############################

BCM95_SWB10_VB10_4.xml

< Changed
> Should be 

------------------------------
REMOTE_LIGHTING_CF
< 0x01
> 0x00
------------------------------
SUPERSLEEP_FITTED_CF
< 0x01
> 0x00
------------------------------
REAR_FOG_OFF_BY_SMART_CF
< 0x01
> 0x00
------------------------------
LONG_PRESS_TM_CF
< 0x05
> 0x00
------------------------------
MAX_TURN_POWER_LEVEL_X_ON_TM_CF
< 0x05
> 0x00
------------------------------
BATTERY_SAVE_CF
< 0x01
> 0x00
------------------------------
MAX_L3_ON_BY_DIAG_TM_CF
< 0x3c
> 0x00
------------------------------
DUAL_BULBS_CF
< 0x01
> 0x00
------------------------------
FOLLOW_ME_HOME_CF
< 0x01
> 0x00
------------------------------
RKE_BACK_DOORS_CF
< 0x01
> 0x00
------------------------------
HAZARD_WITH_CRASH_CF
< 0x01
> 0x00
------------------------------
AMBIANCE_LIGHTS_CF
< 0x01
> 0x00
------------------------------
CHILD_LOCK_SWITCH_REVERSE_CF
< 0x01
> 0x00
------------------------------
CRASH_DETECTION_CF
< 0x01
> 0x00
------------------------------
OVERLOAD_BY_SMART_CF
< 0x01
> 0x00
------------------------------
LVE_CF
< 0x01
> 0x00
------------------------------
SEPARATED_BACK_LIGHT_CF
< 0x01
> 0x00
------------------------------
PLANT_LOCK_WITH_SPEED_CF
< 0x01
> 0x00
------------------------------

cedricp, 2018-09-14:
I isolated BCM sessions where there are 3B commands in the ZIP (session1 is the older, session3, the newer)

session1: https://pastebin.com/wk1Wta0w
session2: https://pastebin.com/RRNDVbve
session3: https://pastebin.com/YBBfpVrh

Sorry don't have time to analyse them further.

cedricp, 2018-09-14:
I'm sorry for you, this has led me to add some useful logs to recover original data. I'll push the code ASAP.
Here's a sample of the output : 
`ECU : UCT_E84_SERIE_3.json
	@ 14:05:16.137
	Screen : Misc Values and Timings	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:05:16.144
	Screen : Misc Values and Timings	Request : ReadDataByLocalIdentifier: misc timings and values
		{"Voltage High 1": "193", "Timeout Button Windows": "60", "Voltage High 3": "181", "Voltage Low 1": "92", "Timeout Supply Voltage Range 3": "1000", "Voltage Low 3": "103", "Voltage Low 2": "97", "Timeout Runtime Stop Mode": "60000", "Voltage High 2": "187", "Timeout Window Button Short Push": "220", "Maximum Veh. Speed": "3", "Timeout Supply Voltage Range 5": "1000", "Timeout Supply Voltage Range 4": "1000", "Timeout Supply Voltage Range 1": "6000", "Multiplier Thermo Prot.": "10", "Timeout Supply Voltage Range 2": "1000", "Timeout Button Roof": "60", "Timeout Roof Button Window Close": "500", "Tolerance Window Button Open Timer": "5000", "Timeout Runtime Hydr. Pump": "18000", "Timeout Buzzer Alert": "30"}
	@ 14:05:23.714
	Blocked writing parameter : 
		{"Timeout Roof Button Window Close": "500", "Voltage High 1": "193", "Timeout Button Windows": "60", "Voltage High 3": "181", "Maximum Veh. Speed": "3", "Timeout Supply Voltage Range 3": "1000", "Voltage Low 3": "103", "Voltage Low 2": "97", "Timeout Runtime Stop Mode": "60000", "Voltage High 2": "187", "Timeout Window Button Short Push": "220", "Timeout Supply Voltage Range 5": "1000", "Timeout Supply Voltage Range 4": "1000", "Timeout Supply Voltage Range 1": "6000", "Multiplier Thermo Prot.": "10", "Timeout Supply Voltage Range 2": "1000", "Timeout Button Roof": "60", "Voltage Low 1": "92", "Tolerance Window Button Open Timer": "5000", "Timeout Runtime Hydr. Pump": "18000", "Timeout Buzzer Alert": "80"}
	@ 14:05:26.941
	Screen : Settings	Request : ReadDataByLocalIdentifier: traceability data
		{"Traceability Data": "\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}
	@ 14:05:26.942
	Screen : Settings	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:05:26.943
	Screen : Settings	Request : ReadDataByLocalIdentifier: endurance mode
		{"Endurance Mode": "00"}
	@ 14:05:26.944
	Screen : Settings	Request : ReadDataByLocalIdentifier: rollbar, montage
		{"Rollbar Present": "00", "Montage Mode": "00", "Window Mode": "00"}
	@ 14:05:31.938
	Blocked writing parameter : 
		{"Endurance Mode": "Normal Mode"}
	@ 14:05:34.618
	Blocked writing parameter : 
		{"Rollbar Present": "Rollbar present", "Montage Mode": "Montage Mode", "Window Mode": "after 1s"}
	@ 14:06:02.105
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: times latches
		{"Tolerance Right Latch": "0", "Tolerance Left Latch": "0"}
	@ 14:06:02.106
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:06:02.107
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 2
		{"Tolerance Start Hydraulic (close roof)": "0", "Tolerance Close Trunklid (close roof)": "0", "Tolerance Close Roof (close roof)": "0", "Tolerance Open Trunklid (close roof)": "0"}
	@ 14:06:02.108
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 1
		{"Tolerance Open Trunklid (open roof)": "0", "Tolerance Open Roof (open roof)": "0", "Tolerance Start Hydraulic (open roof)": "0", "Tolerance Close Trunklid (open roof)": "0"}
	@ 14:06:02.109
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: timeout values 1
		{"Timeout Open Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 2 (open roof)": "0", "Timeout Unlock Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 1 (open roof)": "0", "Timeout Unlock Trunklid 2 (open roof)": "0", "Timeout Open Roof 1 (open roof)": "0", "Timeout Open Roof 2 (open roof)": "0"}
	@ 14:06:02.111
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: timeout values 2
		{"Timeout Unlock Trunklid 1 (close roof)": "0", "Timeout Close Trunklid 1 (close roof)": "0", "Timeout Unlock Trunklid 2 (close roof)": "0", "Timeout Close Trunklid 2 (close roof)": "0", "Timeout Open Trunklid 1 (close roof)": "0", "Timeout Close Roof 1 (close roof)": "0", "Timeout Close Roof 2 (close roof)": "0"}
	@ 14:06:09.967
	Writing parameter : 
		{"Timeout Open Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 2 (open roof)": "0", "Timeout Unlock Trunklid 1 (open roof)": "0", "Timeout Close Trunklid 1 (open roof)": "56", "Timeout Unlock Trunklid 2 (open roof)": "0", "Timeout Open Roof 1 (open roof)": "0", "Timeout Open Roof 2 (open roof)": "0"}
	@ 14:06:10.092
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: times latches
		{"Tolerance Right Latch": "0", "Tolerance Left Latch": "0"}
	@ 14:06:10.093
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: general system infos
		{"Status EEPROM": "Writing"}
	@ 14:06:10.094
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 2
		{"Tolerance Start Hydraulic (close roof)": "0", "Tolerance Close Trunklid (close roof)": "0", "Tolerance Close Roof (close roof)": "0", "Tolerance Open Trunklid (close roof)": "0"}
	@ 14:06:10.095
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: tolerance values 1
[... cut ...]  "Timeout Open Roof 1 (open roof)": "0", "Timeout Open Roof 2 (open roof)": "0"}
	@ 14:06:10.099
	Screen : Timouts & Tolerances	Request : ReadDataByLocalIdentifier: timeout values 2
		{"Timeout Unlock Trunklid 1 (close roof)": "0", "Timeout Close Trunklid 1 (close roof)": "0", "Timeout Unlock Trunklid 2 (close roof)": "0", "Timeout Close Trunklid 2 (close roof)": "0", "Timeout Open Trunklid 1 (close roof)": "0", "Timeout Close Roof 1 (close roof)": "0", "Timeout Close Roof 2 (close roof)": "0"}`

aliloup85, 2018-09-16:
Thanks for Yr help, I'm still waiting for what I should do once the code is ready😊

cedricp, 2018-09-16:
This patch will unfortunately not help you, it's to avoid that kind of problem i the future. Please try to remember which screen you changed parameters on so I can try to recover your original data.

aliloup85, 2018-09-16:
Thanks CEDRIC, so I think that I have to give my car to a specialist that he'll be able to recover my ECU configuration. Once again, thanks to everyone that tried to help me.

cedricp, 2018-09-17:
Here a list of your write commands (3B) : 
**3B 14** -> Write doorlock 9
**3B 0E** -> Window lift calibration
**3B 11** -> Write doorlock 6
**3B 18** -> Write doorlock 3

cedricp, 2018-09-17:
Here are your original parameters for doorlock screen : 
**Read_Door_Lock_configuration** 
{
"LONG_PRESS_TM_CF": "500",
 "AUTO_RELOCK_TIMEOUT_CF": "2",
 "SEPARATED_TAILGATE_OPENING_CF": "without",
 "BTN_TAILGATE_OPENING_CF": "without",
 "POWER_SLIDING_DOORS_CF": "without",
 "FUEL_WAC_LOCKING_DELAY_TM_CF": "0",
 "ANTI_HIJACK_CF": "without",
 "CHILD_LOCK_CF": "with",
 "KNOB_CF": "without",
 "HANDS_FREE_CF": "without",
 "CHILD_LOCK_SWITCH_REVERSE_CF": "without",
 "RKE_BACK_DOORS_CF": "with",
 "GLASSHATCH_CF": "without",
 "HIGH_LOCK_WITH_SPEED_CF": "without",
 "CAR_BODY_TYPE_E_CF": "false",
 "RKE_BTN_TAILGATE_OPENING_CF": "without",
 "SUPERLOCK_CF": "without",
 "CRASH_DETECTION_CF": "with",
 "PLANT_LOCK_WITH_SPEED_CF": "with"
}

cedricp, 2018-09-17:
And for window calibration : 
**Read_Window_Control_Calibration_Data**
{
"FRONT_PASSENGER_WLAP_ECU_CF": "with",
 "AUTO_WINDOW_CLOSING_CF": "with",
 "SHORT_DROP_CF": "without",
 "REAR_PASSENGER_WLAP_ECU_CF": "with",
 "LVE_CF": "without"
}

Restore these parameters and check if it works. I cannot do better on my side.

aliloup85, 2018-09-17:
Merci beaucoup cher CEDRIC, j'essaierai et vous tiendrai au courant.

aliloup85, 2018-09-22:
Thanks CEDRIC! I fixed the child security thanks to your instructions. I would like to fix the other problems (mirror closing and external temperature display) when you have free time.thanks a lot again.

aliloup85, 2018-09-22:
Hello everyone! Finally I've fixed the mirror folding and the temperature display. I've found the settings on "dcm" where all the settings are disabled so if u enable an option other options will be disabled, so u have to re-enable them at the same time you're enabling other ones. Thanks for everyone.
br,

cedricp, 2018-09-23:
Yes, on this screen, you have to set all drop-down box parameters before writing.

closed 2018-09-23 by aliloup85

kadirayhantr, 2020-01-27:
> Evet, bu ekranda yazmadan önce tüm açılır kutu parametrelerini ayarlamanız gerekir.

Hi,  I want to activate my power folding mirror feature.  First I changed my side mirrors and front left window and mirror open/close switch.  Than I changed the status of mirror folding section from 0 to 1 . But the mirrors didn't move.  My DDCM :DDCM X95 SLOT v4.0   and my UCH : BCM95_SW25_V1.1
Can anybody do this ?  Please help mw how to do. Thanks