cedricp/ddt4all #102: Can not open port

bokkoman, 2018-09-30:
I've been using ddt4all a lot lately to install android auto for people.
But at my latest try, i got an error that ddt4all can not open port, acces denied error 5.

I don't know why this is happening. I can't use my elm device anymore. Tried changing the port, but with no success.

It happened after i tried an ELM test. Someone asked if i could do that too see difference with his ELM.
Didn't think any of it, so i just did.
So 2 days after that, i tried to connect with a car, but that error came up. 

Does someone know why? Or could help me out?

cedricp, 2018-10-03:
What is your OS (Win, Linux, Mac) ?

bokkoman, 2018-10-03:
I'm on Windows 10.
 Is it maybe cause i did the ELM test with the elm adapter connected to my laptop but not to my car?

pcboys, 2018-10-03:
If you elm only connected to PC, then it is no power support, so,  is not work.

cedricp, 2018-10-03:
You indeed need power supply from the car to power on the ELM chip.

bokkoman, 2018-10-03:
The ELM test without connected to car worked, it gave 100%. Now when i connect to car and laptop i get error 5.

KarelSvo, 2018-10-07:
Is your problem also present with this installer version?
https://www.drive2.ru/b/498093336985338243/

bokkoman, 2018-10-07:
I bought a new elm device, and with that one there is no problem. It's the exact same device.
So i guess something internally on the elm went wrong when doing the elm test without connecting to a car.

nickischenker, 2018-10-29:
Same Issue

Using an [OBDLink SX](http://www.obdlink.com/sxusb/) on Win7/32
[gives me an error](https://picload.org/view/dcaiocdr/error_v14.jpg.html)
..,got with one of these Chinese ELM327 forks a similar result. This only happens with v1.4, v1.3 is ok.

cedricp, 2018-10-29:
There's a file missing in the installer, get the latest source code and replace it in the installation directory, it should work...

nickischenker, 2018-10-29:
thx, v1.4 is working. :D

closed 2019-01-08 by cedricp